MichaIng/DietPi #4468: Update Error 7.0.2 to 7.2.3

Schlummifix, 2021-06-05:
#### Details:
- Date           | Sat Jun  5 05:46:37 CEST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
want to update from 7.0.2 to 7.2.3 with dietpi-udate
failed. even reeboot did not solve

Joulinar, 2021-06-05:
Hi,

Some more information about the issue would be helpful.

Please can you post `apt update && apt upgrade`

Schlummifix, 2021-06-05:
Thank You! After doing manually a apt update && apt upgrade
the dietpi-update was success

Joulinar, 2021-06-05:
ok in this case I'm going to close the issue.

closed 2021-06-05 by Joulinar