MichaIng/DietPi #4469: .NET Core [as an independent software title]

fpetru, 2021-06-05:
#### Is your feature request related to a problem? Please describe:
It is not currently possible to select .NET installation - it is only installed & configured as a dependecy 

#### Describe the solution you'd like:
Adding it distinctly, would enable users to select and run .NET projects. Furthermore, using  [dotnet/interactive](https://github.com/dotnet/interactive) it will be so much easier to write code, see the results or share them. 

C# is the 5th programming language [link](https://www.tiobe.com/tiobe-index/) and it is already supported by DietPi (but currently only as dependency).

DocBrown101, 2022-11-26:
Is there anything new here yet or at least a roadmap?

Joulinar, 2022-11-27:
Not actually

MichaIng, 2023-01-14:
Install instructions via APT repo: https://learn.microsoft.com/dotnet/core/install/linux-debian