MichaIng/DietPi #4470: Update shairport-sync binary to publish to pipe/stdout for snapcast server

foxy82, 2021-06-05:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

snapcast server wants shairport to publish audio to a pipe/stdout however that doens't look like it is available:

```
2021-06-05 11-24-59.350 [Info] (AirplayStream) (MyAirport) fatal error: Invalid audio backend "stdout" selected!
```

#### Describe the solution you'd like:

Add the following options to the compilation of shairport-sync:

`--with-stdout` 
`--with-pipe`

MichaIng, 2021-06-05:
As Shairport Sync runs as systemd unit, `stdout` is done to the journal and contains startup and operational logs. Not sure how Snapcast would read an audio stream from Shairport Sync STDOUT in general, if not by redirecting it to a file, FIFO, socket, or such, so I think `pipe` is what we want here, which seems to work like the MPD FIFO and allows to cleanly separate process/daemon operational output from the audio stream.

I just updated the Shairport Sync packages some days ago, so it might take a while, e.g. to include the next upstream version upgrade or so.

foxy82, 2021-06-05:
Snapcast spins up it's own copy of airport sync so will be running inside the snapcast process - I guess this is how it uses stdout. 

I'll see if I can build my own in the meantime I'll try with just pipe to start with and if that is enough we can just use that.

foxy82, 2021-06-05:
I've just tried this and it is in fact --with-stdout that we need.

I used this config line locally:

```
./configure --sysconfdir=/etc --with-alsa --with-soxr --with-avahi --with-ssl=openssl --with-systemd --with-metadata --with-stdout
```

MichaIng, 2021-06-06:
Currently:
```
--with-alsa --with-avahi --with-ssl=openssl --with-soxr --with-metadata --with-systemd --with-dbus-interface --with-mpris-interface --with-mqtt-client
```
Okay if Snapcast spins up an own Shairport Sync process, than this makes sense, and looks actually elegant. The only problem then is that our system service is then obsolete and might block the same network port. When when setting it up that way, the system service needs to be excluded from dietpi-services control. We need to add a include and exclude commands to the dietpi-services CLI to make this easier than navigating through the whiptail menus: `dietpi-services exclude shairport-sync`

samaero, 2022-04-28:
Hi there, wondering if there was any progress integrating this into dietpi? Looks like the only available backend for the bundled shairport-sync in v8.3.1 is still alsa.

Joulinar, 2022-05-10:
not 100% same but there is a request  on our forum to update the package to a specific version to support Airplay 2 https://dietpi.com/phpbb/viewtopic.php?t=10362

MichaIng, 2022-10-15:
New packages are available with pipe and stdout backend support added:
```sh
dietpi-software reinstall 37
```

closed 2022-10-15 by MichaIng