MichaIng/DietPi #4471: Problem with upgrade  to 7.2.3

maury77, 2021-06-05:
E: The repository 'https://dl.bintray.com/openhab/apt-repo2 stable InRelease' is no longer signed.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -q update
[ INFO ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-Bugreport | Packing upload archive
[  OK  ] DietPi-Bugreport | Sending bug report
[ INFO ] DietPi-Bugreport | Reference code: 3367cd8d-87ca-4a59-85e9-807a4979358e

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sat Jun  5 15:47:45 CEST 2021
- Bug report     | 3367cd8d-87ca-4a59-85e9-807a4979358e
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux sdomotica 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:2 https://dl.bintray.com/openhab/apt-repo2 stable InRelease
  403  Forbidden [IP: 18.196.33.98 443]
Hit:3 https://dtcooper.github.io/raspotify jessie InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: Failed to fetch https://dl.bintray.com/openhab/apt-repo2/dists/stable/InRelease  403  Forbidden [IP: 18.196.33.98 443]
E: The repository 'https://dl.bintray.com/openhab/apt-repo2 stable InRelease' is no longer signed.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

ravenclaw900, 2021-06-05:
It appears that this is an issue with your installation of openHAB.  Looking at the docs, that repository seems to be deprecated.  Could you run:
```sh
rm /etc/apt/trusted.gpg.d/openhab.gpg /etc/apt/sources.list.d/openhab.list
curl -sSfL 'https://openhab.jfrog.io/artifactory/api/gpg/key/public' | gpg --dearmor -o /etc/apt/trusted.gpg.d/openhab.gpg
echo 'deb https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable main' > /etc/apt/sources.list.d/openhab.list
```
Then run `apt update && apt upgrade` and post the output.

maury77, 2021-06-05:
thanks I resolved with your help

closed 2021-06-05 by maury77

MichaIng, 2021-06-06:
Should we probably check and remove `dl.bintray.com` from sources lists via pre-patches? Of course an interactive whiptail prompt would be shown as well to explain that this repository does not exist anymore and an alternative would be required, if needed at all.

Joulinar, 2021-06-06:
if that would be a real issue, we would have more messages already, isn't it?

That's the official announcement from end of April https://community.openhab.org/t/the-stable-release-linux-repositories-have-changed-location/121261

MichaIng, 2021-06-06:
> if that would be a real issue, we would have more messages already, isn't it?

We did, I guess 5 - 10 cases appeared across GitHub and forum for the multiple 3rd party repos that were hosted by Bintray. But of course relevance drops over time, as the service shutdown happened at end of March.

Joulinar, 2021-06-06:
For `openhab` ? I guess it was Ondrej PHP repository. Maybe I mixed it. 🤣

MichaIng, 2021-06-06:
Not (only) openHAB, but Bintray hosted a lot of 3rd party repositories and binaries, openHAB was only one:
- https://dietpi.com/phpbb/search.php?keywords=bintray
- https://github.com/MichaIng/DietPi/search?q=bintray&type=issues

Grafana, InfluxDB, Tvheadend, the TigerVNC binaries.