MichaIng/DietPi #4472: Gift that keeps on giving: unable to complete a simple update/upgrade

MAGA6226, 2021-06-06:
#### Details:
- Date           | Sun Jun  6 16:12:01 CDT 2021
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux dietpi20053101 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...routine update
2. ...routine upgrade
#### Expected behaviour:

- ...exactly that
#### Actual behaviour:

- ...not exactly that
#### Extra details:

- ...per instructions
#### Additional logs:
```
Setting up bind9 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
bind9-pkcs11.service is a disabled or a static unit not running, not starting it                         .
bind9-resolvconf.service is a disabled or a static unit not running, not starting it.
Job for bind9.service failed because the control process exited with error code.
See "systemctl status bind9.service" and "journalctl -xe" for details.
invoke-rc.d: initscript bind9, action "restart" failed.
● bind9.service - BIND Domain Name Server
   Loaded: loaded (/lib/systemd/system/bind9.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-06-06 16:12:00 CDT; 55ms ago
     Docs: man:named(8)
  Process: 16079 ExecStart=/usr/sbin/named $OPTIONS (code=exited, status=1/FAILURE)

Jun 06 16:12:00 dietpi20053101 named[16083]: found 4 CPUs, using 4 worker threads
Jun 06 16:12:00 dietpi20053101 named[16083]: using 3 UDP listeners per interface
Jun 06 16:12:00 dietpi20053101 named[16083]: using up to 4096 sockets
Jun 06 16:12:00 dietpi20053101 named[16083]: loading configuration from '/etc/bind/named.conf'
Jun 06 16:12:00 dietpi20053101 named[16083]: /etc/bind/named.conf:13: unexpected end of input
Jun 06 16:12:00 dietpi20053101 named[16083]: loading configuration: unexpected end of input
Jun 06 16:12:00 dietpi20053101 named[16083]: exiting (due to fatal error)
Jun 06 16:12:00 dietpi20053101 systemd[1]: bind9.service: Control process exited, code=exited, status=1/FAILURE
Jun 06 16:12:00 dietpi20053101 systemd[1]: bind9.service: Failed with result 'exit-code'.
Jun 06 16:12:00 dietpi20053101 systemd[1]: Failed to start BIND Domain Name Server.
dpkg: error processing package bind9 (--configure):
 installed bind9 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 bind9
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Joulinar, 2021-06-06:
Hi,

looks like you have `BIND Domain Name Server` installed. This software package is going to be updated by `apt upgrade`, which is failing due to incorrect configuration file `/etc/bind/named.conf`

```
Jun 06 16:12:00 dietpi20053101 named[16083]: /etc/bind/named.conf:13: unexpected end of input
Jun 06 16:12:00 dietpi20053101 named[16083]: loading configuration: unexpected end of input
Jun 06 16:12:00 dietpi20053101 named[16083]: exiting (due to fatal error)
```

looks the file is not ending as expected by `bind`. I guess you would need to check your config.

MichaIng, 2021-06-07:
Here is the default config, as shipped by the Debian Buster package. It only loads additional config files:
```
// This is the primary configuration file for the BIND DNS server named.
//
// Please read /usr/share/doc/bind9/README.Debian.gz for information on the 
// structure of BIND configuration files in Debian, *BEFORE* you customize 
// this configuration file.
//
// If you are just adding zones, please do that in /etc/bind/named.conf.local

include "/etc/bind/named.conf.options";
include "/etc/bind/named.conf.local";
include "/etc/bind/named.conf.default-zones";
```

But as often, the question is whether you actually use that DNS server (as the systems itself obviously does not use it and the services was likely failing before?). If not, I suggest to purge it: `apt purge bind9`

MichaIng, 2021-06-27:
I mark this issue as closed. Feel free to reopen if required.

closed 2021-06-27 by MichaIng