MichaIng/DietPi #4473: Diet-Pi Software | Prowlarr - Indexer Manager & Proxy for PVR

EthanC, 2021-06-07:
### Creating a software request

#### Formal software information
- Software title | Prowlarr
- Short description | Prowlarr is an indexer manager/proxy built on the popular arr .net/reactjs base stack, to integrate with your various PVR apps.
- Official URL (if available) | https://wikijs.servarr.com/prowlarr
- GitHub URL (if available) | https://github.com/prowlarr/prowlarr
- (Official) install documentation | https://wikijs.servarr.com/prowlarr/installation

#### Are there similar/alternative software titles available with DietPi-Software?
- Jackett
- NZBHydra2

#### What makes your requested software better than the above solutions, if available?
- Prowlarr supports both Torrent Trackers and Usenet Indexers. It integrates seamlessly with Sonarr, Radarr, Lidarr, and Readarr offering complete management of your indexers with no per-app Indexer setup required.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- Prowlarr is based on the software stack used by both Radarr and Sonarr, both of which are currently available in DietPi-Software. There are multiple dependencies as well as assumptions of which user and group the software is being run under.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
- I have yet to successfully install Prowlarr on DietPi as I, and I imagine many other users, would prefer to have it done in the most optimal way by DietPi-Software.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.?
- Unfortunately at this time I would not be of much help with this solution.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/
- https://feathub.com/MichaIng/DietPi/+233

Tooloco, 2021-11-26:
Hello, I was just going to request this software as well. As apparently there's a bug with Jackett that uses 100% of cpu when enabling proxy config. Prowlarr is also easier to setup and belongs to the arr family. As far as I can see (haven't tested) the only downside would be the limited availability of indexers (only 24) but we can always have jackett and prowlarr installed. Info from: https://github.com/Prowlarr/Prowlarr and https://wiki.servarr.com/prowlarr

MichaIng, 2021-12-06:
For the record: This software title got 10 votes in ~6 months on FeatHub: https://feathub.com/MichaIng/DietPi/+233

However, since FeatHub is not reliable, and even that the HTTPS certificate got renewed recently, GitHub authentication still does not work. We hence move software requests for voting to GitHub discussions. Place your vote via up arrow at the first post above.