MichaIng/DietPi #4475: Shairport Sync Install issue

mprince28117, 2021-06-08:
When upgrading to the current DietPi version, the install errored out on a Shairport sync issue.  To get around this, I uninstalled the package and completed the upgrade.  Now I can't reinstall the package, bug report below:

#### Details:
- Date           | Tue Jun  8 11:35:51 EDT 2021
- Bug report     | ed987da7-8247-4ac6-ac0e-f58a33c89e02
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install ./shairport-sync_armv7l.deb
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install or reinstall Shairport Sync from DietPi-Software...
2. ...
#### Expected behaviour:

- Package should install
#### Actual behaviour:

- Install errors out, indicates package(s) are bad
#### Extra details:

- [FAILED] DietPi-Software | APT install for: ./shairport-sync_armv7l.deb
 - Command: apt-get -qq --allow-change-held-packages install ./shairport-sync_armv7l.deb
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2021-06-08:
Many thanks for your report. Issue was that the package from the Debian repository has a newer version string of one of the dependencies than the Raspbian repository. I just fixed it, please retry:
```
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/stretch/shairport-sync_armv7l.deb
apt install ./shairport-sync_armv7l.deb
rm shairport-sync_armv7l.deb
```

mprince28117, 2021-06-08:
Success, many thanks!

closed 2021-06-08 by mprince28117