MichaIng/DietPi #4478: Unable to locate package python3-rpi.gpio on VMware.

TheAdminFrmoHell, 2021-06-10:
Not sure if it's something I'm doing or if it is a bug, someone else confirm?
I am using the latest DietPi_VMware-x86_64-Buster image on a standard PC running VMware ESXi. (The image is converted using VMware vCenter Converter Standalone Client)

Reproduce: Select the node-red package during install.

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Python 3 RPi.GPIO: Control Raspberry Pi GPIO channels in Python 3

[ INFO ] DietPi-Software | APT install for: python3-rpi.gpio, please wait...
E: Unable to locate package python3-rpi.gpio
E: Couldn't find any package by glob 'python3-rpi.gpio'
E: Couldn't find any package by regex 'python3-rpi.gpio'
**[FAILED] DietPi-Software | APT install for: python3-rpi.gpio
 - Command: apt-get -qq --allow-change-held-packages install python3-rpi.gpio**
[ INFO ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-Bugreport | Packing upload archive
[  OK  ] DietPi-Bugreport | Sending bug report
[ INFO ] DietPi-Bugreport | Reference code: **53214139-6b4c-4fc3-b145-1102914bdca5**
root@DietPi:/tmp/DietPi-Software# apt-get -qq --allow-change-held-packages install python3-rpi.gpio
**E: Unable to locate package python3-rpi.gpio
E: Couldn't find any package by glob 'python3-rpi.gpio'
E: Couldn't find any package by regex 'python3-rpi.gpio'**
root@DietPi:/tmp/DietPi-Software#

Joulinar, 2021-06-10:
Hi,

Well this is a package to manage GPIO on Raspberry Pi. Why do you like to install it on a VM?

TheAdminFrmoHell, 2021-06-10:
> 
> 
> Hi,
> 
> Well this is a package to manage GPIO on Raspberry Pi. Why do you like to install it on a VM?

it's not a something I am selecting to use, it's done automagically when I try to install node-red :)

Joulinar, 2021-06-10:
Ok let me check

TheAdminFrmoHell, 2021-06-10:
Cheers

Joulinar, 2021-06-10:
@MichaIng 
can you have a look pls. I guess this was introduced with the new dependency system. 

For `Node-Red` we have a defined dependency of `Python 3 RPi.GPIO` (ID 69)
https://github.com/MichaIng/DietPi/blob/4ffa0e5ef0d8fd6fa241535ad96eae5482135797/dietpi/dietpi-software#L1241

However, `Python 3 RPi.GPIO` is available on RPi only and deactivated for other devices
https://github.com/MichaIng/DietPi/blob/4ffa0e5ef0d8fd6fa241535ad96eae5482135797/dietpi/dietpi-software#L1179-L1183

Before this change, we invoked `python-rpi.gpio` package directly inside the `Node-Red` install

```
			# Pre-reqs
			local apackages=('python' 'python3')
			# - RPi: GPIO control for Node-RED
			(( $G_HW_MODEL < 10 )) && apackages+=('python-rpi.gpio')
			G_AGI "${apackages[@]}"
```

but atm, this is not the case anymore. Probably we would need to revert this back. What do you think?

---

@TheAdminFrmoHell 
as a workaround, pls do following

1. once hit by the error, you will get some error handle options
2. pls select `Change command`
3. remove the entire command you see 
4. and replace it with the word `true`
5. you should be able to complete the install now

TheAdminFrmoHell, 2021-06-10:
Thanks, I manually installed it node-red before I made the first post, it seems to be working fine. But, I'm bound to mess this sh*t up so I will most def do this on my next reinstall.

MichaIng, 2021-06-10:
Many thanks for your report. Indeed this dependency should have been set on RPi only.
Fixed with: https://github.com/MichaIng/DietPi/commit/9f1094f96dfbf4dd7de4cabe9c3fb9ae22e53a4c
Changelog: https://github.com/MichaIng/DietPi/commit/f548f9710baf5ee550223b95a24f17f86829ff60

TheAdminFrmoHell, 2021-06-15:
Happy days, I'll make a new ticket if I find any other issues with my weird setup. Keep the bucket safe 🤘

closed 2021-06-15 by TheAdminFrmoHell