MichaIng/DietPi #4479: dietpi-drive_manager gives errors for mounted network drives

maartenlangeveld, 2021-06-10:
```
 ─────────────────────────────────────────────────────
 DietPi v7.2.3 : 10:32 - Thu 06/10/21
 ─────────────────────────────────────────────────────
 - Device model : NanoPi NEO2 (aarch64)
 - Uptime : up 25 minutes
 - CPU temp : 40'C : 104'F (Optimal temperature)
 - FQDN/hostname : M-DNS
 - LAN IP : 192.x.x.x (eth0)
 - WAN IP : x.x.x.x South Holland Netherlands
 - Freespace (RootFS) : 23G
 - MOTD : DietPi v7.2 has been released, please update now.
 ─────────────────────────────────────────────────────

root@M-DNS:/mnt# dietpi-drive_manager
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected mounted physical drive: /dev/mmcblk0p1 > /
[ INFO ] DietPi-Drive_Manager |  - Detected mounted physical drive: 192.x.x.x:/ > /mnt/nanopi-backup-rpi-zero-w
[FAILED] DietPi-Drive_Manager | The following drive seems to not have a UUID, skipping fstab entry:
SOURCE           TARGET                        FSTYPE UUID
192.x.x.x:/ /mnt/nanopi-backup-rpi-zero-w nfs4
[ INFO ] DietPi-Drive_Manager |  - Detected mounted physical drive: 192.x.x.x:/mnt/main/backup/nanopi-backup/dietpi-backup > /mnt/nanopi-backup-m-server
[FAILED] DietPi-Drive_Manager | The following drive seems to not have a UUID, skipping fstab entry:
SOURCE                                                     TARGET                      FSTYPE UUID
192.x.x.x:/mnt/main/backup/nanopi-backup/dietpi-backup /mnt/nanopi-backup-m-server nfs4
[ INFO ] DietPi-Drive_Manager |  - Detected unmounted drive: /dev/sda1
```

Joulinar, 2021-06-10:
Hi,

what typ of network drive is this? NFS?

MichaIng, 2021-06-10:
Many thanks for reporting.

A tricky thing. The physical drive detection was not optimal anyway, as it theoretically allowed to have drives detected as physical ones, even when they had no `/dev/` entry, respectively their mount source did not start with `/dev/`, which is expected in many other places. With this change we use this prefix now to device whether it's physical or not: https://github.com/MichaIng/DietPi/commit/cb95aa5d51c19a910908fd3daa36935ec2860ada

But the other way round, there are potentially mount types (encrypted filesystems, LVM drives) now detected as physical ones, which were detected as network drives before. The script will hence now add fstab entries for those, at least if it has a UUID as well, while before, any existing fstab entry for it would have been lost. Not sure whether it's better or not (if the fstab entry is correct in these cases), but it needs to be kept an eye one.

maartenlangeveld, 2021-06-10:
It concerns NFS network drives, have not tested with SAMBA network drives.

MichaIng, 2021-06-10:
It did affect all network drives. At least the result is exactly the same as if it was detected correctly. Only you'll see some unintended options, like the ability to format it or create a swap file, which both does not make sense for a network drive 😄.

closed 2021-06-26 by MichaIng