MichaIng/DietPi #4480: Docker install failed on Allo Sparky

davindisko, 2021-06-10:
Hi,
I reinstalled Dietpi from scratch and when I try to install Roon Extension, Docker fails during update.

```
#### Details:
- Date           | Thu Jun 10 20:55:05 BST 2021
- Bug report     | 3e13077a-1e6d-457e-bb6f-d9fc8b7acd49
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | DietPi
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages dist-upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
Setting up docker-ce (5:20.10.7~3-0~debian-buster) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2021-06-10 20:55:05 BST; 34ms ago
     Docs: https://docs.docker.com
  Process: 3158 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
```

davindisko, 2021-06-10:
```
Jun 10 21:09:04 DietPi systemd[1]: Starting Docker Application Container Engine...
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.928923432+01:00" level=info msg="Starting up"
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.935112849+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.935290016+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.935492599+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0
 <nil>}] <nil> <nil>}" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.935743349+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.943157932+01:00" level=info msg="parsed scheme: \"unix\"" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.943339016+01:00" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.943562849+01:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock  <nil> 0
 <nil>}] <nil> <nil>}" module=grpc
Jun 10 21:09:04 DietPi dockerd[4118]: time="2021-06-10T21:09:04.943735932+01:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Jun 10 21:09:05 DietPi dockerd[4118]: time="2021-06-10T21:09:05.019803349+01:00" level=error msg="[graphdriver] prior storage driver devicemapper failed: devicemapper: Error running device
Create (CreatePool) dm_task_run failed"
Jun 10 21:09:05 DietPi dockerd[4118]: failed to start daemon: error initializing graphdriver: devicemapper: Error running deviceCreate (CreatePool) dm_task_run failed
Jun 10 21:09:05 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Jun 10 21:09:05 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Jun 10 21:09:05 DietPi systemd[1]: Failed to start Docker Application Container Engine.
```

davindisko, 2021-06-10:
Same behaviour... 
It looks like Allo Sparky is no longer supported...
[https://github.com/MichaIng/DietPi/issues/4080](https://github.com/MichaIng/DietPi/issues/4080)

Joulinar, 2021-06-10:
probably a stupid question. You have sufficient disk space?

davindisko, 2021-06-10:
It's a 16Go eMMC card and 13Go are available on /

MichaIng, 2021-06-10:
Can you run this command and paste the output here?
```
bash -c "$(curl -sSfL https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh)"
```
Generally Docker is compatible with Linux 3.10, although it is he minimum version indeed, as long as features have not been backported.

When I find some time, I'll try to rebuild the Sparky SBC kernel from scratch, probably this helps.

davindisko, 2021-06-11:
```
root@DietPi:~# bash -c "$(curl -sSfL https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh)"
warning: /proc/config.gz does not exist, searching other paths for kernel config ...
error: cannot find kernel config
  try running this script again, specifying the kernel config:
    CONFIG=/path/to/kernel/.config bash or bash /path/to/kernel/.config
```

MichaIng, 2021-06-11:
Ah, like on RPi, the module is not loaded automatically. Please try:
```
modprobe configs
bash -c "$(curl -sSfL https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh)"
```

davindisko, 2021-06-11:
```
root@DietPi:~# modprobe configs
modprobe: FATAL: Module configs not found in directory /lib/modules/3.10.38
```
And same result than previously after check-config.sh

MichaIng, 2021-06-12:
[config-3.10.38.txt](https://github.com/MichaIng/DietPi/files/6641983/config-3.10.38.txt)

Another attempt:
```
curl -sSfL (https://github.com/MichaIng/DietPi/files/6641983/config-3.10.38.txt -o /boot/config-3.10.38
bash -c "$(curl -sSfL https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh)"
```

davindisko, 2021-06-12:
```
root@DietPi:~# curl -sSfL https://github.com/MichaIng/DietPi/files/6641983/config-3.10.38.txt -o /boot/config-3.10.38
root@DietPi:~# bash -c "$(curl -sSfL https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh)"
warning: /proc/config.gz does not exist, searching other paths for kernel config ...
info: reading kernel config from /boot/config-3.10.38 ...

Generally Necessary:
- cgroup hierarchy: properly mounted [/sys/fs/cgroup]
- CONFIG_NAMESPACES: enabled
- CONFIG_NET_NS: enabled
- CONFIG_PID_NS: enabled
- CONFIG_IPC_NS: enabled
- CONFIG_UTS_NS: enabled
- CONFIG_CGROUPS: enabled
- CONFIG_CGROUP_CPUACCT: enabled
- CONFIG_CGROUP_DEVICE: missing
- CONFIG_CGROUP_FREEZER: enabled
- CONFIG_CGROUP_SCHED: enabled
- CONFIG_CPUSETS: missing
- CONFIG_MEMCG: enabled
- CONFIG_KEYS: enabled
- CONFIG_VETH: enabled (as module)
- CONFIG_BRIDGE: missing
- CONFIG_BRIDGE_NETFILTER: missing
- CONFIG_IP_NF_FILTER: enabled
- CONFIG_IP_NF_TARGET_MASQUERADE: enabled
- CONFIG_NETFILTER_XT_MATCH_ADDRTYPE: enabled (as module)
- CONFIG_NETFILTER_XT_MATCH_CONNTRACK: enabled
- CONFIG_NETFILTER_XT_MATCH_IPVS: missing
- CONFIG_NETFILTER_XT_MARK: enabled
- CONFIG_IP_NF_NAT: missing
- CONFIG_NF_NAT: enabled
- CONFIG_POSIX_MQUEUE: enabled
- CONFIG_DEVPTS_MULTIPLE_INSTANCES: missing
- CONFIG_NF_NAT_IPV4: enabled
- CONFIG_NF_NAT_NEEDED: enabled

Optional Features:
- CONFIG_USER_NS: missing
- CONFIG_SECCOMP: enabled
- CONFIG_SECCOMP_FILTER: enabled
- CONFIG_CGROUP_PIDS: missing
- CONFIG_MEMCG_SWAP: missing
- CONFIG_MEMCG_SWAP_ENABLED: missing
- CONFIG_MEMCG_KMEM: missing
- CONFIG_RESOURCE_COUNTERS: enabled
- CONFIG_IOSCHED_CFQ: enabled
- CONFIG_CFQ_GROUP_IOSCHED: missing
- CONFIG_BLK_CGROUP: missing
- CONFIG_BLK_DEV_THROTTLING: missing
- CONFIG_CGROUP_PERF: missing
- CONFIG_CGROUP_HUGETLB: missing
- CONFIG_NET_CLS_CGROUP: missing
- CONFIG_NETPRIO_CGROUP: missing
- CONFIG_CFS_BANDWIDTH: missing
- CONFIG_FAIR_GROUP_SCHED: enabled
- CONFIG_RT_GROUP_SCHED: enabled
- CONFIG_IP_NF_TARGET_REDIRECT: enabled
- CONFIG_IP_VS: enabled (as module)
- CONFIG_IP_VS_NFCT: enabled
- CONFIG_IP_VS_PROTO_TCP: enabled
- CONFIG_IP_VS_PROTO_UDP: enabled
- CONFIG_IP_VS_RR: enabled (as module)
- CONFIG_SECURITY_SELINUX: enabled
- CONFIG_SECURITY_APPARMOR: enabled
- CONFIG_EXT3_FS: enabled
- CONFIG_EXT3_FS_XATTR: enabled
- CONFIG_EXT3_FS_POSIX_ACL: missing
- CONFIG_EXT3_FS_SECURITY: missing
    (enable these ext3 configs if you are using ext3 as backing filesystem)
- CONFIG_EXT4_FS: enabled
- CONFIG_EXT4_FS_POSIX_ACL: missing
- CONFIG_EXT4_FS_SECURITY: enabled
    enable these ext4 configs if you are using ext4 as backing filesystem
- Network Drivers:
  - "overlay":
    - CONFIG_VXLAN: missing
    - CONFIG_BRIDGE_VLAN_FILTERING: missing
      Optional (for encrypted networks):
      - CONFIG_CRYPTO: enabled
      - CONFIG_CRYPTO_AEAD: enabled
      - CONFIG_CRYPTO_GCM: missing
      - CONFIG_CRYPTO_SEQIV: missing
      - CONFIG_CRYPTO_GHASH: missing
      - CONFIG_XFRM: enabled
      - CONFIG_XFRM_USER: enabled
      - CONFIG_XFRM_ALGO: enabled
      - CONFIG_INET_ESP: enabled
      - CONFIG_INET_XFRM_MODE_TRANSPORT: enabled
  - "ipvlan":
    - CONFIG_IPVLAN: missing
  - "macvlan":
    - CONFIG_MACVLAN: enabled (as module)
    - CONFIG_DUMMY: missing
  - "ftp,tftp client in container":
    - CONFIG_NF_NAT_FTP: enabled
    - CONFIG_NF_CONNTRACK_FTP: enabled
    - CONFIG_NF_NAT_TFTP: enabled
    - CONFIG_NF_CONNTRACK_TFTP: enabled
- Storage Drivers:
  - "aufs":
    - CONFIG_AUFS_FS: missing
  - "btrfs":
    - CONFIG_BTRFS_FS: enabled (as module)
    - CONFIG_BTRFS_FS_POSIX_ACL: enabled
  - "devicemapper":
    - CONFIG_BLK_DEV_DM: enabled
    - CONFIG_DM_THIN_PROVISIONING: missing
  - "overlay":
    - CONFIG_OVERLAY_FS: missing
  - "zfs":
    - /dev/zfs: missing
    - zfs command: missing
    - zpool command: missing

Limits:
- /proc/sys/kernel/keys/root_maxkeys: 1000000
```

MichaIng, 2021-06-15:
So the `devicemapper` storage driver is used, but `CONFIG_DM_THIN_PROVISIONING: missing`, which seems to cause that error: https://github.com/moby/moby/issues/6325

But also a full disk can be the reason, did you check that it is sufficient for the containers you want to run? `df`

The only doable alternative is Btrfs, but this requires a dedicated drive, formatted with Btrfs filesystem. If you have your `/mnt/dietpi_userdata` moved to an external drive, it would be a try to re-format it with Btrfs and then switch the Docker storage driver accordingly.

But no guarantee that it works, as a few things from "Generally Necessary" are missing as well, although there are other cases where some are missing and Docker still runs fine.

davindisko, 2021-06-15:
Hi, it doesn't look like disk space is unsufficient

```
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
none            495M  6.6M  489M   2% /run
udev             10M     0   10M   0% /dev
tmpfs           495M     0  495M   0% /dev/shm
/dev/mmcblk0p2   15G  1.8G   13G  13% /
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           495M     0  495M   0% /sys/fs/cgroup
tmpfs            50M   32K   50M   1% /var/log
tmpfs           1.0G     0  1.0G   0% /tmp
/dev/mmcblk0p1   48M   13M   36M  26% /boot
```

MichaIng, 2021-06-15:
That is very sufficient. So a dedicated Btrfs-formatted drive is then what can be tested.

I was checking for mainline Linux support for Sparky SBC and found this: https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.16-New-ARM-Hardware
> Along with this new SoC, the CubieBoard7 board is now supported and the Allo.com Sparky SBC.

Interesting. Would be interesting to test and give that board a new life. But I guess the bootloader won't work with the mainline kernel 🤔.

It's the Actions Semiconductor S500 SoC. Armbian had two SBCs with it supported, the Roseapple Pi and Lemaker Guitar. Both have been removed from support and the archived last images are with Linux 3.10.105, hence a new RC version of the Sparky SBC kernel. That all looks like mainline kernel support is not really there.

I'll also check back with the Allo guys, if there is any chance to get a newer Linux major version, but I think kernel development time for this SBC is very limited.

davindisko, 2021-06-15:
Ok, thank you so much for all the research.
But yes, it looks like we're close to the end of support for Sparky SBC... Even if it's only 4 years old :/

MichaIng, 2021-09-18:
Even that it was on Odroid XU4 and Bullseye and a different issues, probably it is related indirectly: #4705

MichaIng, 2022-02-02:
@davindisko 
If you find time, just to rule it out, could you try to add `systemd.unified_cgroup_hierarchy=0` to the end of the cmdline in `/boot/uenv.txt`, reboot and see whether Docker starts then?

Also an upgrade to Debian Bullseye could be tested: https://dietpi.com/blog/?p=811

But it indeed looks like the older kernel is simply not supported by Docker anymore 😞.