MichaIng/DietPi #4481: error install rpi-eeprom

timoteicampian, 2021-06-11:
#### Details:
- Date           | Fri Jun 11 11:12:26 EEST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install rpi-eeprom
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. dietpi image copied to sdcard
2. /boot/dietpi.txt edited
3. first boot
4. error found in /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/e/expat/libexpat1_2.2.6-2+deb10u1_armhf.deb  Could not connect to raspbian.raspberrypi.org:80 (93.93.128.193), connection timed out [IP: 93.93.128
.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7-minimal_3.7.3-2+deb10u3_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3-defaults/python3-minimal_3.7.3-1_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mime-support/mime-support_3.62_all.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mpdecimal/libmpdec2_2.4.2-2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/libpython3.7-stdlib_3.7.3-2+deb10u3_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7_3.7.3-2+deb10u3_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3-defaults/libpython3-stdlib_3.7.3-1_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3-defaults/python3_3.7.3-1_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/pciutils/libpci3_3.5.2-1_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/pciutils/pciutils_3.5.2-1_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils-common_2.31.1-16+rpi2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/libbinutils_2.31.1-16+rpi2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/f/flex/libfl2_2.6.4-6.2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils-arm-linux-gnueabihf_2.31.1-16+rpi2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Joulinar, 2021-06-11:
Hi,

looks like you have issues to connect to global Raspbian `apt` repository. Can you try following

```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb
```

timoteicampian, 2021-06-11:
root@DietPi:~# cd /tmp
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb
--2021-06-11 12:35:54--  http://raspbian.raspberrypi.org/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirrors.pidginhost.com/raspbian/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb [following]
--2021-06-11 12:35:55--  http://mirrors.pidginhost.com/raspbian/raspbian/pool/main/b/binutils/binutils_2.31.1-16+rpi2_armhf.deb
Resolving mirrors.pidginhost.com (mirrors.pidginhost.com)... 176.124.104.176, 2001:67c:744:7::2
Connecting to mirrors.pidginhost.com (mirrors.pidginhost.com)|176.124.104.176|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 57028 (56K) [application/octet-stream]
Saving to: 'binutils_2.31.1-16+rpi2_armhf.deb'

binutils_2.31.1-16+rpi2_armhf.deb                     100%[=========================================================================================================================>]  55.69K  --.-KB/s    in 0.06s

2021-06-11 12:35:55 (930 KB/s) - 'binutils_2.31.1-16+rpi2_armhf.deb' saved [57028/57028]

timoteicampian, 2021-06-11:
apt-get --allow-change-held-packages install rpi-eeprom
Reading package lists... Done
Building dependency tree
Reading state information... Done
rpi-eeprom is already the newest version (12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:/tmp# apt-get -qq --allow-change-held-packages install rpi-eeprom
root@DietPi:/tmp# echo $?

timoteicampian, 2021-06-11:
besides ethernet, wifi was also enabled. could this have been the culprit?

Joulinar, 2021-06-11:
are you connected with Ethernet as well as WiFi to same network? That's usually not a good idea

To check if all packages updated, just run `apt update && apt upgrade`

timoteicampian, 2021-06-11:
yes, I am using ethernet also, but only wifi seems to be active

root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether dc:a6:32:88:a6:a4 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:88:a6:a5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.68.200/24 brd 192.168.68.255 scope global wlan0
       valid_lft forever preferred_lft forever
       
       
       
apt update && apt upgrade works well

Joulinar, 2021-06-11:
> apt update && apt upgrade works well

maybe a temp issue you faced at the beginning.

MichaIng, 2021-06-11:
This can happen during updates at the APT repository itself, when some files are changed already but others not yet. It *should* be atomic, but experience say that it is not always 😉.

Joulinar, 2021-06-11:
I will close this issue as system is working now. Feel free to reopen if needed

closed 2021-06-11 by Joulinar