MichaIng/DietPi #4482: MOTD - odd message?

timoteicampian, 2021-06-11:
─────────────────────────────────────────────────────
 DietPi v7.2.3 : 12:34 - Fri 06/11/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 53'C : 127'F (Running warm, but safe)
 - LAN IP : 192.168.68.200 (wlan0)
 **_- MOTD : DietPi v7.2 has been released, please update now._**
 ─────────────────────────────────────────────────────

as you can see, I already have the latest version, although MOTD oddly suggests to update

Joulinar, 2021-06-11:
the MOTD is not something depending on the version of your system. It's a massage that is maintained centrally and will be displayed all user if MOTD is active. It's changed in regular basis, 1 or 2 times a month.

MichaIng, 2021-06-11:
Yes, it is not smart in most cases. But you're right, v7.2 has been released long enough to have it changed 👍.

Joulinar, 2021-06-11:
going to close this. Feel free to reopen if needed.

closed 2021-06-11 by Joulinar