MichaIng/DietPi #4483: soundcard selected in dietpi.txt. incomplete install

timoteicampian, 2021-06-11:
### Creating a bug report/issue

#### Details:
- Date           | Fri Jun 11 11:12:26 EEST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install rpi-eeprom
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | 9b22de83-0078-443b-b589-2b307139b601

#### Steps to reproduce

1. download latest dietpi image 
2. modified dietpi.txt soundcard options: 
# Sound card
CONFIG_SOUNDCARD=rpi-bcm2835-auto

#### Expected behaviour

- alsa-util should be installed too

#### Actual behaviour

- alsa-utils was not installed
- 
#### Extra details

- fixed manually by enabling soundcard using dietpi-config

timoteicampian, 2021-06-11:
I also configured the hostname using dietpi.txt but at the login screen the hostname remained DietPi
I also tried the following

```
root@DietPi:~# /boot/dietpi/func/change_hostname lv01
root@DietPi:~# hostname
DietPi
```

```
root@DietPi:~# hostname lv01
root@DietPi:~# hostname
lv01
```

```
root@DietPi:~#  exec bash
root@lv01:~#
```

Joulinar, 2021-06-11:
you would need to reboot to get the new host name.

MichaIng, 2021-06-11:
But actually we could call the `hostname` command in our script as well. Not sure whether it is respected that way by all parts of the system, so that a reboot is indeed not required (but only a refresh, login/logout of the current shell session). At least it does not hurt.

MichaIng, 2021-07-26:
Done with DietPi v7.4: https://github.com/MichaIng/DietPi/commit/b5d86e4905359cde933e2d48412c854417a15752

closed 2021-07-26 by MichaIng