MichaIng/DietPi #4484: Boot process slow on NanoPi NEO Core becaause of EHCI timed out on TD - token=0xd8d80

orotarobas, 2021-06-11:
**Dietpi Version**:
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
**Distro Version**: bustercat 10.9
**Kernel Version**: Linux DietPi 5.10.34-sunxi #21.05.1 SMP Thu May 6 20:13:21 UTC 2021 armv7l GNU/Linux
**SBC Version**: NanoPi NEO (armv7l)
**Power Supply**: Benchtop power supply, 5V 3A
**SD Card**: board emmc 

Even if the installation properly succeed the boot process is always very slow since it hangs for several seconds at USB initalization because of EHCI timeout issue, here follows the boot log:

U-Boot SPL 2020.10-armbian (Dec 11 2020 - 23:42:28 +0100)
DRAM: 512 MiB
Trying to boot from MMC2

U-Boot 2020.10-armbian (Dec 11 2020 - 23:42:28 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi NEO
DRAM:  512 MiB
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial
Out:   serial
Err:   serial
Net:   phy interface0

Error: ethernet@1c30000 address not set.
No ethernet found.

starting USB...
Bus usb@1c1a000: USB EHCI 1.00
Bus usb@1c1a400: USB OHCI 1.0
Bus usb@1c1d000: USB EHCI 1.00
Bus usb@1c1d400: USB OHCI 1.0
scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
scanning bus usb@1c1d000 for devices... EHCI timed out on TD - token=0xd8d80
Device NOT ready
   Request Sense returned 00 00 00
2 USB Device(s) found
scanning bus usb@1c1d400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
3967 bytes read in 2 ms (1.9 MiB/s)
## Executing script at 43100000
U-boot loaded from eMMC or secondary SD
MMC: no card present
Boot script loaded from mmc
275 bytes read in 2 ms (133.8 KiB/s)
8969831 bytes read in 680 ms (12.6 MiB/s)
7973464 bytes read in 605 ms (12.6 MiB/s)
Found mainline kernel configuration
32430 bytes read in 6 ms (5.2 MiB/s)
504 bytes read in 4 ms (123 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost1.dtbo
504 bytes read in 4 ms (123 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost2.dtbo
339 bytes read in 4 ms (82 KiB/s)
Applying kernel provided DT overlay sun8i-h3-analog-codec.dtbo
502 bytes read in 4 ms (122.1 KiB/s)
Applying kernel provided DT overlay sun8i-h3-uart1.dtbo
4185 bytes read in 4 ms (1021.5 KiB/s)
Applying kernel provided DT fixup script (sun8i-h3-fixup.scr)
[... 138 lines cut ...]
[  OK  ] Reached target Network.
         Starting Permit User Sessions...
[  OK  ] Started Permit User Sessions.
[  OK  ] Started Serial Getty on ttyGS0.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.
Starting DietPi-Autostart (Custom) script...

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : cron

 ─────────────────────────────────────────────────────
 DietPi v7.2.3 : 15:08 - Fri 06/11/21
 ─────────────────────────────────────────────────────
 - LAN IP : fe80::3c6d:ccff:fe6f:58e0 (eth0)

 Please hit <return> to login

orotarobas, 2021-06-11:
> Many thanks for your report. I wonder whether this is as the NEO Core does
> not have a USB port like NEO (non-Core). On the other hand, the USB hosts
> are present, available via pitch pin header.
>
> Let me create a new image, just in case it got resolved in the meantime.
> Else we'll need to report it to Armbian.

Allright, let's Hope to be lucky.
Thanks for your support

orotarobas, 2021-06-29:
Hello, any news about this issue??

MichaIng, 2021-06-30:
New image uploaded: https://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Buster.7z

MichaIng, 2021-09-01:
Did the new image help?

orotarobas, 2021-09-07:
As far as i remember no, i will check again this and the issue about 
self update of old image and will follow up about both by end of week.

------ Messaggio originale ------

becaause of EHCI timed out on TD - token=0xd8d80 (#4484)

MichaIng, 2021-10-20:
A new Debian Bullseye based image is available now. I'll close the issue, but feel free to reopen if problems persist.

orotarobas, 2022-06-20:
Hello

this message is posted here since it's not clear how to get in touch 
directly for this pouropuse.

We have a business opportunity that we would like to develop using 
NanoPi R2S+ DietPi.

The board has to replicate messages that is getting from a usb keyboard 
(NanoPi has to act as usb host on the USB 2.0 port) to a PC (NanoPi has 
to emulate a USB device on the USB C port) and a Robot through Ethernet 
connection.

Would you be open to support us into this project? Can you make us a 
commercial proposal so we can add your work to implement such behaviour 
in our offer to the customer?

We think we miss the USB Device driver configuration on the USB C port.

Please let us know your toughts

Bests

Vesentini Lorenzo

Logos01 Srl

------ Messaggio originale ------
Da "MichaIng" ***@***.***>
A "MichaIng/DietPi" ***@***.***>
Cc "orotarobas" ***@***.***>; "Author" 
***@***.***>
Data 20/10/2021 14:22:00
Oggetto Re: [MichaIng/DietPi] Boot process slow on NanoPi NEO Core 
becaause of EHCI timed out on TD - token=0xd8d80 (#4484)

MichaIng, 2022-06-25:
@orotarobas 
In any case, please open a new issue, since your request is not related to this 1 year old issue and is even about a different NanoPi model.

I'm not sure whether what you are asking for is something we are able to achieve without a new USB driver, on NanoPi side and/or PC side, also depending on whether NanoPi or PC is the USB host, since I'm not aware of a method to enable gadget mode on NanoPi. We may look into it when we find time, but as of limited spare development time and as we do no kernel development, I cannot make promises.

An alternative would be to discuss the question at Armbian, which develops the kernel we're using for NanoPi R2S: https://forum.armbian.com/forum/156-nanopi-r2s/
Feel free to ping me there as well via `@MichaIng`.

orotarobas, 2022-06-27:
Allright, well noted.

I will open a discussion topic on Armbian.

Bests

Lorenzo

------ Messaggio originale ------
Da "MichaIng" ***@***.***>
A "MichaIng/DietPi" ***@***.***>
Cc "orotarobas" ***@***.***>; "Mention" 
***@***.***>
Data 25/06/2022 18:19:22
Oggetto Re: [MichaIng/DietPi] Boot process slow on NanoPi NEO Core 
becaause of EHCI timed out on TD - token=0xd8d80 (#4484)

***@***.*** <https://github.com/orotarobas>
***@***.***>

MichaIng, 2022-06-27:
Ah, it seems you can use the mini USB port in gadget mode OOTB, when using PoE for powering the R2S instead. Then probably a serial console is a way to pass input to the PC connected via serial console?
```sh
sudo modprobe g_serial
sudo systemctl restart serial-getty@ttyGS0
echo test > /dev/ttyGS0
```