MichaIng/DietPi #4485: Pi hole installation issue

NgTrThanh, 2021-06-12:
#### Details:
- Date           | Sat Jun 12 10:43:27 +07 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.10.17+ #1421 Thu May 27 13:58:02 BST 2021 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ping -c 1 -W 10 dns9.quad9.net
- Exit code      | 2
- Software title | DietPi-Login
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: dns9.quad9.net: Temporary failure in name resolution

Joulinar, 2021-06-12:
Hi,

You are failing on a very basic internet connection check. Can you post following

```
ping -c 1 -W 10 dns9.quad9.net
cat /etc/resolv.conf
readlink /etc/resolv.conf
```

MichaIng, 2021-06-27:
I mark this issue as closed. Feel free to reopen if required.

closed 2021-06-27 by MichaIng