MichaIng/DietPi #4486: No render in Chromium

TamaTamaGoGo, 2021-06-12:
### No render in Chromium

Sorry for my week English.

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | 
10.9
- Kernel version | 
Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- SBC model | 
RPi 4 Model B (aarch64)
- Power supply used | 
5V 
- SDcard used | 
I-O Data MicroSD 32GB

#### Additional Information (if applicable)
- Software title | 
Chromium
- Was the software title installed freshly or updated/migrated?
APT(dietpi-software)
- Can this issue be replicated on a fresh installation of DietPi?
I think **YES**.
- Bug report ID | 
256d082f-b7bd-4723-b653-019799e0b191

#### Steps to reproduce

1. Install chromium at first boot
2. Install any desktop and run chromium

#### Expected behaviour

- Website is not render.

#### Actual behaviour

- Website is not render.

#### Extra details

- Running `chromium` at Xfce Terminal,

```root@DietPi:~# chromium
[5230:5230:0612/133623.413425:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133623.414634:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133623.425941:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.432017:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.434040:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.435958:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.626468:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133623.631106:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133623.727616:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.730170:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.732080:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.734286:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.736342:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.843640:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133623.846211:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133623.869135:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.872945:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.876989:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.883411:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133623.885379:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133624.579402:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133624.580161:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133624.743826:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133624.747559:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133625.533015:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133625.537694:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133626.381891:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133626.382306:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133626.661530:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133626.662344:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133626.683599:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.685666:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.687614:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.690379:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.692512:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.694620:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.696456:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.698303:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.700123:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.702432:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.704138:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.706714:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.709092:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.710939:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[5230:5230:0612/133626.712762:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.[5230:5230:0612/133628.436749:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133628.437607:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133628.548962:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133628.549437:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133628.672926:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133628.679288:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133628.788257:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133628.788700:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[5230:5230:0612/133628.949963:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[5230:5230:0612/133628.950945:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
```

Joulinar, 2021-06-12:
Hi,

Found something similar. But huge difference, you are running 64bit while this is on 32bit I guess https://github.com/RPi-Distro/chromium-browser/issues/16

What is the user you are using to run the desktop?

TamaTamaGoGo, 2021-06-12:
Using **Cinnamon** installed by apt.
But that’s occurred by **MATE**, **LXQt**, **Xfce** and **KDE**.
So, I think we can found that issue by any desktops.

(FireFox works fine!)

TamaTamaGoGo, 2021-06-12:
### Additional Information
Using **root(sudo) privileges**.
Thank you.

Joulinar, 2021-06-12:
How does it behaves if you use a plain root desktop (without sudo)

TamaTamaGoGo, 2021-06-12:
OK. I’ll try it.

TamaTamaGoGo, 2021-06-13:
[image]

There are same issue at other privileges.

Log is here.
```
[2856:2856:0613/122053.133669:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.134912:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.136560:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.137976:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.139067:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.140318:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.141385:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.142586:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.143163:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.165160:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.165545:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.215002:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.218173:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.224057:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.226089:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.227351:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.228637:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.229817:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.230870:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.231915:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.232918:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.233883:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.235019:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.236197:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.240352:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.241974:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.298370:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.299215:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.337855:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.338387:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.364403:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.364918:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.398599:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122053.398959:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
[2856:2856:0613/122053.409279:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122053.410428:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[... cut ...]
[2856:2856:0613/122055.819879:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122055.820658:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[2856:2856:0613/122055.873736:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
[2856:2856:0613/122055.875052:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
server@DietPi:~ $ 
```

TamaTamaGoGo, 2021-06-13:
Sorry, I forgot to use debug option.
I’ll try with command
`chromium --enable-logging --v=1`.
Thank you.

TamaTamaGoGo, 2021-06-13:
[image]

Log is here.
https://raw.githubusercontent.com/TamaTamaGoGo/tamatamagogo/master/logs/DietPi/%234486/chrome_debug.log

Joulinar, 2021-06-13:
I guess you are still not running it as `root` user. correct?

TamaTamaGoGo, 2021-06-13:
There are same behaviors at `root` user.
But, I’ll test again with `root` user with command `chromium —enable-logging —v=1`.

TamaTamaGoGo, 2021-06-13:
### Additional Information
`chromium --version`
`Chromium 89.0.4389.114 built on Debian 10.9, running on Debian 10.9`

**Current resolution**
`vc4-kms-v3d-pi4`

TamaTamaGoGo, 2021-06-13:
So, I found that **89.0.4389.114** version is the bad version.

When I run command `apt list chromium -a`
```
chromium/stable,now **89.0.4389.114-1~deb10u1** arm64 [upgradable from: 88.0.4324.182-1~deb10u1]
chromium/stable **88.0.4324.182-1~deb10u1** arm64 [installed,upgradable to: 89.0.4389.114-1~deb10u1]
```
So, I found there are the version what hide but stable version.
I delete 89.0.4389.114-1~deb10u1 with `dietpi-software`.

```
apt install -y chromium-common=88.0.4324.182-1~deb10u1
apt install -y chromium=88.0.4324.182-1~deb10u1
```
After installation, `chromium --no-sandbox`(because running by `root`(privilege)), that's worked finw.
[image]

That's works fine.
That mean's **chromium=88.0.4324.182-1~deb10u1** is the latest version what works fine for DietPi.
Sorry for my week English and long posts.

Joulinar, 2021-06-13:
but this is an issue of the official `chromium` version 89. DietPi is using the global package, available via global Debian Repository. There is nothing DietPi specific on this. Furthermore, DietPi is not an own OS, it's a diet version of Raspberry OS 64bit in your case.

TamaTamaGoGo, 2021-06-14:
Now it working fine.
It might be caused by my SD card's life or flash error (or unwanted error at installing).

`chromium --version`
**Chromium 89.0.4389.114 built on Debian 10.9, running on Debian 10.9**

Sorry for my issue.
I'll close this issue by my own decision.
Thanks again.

closed 2021-06-14 by TamaTamaGoGo

MichaIng, 2021-06-14:
Great that it has been solved. Chromium releases on Debian are very frequent, so it's not unthinkable that there is one that causes issues on certain boards, architecture, GPU drivers. Another release is available in the meantime, but for some reason not yet for arm64: https://packages.debian.org/buster/chromium