MichaIng/DietPi #4487: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions

bartplessers, 2021-06-13:
#### Details:
- Date           | Sat Jun 12 10:25:50 CEST 2021
- Bug report     | 9056d18b-7cba-4d87-8b68-af25f4fff6da
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux rpi-84 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:6 https://archive.raspberrypi.org/debian buster/main armhf Packages [376 kB]
Error: No information about packages! (Maybe no deb entries?)
Fetched 13.6 MB in 19s (733 kB/s)
Reading package lists...
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code
```

Joulinar, 2021-06-13:
Hi,

Can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update

apt upgrade
```

MichaIng, 2021-06-13:
It is a [known bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=617856) with `apt-show-versions`. Do you require it for a specific reason, or did you have Webmin installed? In the past, Webmin had `apt-show-versions` as dependency, but that is not the case anymore.

Generally I had a closer look into it, and nowadays you can do everything that `apt-show-versions` does with plain `apt` as well, much fast and with less overhead. `apt-show-versions` creates and updates an own APT cache and requires the original APT package lists in plain text, while on DietPi they are compressed now by default. This can be changed, if you required it, but otherwise I suggest to `apt purge apt-show-versions`.

Joulinar, 2021-07-25:
Going to close this issue. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar