MichaIng/DietPi #4488: PI-SPC install issue: pi-spc/sds.sh lin 18 unexpected operator

renaudlarzilliere, 2021-06-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install dietpi from latest image
2. install pi-spc, shaireport and raspotify
3. updates run and when reboot, then repeated error message:
4

#### Expected behaviour

- reboot normally

#### Actual behaviour

- repeated error message: /var/lib/dietpi/dietpi-software/installed/pi-spc/sds.sh : 18: [: 0: unexpected operatoro 

#### Extra details

- ...

Joulinar, 2021-06-13:
Hi,

pls can you post content of `sds.sh` file

`cat /var/lib/dietpi/dietpi-software/installed/pi-spc/sds.sh`

renaudlarzilliere, 2021-06-13:
[image]

MichaIng, 2021-06-13:
Dammit, there is a syntax bug in the script. Please try to resolve with this command:
```
sed -i 's/==/=/' /var/lib/dietpi/dietpi-software/installed/pi-spc/sds.sh
```

MichaIng, 2021-06-13:
Fixed with: https://github.com/MichaIng/DietPi/commit/96a78c445107efd328af71689d8b8b21fc8bc620
Changelog: https://github.com/MichaIng/DietPi/commit/33f572fed1b77f7cca3f842ddcd4f92fe0e2708f

renaudlarzilliere, 2021-06-13:
> Dammit, there is a syntax bug in the script. Please try to resolve with this command:
> 
> ```
> sed -i 's/==/=/' /var/lib/dietpi/dietpi-software/installed/pi-spc/sds.sh
> ```
Worked ! Thanks

closed 2021-06-26 by MichaIng