MichaIng/DietPi #4490: RPi 3 B Freezing Up

DarkVader-cell, 2021-06-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.2.3`
- Distro version | `Debian Buster`
- Kernel version | `Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux`
- SBC model | `RPi 3 Model B (aarch64)`
- Power supply used | `5V 2A`
- SDcard used | `SanDisk Ultra Dual 64GB USB`

#### Additional Information (if applicable)
- Software title | 
1. Docker (Multiple containers)
2. Docker- Compose
3. Portainer
4. Jellyfin
5. Radarr
6. Sonarr
7. Bazarr
8. Webmin
9. vsftpd

- Was the software title installed freshly or updated/migrated? `Nothing New`

- Bug report ID | `45b728e2-db1d-455f-a182-5659b996f9c5`

#### Expected behaviour

The Pi handled all these smoothly as I set a swapfile of 8GB, and the governor was set to Performance mode. I also have active cooling and temps stay well below 55 °C.  Also, the Pi boots from a 64 GB SanDisk Ultra 3.0 USB.

#### Actual behaviour

The Pi works well, but I tried rebooting using the `reboot` command, and it just freezes up after reboot, i.e, I can ping it, but no services like SSH/FTP are running. I then require multiple hard reboots and power cycling to get it into a working state. So I was wondering if this is a hardware limitation or a software one? Also, another interesting fact is that, after getting it into a working state, docker suddenly stopped working on boot( had to reinstall manually).

#### Extra details

I have attached the output of `dmesg` command here
[DIETPI logs.txt](https://github.com/MichaIng/DietPi/files/6653396/DIETPI.logs.txt)

Here's the Docker log in case it's helpful,
[Docker log.txt](https://github.com/MichaIng/DietPi/files/6653414/Docker.log.txt)
 Also, this is very similar to #2133, but my doubt is it was running without any issues prior to this and rebooted well with all these installed. There were no power outages, and I tried different USB cables :( 

 ...

DarkVader-cell, 2021-06-15:
Just observed this using `nmap` after another reboot, a lot of ports show up as filtered, mainly `ssh`, `http`, `https` but ports of `Radarr`, `Sonarr` all seem to be opening fine.

Joulinar, 2021-06-15:
quite some applications installed. Did you had a look how memory and swap are behaving over time? Maybe they are going to be filled up, causing the system to slow down and probably freeze at the end.

MichaIng, 2021-06-15:
To get logs from before the freeze, you could uninstall RAMlog and enable persistent system logging:
```
dietpi-software uninstall 103
mkdir -p /var/log/journal
```

DarkVader-cell, 2021-06-15:
> quite some applications installed. Did you have a look at how memory and swap are behaving over time? Maybe they are going to be filled up, causing the system to slow down and probably freeze at the end.
Oh, at boot all cores are at 90% usage, this drops down to 20-30% after a min or two. Recently I split my memory, and gave 320MB for the GPU, leaving close to 660MB for another usage. It seems to hit 530MB/667MB and another 100MB or so spills over to the SWAP. Shouldn't the existence of a swapfile avoid this tho?

DarkVader-cell, 2021-06-15:
@MichaIng Ok, I have done the above steps, now I need to present the output of `dmesg` next time it freezes? (i.e after a reboot) or should I upload logs from a certain folder?

MichaIng, 2021-06-15:
The output of `journalctl` will then show system logs, including the ones prior to crash.

DarkVader-cell, 2021-06-15:
@MichaIng Ok, I have managed to retrieve some logs, there seem to be a few (from each reboot/freeze I presume), or are they split into files of 8MB each? Anyways I'll attach them below.

[system.journal LOGS.zip](https://github.com/MichaIng/DietPi/files/6656759/system.journal.LOGS.zip)

Joulinar, 2021-06-15:
you could use `free -m` to watch SWAP usage from time to time. In general using SWAP is one of the worst thinks you can do as it's slowing down your systems performance. Because system is writing down part of the phys memory to disk, which is much slower than phys memory. Probably SWAP usage will increate over time, as your phys memory is already exhausted after reboot.

DarkVader-cell, 2021-06-15:
@Joulinar oh...okay, will keep that in mind next time, oh and were the logs useful? Was there something, in particular, triggering it off?

Joulinar, 2021-07-05:
@DarkVader-cell how is it going?

MichaIng, 2021-07-05:
I checked the logs. They are rotated on each boot, yes

To remove an obsolete cmdline entry:
```sh
sed -i 's/elevator=deadline //' /boot/cmdline.txt
```

But what likely causes your issue is:
```
kernel: Under-voltage detected! (0x00050005)
...
kernel: rpi_firmware_get_throttled: 9 callbacks suppressed
```

Is the USB drive powered via dedicated PSU or via USB only? The letter may be related to the voltage issues. Keep an eye on `dmesg -l emerg,alert,crit,err` for further such voltage and other probably related errors.

Joulinar, 2021-07-25:
Going to close the issue due to missing feedback. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar