MichaIng/DietPi #4493: hcitool support in nanopi air.

Ataraxiall, 2021-06-16:
Good morning,

I have a NanoPi Air with the latest version of DietPi.
For a program, I need to use the hcitool option, but both when haver scan or dev, I can't find the bluetooth on the board and I don't know how to activate it.

From dietpi-config I have tried putting it on, but it doesn't solve anything. Any ideas?

Thanks!

MichaIng, 2021-06-16:
Many thanks for your report.

Does the following help?
```
hciconfig hci0 up
systemctl restart bluetooth
```

And what does the following show?
```
systemctl status bluetooth
dmesg | grep -E 'hci|blue'
```

Ataraxiall, 2021-06-16:
> Many thanks for your report.
> 
> Does the following help?
> 
> ```
> hciconfig hci0 up
> systemctl restart bluetooth
> ```
> 
> And what does the following show?
> 
> ```
> systemctl status bluetooth
> dmesg | grep -E 'hci|blue'
> ```

Thanks for the fast response.

`root@DietPi:~# hciconfig hci0 up
Can't open HCI socket.: Address family not supported by protocol
`
`root@DietPi:~# systemctl start bluetooth
root@DietPi:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset
: enabled)
   Active: inactive (dead)
     Docs: man:bluetoothd(8)

Jun 16 15:29:40 DietPi systemd[1]: Condition check resulted in Bluetooth service
 being skipped.
`

`root@DietPi:~# dmesg | grep -E 'hci|blue'
[    3.074730] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    3.074766] ehci-platform: EHCI generic platform driver
[    3.075181] ehci-platform 1c1a000.usb: EHCI Host Controller
[    3.075253] ehci-platform 1c1a000.usb: new USB bus registered, assigned bus number 1
[    3.076092] ehci-platform 1c1a000.usb: irq 37, io mem 0x01c1a000
[    3.089896] ehci-platform 1c1a000.usb: USB 2.0 started, EHCI 1.00
[    3.090486] usb usb1: Manufacturer: Linux 5.10.4-sunxi ehci_hcd
[    3.092843] ehci-platform 1c1c000.usb: EHCI Host Controller
[    3.092889] ehci-platform 1c1c000.usb: new USB bus registered, assigned bus number 2
[    3.093282] ehci-platform 1c1c000.usb: irq 39, io mem 0x01c1c000
[    3.105875] ehci-platform 1c1c000.usb: USB 2.0 started, EHCI 1.00
[    3.106428] usb usb2: Manufacturer: Linux 5.10.4-sunxi ehci_hcd
[    3.108737] ehci-platform 1c1d000.usb: EHCI Host Controller
[    3.108779] ehci-platform 1c1d000.usb: new USB bus registered, assigned bus number 3
[    3.109215] ehci-platform 1c1d000.usb: irq 41, io mem 0x01c1d000
[    3.121891] ehci-platform 1c1d000.usb: USB 2.0 started, EHCI 1.00
[    3.122411] usb usb3: Manufacturer: Linux 5.10.4-sunxi ehci_hcd
[    3.124466] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    3.124506] ohci-platform: OHCI generic platform driver
[    3.124911] ohci-platform 1c1a400.usb: Generic Platform OHCI controller
[    3.124957] ohci-platform 1c1a400.usb: new USB bus registered, assigned bus number 4
[    3.125289] ohci-platform 1c1a400.usb: irq 38, io mem 0x01c1a400
[    3.186253] usb usb4: Manufacturer: Linux 5.10.4-sunxi ohci_hcd
[    3.188364] ohci-platform 1c1c400.usb: Generic Platform OHCI controller
[    3.188408] ohci-platform 1c1c400.usb: new USB bus registered, assigned bus number 5
[    3.188728] ohci-platform 1c1c400.usb: irq 40, io mem 0x01c1c400
[    3.250239] usb usb5: Manufacturer: Linux 5.10.4-sunxi ohci_hcd
[    3.252471] ohci-platform 1c1d400.usb: Generic Platform OHCI controller
[    3.252516] ohci-platform 1c1d400.usb: new USB bus registered, assigned bus number 6
[    3.252835] ohci-platform 1c1d400.usb: irq 42, io mem 0x01c1d400
[    3.314242] usb usb6: Manufacturer: Linux 5.10.4-sunxi ohci_hcd
`

dont work.

Thnaks!

MichaIng, 2021-06-16:
It's attached via UART. Do you use a serial console, or only SSH?

Ataraxiall, 2021-06-16:
Im connected via serial console (microusb port) to pc with putty.

MichaIng, 2021-06-16:
Okay, this is the gadget mode console on `ttyGS0`, so the debug console on `ttyS0` should be theoretically free.

Please try `hciattach /dev/ttyS0 bcm43xx 1500000`.
If it does not work, try:
- add `uart1` to the `overlays=` line in `/boot/armbianEnv.txt`
- run `hciattach /dev/ttyS1 bcm43xx 1500000`

Joulinar, 2021-07-25:
Going to close this issue. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar