MichaIng/DietPi #4498: Open Beta v7.3 | Please help testing and hardening the upcoming release

MichaIng, 2021-06-19:
RC version | v7.3.0
---------- | -------
v7.3.0 PR | #4495
Release plan | #4452
Release docs | TBD
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2021-06-26**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.3
____
### Beta v7.3.0
_(2021-06-19)_

#### Changes
- DietPi-Automation | A new dietpi.txt setting "AUTO_SETUP_DHCP_TO_STATIC" has been added. When set to "1", DHCP leased network settings will be applied as static network settings automatically during first run setup. This works as well with older images, by adding the above setting to dietpi.txt.
- DietPi-Backup | The include/exclude filter handling has been reworked. /mnt (dietpi_userdata) and /media related rules are added now via the editable custom filter file, which gives users more control over these. Especially it allows to include other mount points below /mnt, hence external dietpi_userdata, which was previously impossible due to the order in which those filter rules are applied.

#### New Software
- AdGuard Home | This DNS sinkhole ad blocker for your LAN, similar to Pi-hole, has been added with software ID 126.
- Beets | This command line music organizer and manager has been added with software ID 190.
- frp | This proxy system has been added as install option with ID 171.
- Snapcast Server | This multiroom audio server component has been added with software ID 191. Many thanks to @foxy82 for implementing this software title: https://github.com/MichaIng/DietPi/pull/4465
- Snapcast Client | The multiroom audio client component has been added with software ID 192. Many thanks to @foxy82 for implementing this software title: https://github.com/MichaIng/DietPi/pull/4465
- K3s | Added a lightweight Kubernetes implementation as install option with ID 193. Many thanks to @mortenlj for implementing this software title: https://github.com/MichaIng/DietPi/pull/4476

#### Fixes
- DietPi-JustBoom | Resolved an issue where the equalizer was always shown as "Off" even when it was just or previously enabled (v7.2 regression). Many thanks to `@shao` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=35072#p35072
- DietPi-Drive_Manager | Resolved an issue where network drives were detected as physical drives (v7.2 regression). Many thanks to @maartenlangeveld for reporting this issue: https://github.com/MichaIng/DietPi/issues/4479
- DietPi-Software | Node-RED: Resolved an issue where the Python 3 RPi.GPIO module was tried to be installed as dependency on non-RPi devices (v7.2 regression). Many thanks to @TheAdminFrmoHell for reporting this issue: https://github.com/MichaIng/DietPi/issues/4478
- DietPi-Software | PI-SPC: Resolved a syntax error in the shutdown script loop. Many thanks to @renaudlarzilliere for reporting this issue: https://github.com/MichaIng/DietPi/issues/4488

Joulinar, 2021-06-22:
Hi,

probably found an issue on `frp`. During install I'm ask to select `Server`, `Client` or `Both`. And now I try to `cancel` the dialog, which is not working. I'm locked into a loop as I need to do a selection

`[ERROR] A choice is required to finish the frp install.`

MichaIng, 2021-06-23:
Jep, that is expected as a choice is required to know which service to install and setup, which is also the reason this install option has the aSOFTWARE_INTERACTIVE flag. Or shall we allow to cancel and then do a default choice? Which default would be best?
A `dietpi.txt` option would be actually nice as well. But sooner or later we need to split those out into a dedicated software config file, to not overload `dietpi.txt` too much 😄.

Joulinar, 2021-06-23:
If I cancel at this stage, I would expect to cancel the installation. What would be the meaning of a cancellation button, if I can't use it? 😉

MichaIng, 2021-06-23:
whiptail actually has the `--nocancel` option to hide the cancel button, I just never implemented it into G_WHIP functions, as the cases are very rare. But indeed it makes sense here.

An option to really cancel the install doesn't make sense, does it? I mean who starts to install a software title to then abort it when being asked for a setup choice?

Joulinar, 2021-06-23:
True, might be an extremely rar case.

closed 2021-06-26 by MichaIng

MichaIng, 2021-06-26:
DietPi v7.3 has been released: https://dietpi.com/docs/releases/#june-2021-version-73

Many thanks to all testers 👍.