MichaIng/DietPi #4499: USB hard disk device name changes soon after boot (but not immediately)

nave7693, 2021-06-21:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | official RPi power adapter (5V 3A)
- SDcard used | 32GB Sandisk

#### Additional Information (if applicable)
- Software title | dietpi-driver_manager (presumed)
- Was the software title installed freshly or updated/migrated? migrated (dietpi upgrade from 7.1.2 to 7.2.3)
- Can this issue be replicated on a fresh installation of DietPi? unsure
- Bug report ID | N/A

#### Steps to reproduce
1. Connect two USB hard drives to the pi
2. Boot and observe

#### Expected behaviour
- the two hard drives will appear as /dev/sda and /dev/sdb

#### Actual behaviour
- the second hard drive switches from being /dev/sda to /dev/sdc within 10 minutes after boot

#### Extra details
- I mounted /dev/sda2 as /mnt/media and contains the dietpi user data directory. This behavior was observed after first reboot after upgrading to 7.2 so I was thinking maybe it has to do with the changes to automount for dietpi-drive_manager. I don't know of an easy way to revert back to 7.1 (I overwrote my backup after mistakenly thinking that my containers which live in /dev/sda2 were corrupted, but in fact the entire mount point is missing due to the drive name changing from sda to sdc)

Please let me know if there are any troubleshooting steps.

```
dietpi@DietPi:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  6.1G   22G  23% /
devtmpfs        926M     0  926M   0% /dev
tmpfs           959M   24K  959M   1% /dev/shm
tmpfs           959M   67M  892M   7% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           959M     0  959M   0% /sys/fs/cgroup
tmpfs           1.5G  148M  1.3G  11% /tmp
tmpfs            50M  9.1M   41M  19% /var/log
/dev/mmcblk0p1  253M   50M  203M  20% /boot
/dev/sda2       3.7T  2.0T  1.7T  54% /mnt/media
/dev/sdb2       3.6T  423G  3.0T  13% /mnt/mac-backup
/dev/sdb3       9.1T  4.5T  4.7T  49% /mnt/win-backup

dietpi@DietPi:~$ mount | grep sd
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
/dev/sda2 on /mnt/media type fuseblk (rw,noatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096,x-systemd.automount)
/dev/sdb2 on /mnt/mac-backup type ext4 (rw,noatime,lazytime,x-systemd.automount)
/dev/sdb3 on /mnt/win-backup type fuseblk (rw,noatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096,x-systemd.automount)

dietpi@DietPi:~$ ls /dev/sd*
/dev/sdb  /dev/sdb1  /dev/sdb2  /dev/sdb3  /dev/sdc  /dev/sdc1  /dev/sdc2

dietpi@DietPi:~$ udevadm info --attribute-walk /dev/sdc

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

[... 228 lines cut ...]
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1437M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=907af7d0-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=907af7d0-01 /boot vfat noatime,lazytime,rw 0 2
UUID=02061B92061B8637 /mnt/win-backup ntfs noatime,lazytime,rw,permissions,big_writes,nofail,noauto,x-systemd.automount
UUID=4CE4BDF9E4BDE4FC /mnt/media ntfs noatime,lazytime,rw,permissions,big_writes,nofail,noauto,x-systemd.automount
UUID=3e945285-c803-4388-b09d-a4370900c631 /mnt/mac-backup ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```

MichaIng, 2021-06-21:
Many thanks for your report.

It looks like the drive connection was lost completely and it then it was re-detected as new device. Can you please run and paste the following, shortly after the drive changed its device name?
```
dmesg | tail -20
journalctl -u mnt-media.automount -u mnt-media.mount
```

nave7693, 2021-06-22:
dmesg output:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-g$
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, $
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shar$
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 517632
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 321024 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50700 r8192 d23028 u81920
[    0.000000] pcpu-alloc: s50700 r8192 d23028 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 515328
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm283$
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes,$
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, l$
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000025400000-0x00000000294000$
[    0.000000] Memory: 1894700K/2070528K available (10240K kernel code, 1358K r$
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33942 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 ji$
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c wi$
[    0.000004] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every $
[    0.000011] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max$
[    0.000040] bcm2835: system timer (irq = 25)
[    0.000301] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000308] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles$
[    0.000316] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398$
[    0.000322] Switching to timer-based delay loop, resolution 18ns
[    0.000434] Console: colour dummy device 80x30
[    0.000463] printk: console [tty1] enabled
[    0.000489] Calibrating delay loop (skipped), value calculated using timer f$
[    0.000502] pid_max: default: 32768 minimum: 301
[    0.000575] LSM: Security Framework initializing
[    0.000656] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, line$
[    0.000669] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes,$
[    0.001380] Disabling memory control group subsystem
[    0.001436] CPU: Testing write buffer coherency: ok
[    0.001651] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002160] Setting up static identity map for 0x200000 - 0x20003c
[    0.002243] rcu: Hierarchical SRCU implementation.
[    0.002642] smp: Bringing up secondary CPUs ...
[    0.003202] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003820] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.004397] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 477 lines cut ...]
Jun 22 01:47:17 DietPi systemd[1]: mnt-media.automount: Got automount request fo
r /mnt/media, triggered by 241 (dietpi-arr_to_R)
Jun 22 01:47:17 DietPi systemd[1]: Mounting /mnt/media...
Jun 22 01:47:18 DietPi mount[282]: Using default user mapping
Jun 22 01:47:19 DietPi ntfs-3g[304]: Version 2017.3.23AR
.3 integrated FUSE 28
Jun 22 01:47:19 DietPi ntfs-3g[304]: Mounted /dev/sda2 (
Read-Write, label "Media", NTFS 3.1)
Jun 22 01:47:19 DietPi ntfs-3g[304]: Cmdline options: rw
,noatime,lazytime,permissions,big_writes
Jun 22 01:47:19 DietPi ntfs-3g[304]: Mount options: lazy
time,allow_other,nonempty,default_permissions,noatime,rw,fsname=/dev/sda2,blkdev
,blksize=4096
Jun 22 01:47:19 DietPi ntfs-3g[304]: User mapping built,
 Posix ACLs not used, configuration type 7
Jun 22 01:47:19 DietPi systemd[1]: Mounted /mnt/media.
Jun 22 01:54:13 DietPi ntfs-3g[304]: ntfs_attr_pread_i:
ntfs_pread failed: Input/output error
< MORE OUTPUT SNIPPED >
Jun 22 01:54:13 DietPi ntfs-3g[304]: ntfs_attr_pread err
or reading '/dietpi_userdata/docker-data/devicemapper/devicemapper/metadata' at
offset 786432: 4096 <> -1: Input/output error
Jun 22 01:54:13 DietPi ntfs-3g[304]: ntfs_attr_pread_i:
ntfs_pread failed: Input/output error
```

MichaIng, 2021-06-22:
So at 320s a series of error messages around the drive appear. Can you check if voltage is involved in any way?
```sh
vcgencmd get_throttled
```
Since you use multiple external drives, how are those powered, via USB or dedicated PSU?

And if voltage is not involved + drives are powered via dedicated PSU, I see that error messages like `uas_eh_abort_handler` contain the "UAS" keyword. UAS is enabled by default since RPi4 but is known to cause issues with some drives. So it could be tried to disable/blacklist it:
```sh
echo 'blacklist uas' > /etc/modprobe.d/disable_uas.conf
reboot
```

nave7693, 2021-06-23:
Voltage looks fine:
```
dietpi@DietPi:~$ sudo vcgencmd get_throttled
throttled=0x0
```

Looks like it is indeed UAS. The blacklist file in modprobe.d did not work because the driver seems to be compiled into the kernel. Adding this parameter to the kernel command line worked:
```
usb-storage.quirks=1e91:a3a6:u
```

Maybe something snuck into the kernel code that broke my particular USB-to-SATA chipset. It is a JMicron one which should be fine from what I Googled). Bummed that it would be slightly slower but it's better than not working :) Was there a kernel upgrade between 7.1 and 7.2?

Joulinar, 2021-06-23:
We had a couple of reports on the forum about the `JMicron` chip. Usually issues solved by getting a different adapter. 🙄

MichaIng, 2021-06-23:
Best individual solution with the USB quick 👍. That way UAS is still available for the other drives.

Kernel upgrades are independent from DietPi updates, as `apt upgrade` will apply them as well. But let me see... current kernel package was released on May 28, DietPi v7.2 was released on May 29, so jep there as definitely a kernel package upgrade. But the last one actually did not include a kernel version bump but only a firmware fix, so did the one before on May 12, ... even March 3 was Linux v5.10.17 already, so that was long before DietPi v7.1.

UAS is definitely not built into the kernel, as this would break older RPi models:
```
# modinfo uas
filename:       /lib/modules/5.10.17-v7+/kernel/drivers/usb/storage/uas.ko
```
But probably it is still pulled in automatically. A blacklist e.g. does not prevent the module from being loaded as dependency or manually. To check, whether it's loaded:
```
lsmod | grep uas
```
It doesn't make sense in your case, as the drive-specific USB quirk works, but to prevent the module from being loaded at all:
```
echo 'install uas /bin/true' >> /etc/modprobe.d/disable_uas.conf
```
Appended, to preserve the blacklist, to prevent a possible warning/error at boot.

MichaIng, 2021-09-18:
I'll mark this as closed as known issue.

closed 2021-09-18 by MichaIng