MichaIng/DietPi #4500: Upgrade Home assistant python 3.9

maury77, 2021-06-21:
what is the procedurre for this upgrade?

maury77, 2021-06-21:
I have the last version 2021.6.5 with python 3.8.5 on dietpi , but I have a problem with an integration & dietpi and I suppose the issue is related to python version

the process is blockend on this istruction  await hass.async_add_executor_job. with no logs of errors

In other distribution the problem is not exists.

MichaIng, 2021-06-21:
A reinstall will compile Python 3.8.10 currently. The last time I checked, Home Assistant stated compatibility with Python 3.8 but not Python 3.9 yet.

MichaIng, 2021-06-27:
Did the reinstall help?

maury77, 2021-07-04:
in the end I haven't reinstalled yet. 

I don't understand why on my installation the landroid cloud plugin (https://github.com/MTrab/landroid_cloud/blob/master/README.md)

stops on this instruction

**hass.async_add_executor_job(client[dev].connect, device, False)**

and then timeout connection in other installation this problem is not present, in the log no information or error

MichaIng, 2021-07-24:
I mark this as closed. Feel free to reopen if required.

closed 2021-07-24 by MichaIng

PrzemekSkw, 2021-11-04:
> ```
> sudo -u homeassistant bash
> . /home/homeassistant/pyenv-activate.sh
> pip3 install -U <module>
> ```

Hello, I have last HA version and have notification to upgrade python to 3.9. Can You tell me what to put in <module> to upgrade python?
Regards.

Joulinar, 2021-11-04:
usually a reinstall of HA should do this.

MichaIng, 2021-11-04:
If you do **not** run a Stretch or Buster Raspberry Pi ARMv6 or ARMv7 system, simply reinstall it:
```sh
dietpi-software reinstall 157
```
If you runs such a system, wait with the reinstall until we fixed: https://github.com/MichaIng/DietPi/issues/4928

PrzemekSkw, 2021-11-05:
@MichaIng I install Dietpi on intel MiniPC x64. Can I reinstall?
Regards.

MichaIng, 2021-11-05:
Yes exactly.