MichaIng/DietPi #4502: Error when updating

tanandry, 2021-06-22:
#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7 G_DIETPI_VERSION_SUB=2 G_DIETPI_VERSION_RC=3
- Distro version | 10.10
- Kernel version | Linux DietPi-RASPI 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official power supply
- SDcard used | 128GB SSD

#### Additional Information (if applicable)
- Bug report ID | c5d234c9-c845-4aeb-b795-8408eb50c6e0

#### Steps to reproduce

1. G_AGUP && G_AGUG
2. Wait and the error happen

#### Extra details

```
[  OK  ] APT update
[ INFO ] APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 56766 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+deb10u10_arm64.deb ...
Unpacking base-files (10.3+deb10u10) over (10.3+deb10u9) ...
Setting up base-files (10.3+deb10u10) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 56766 files and directories currently installed.)
Preparing to unpack .../mariadb-common_1%3a10.3.29-0+deb10u1_all.deb ...
Unpacking mariadb-common (1:10.3.29-0+deb10u1) over (1:10.3.27-0+deb10u1) ...
Preparing to unpack .../libnettle6_3.4.1-1+deb10u1_arm64.deb ...
Unpacking libnettle6:arm64 (3.4.1-1+deb10u1) over (3.4.1-1) ...
Setting up libnettle6:arm64 (3.4.1-1+deb10u1) ...
(Reading database ... 56766 files and directories currently installed.)
Preparing to unpack .../libhogweed4_3.4.1-1+deb10u1_arm64.deb ...
Unpacking libhogweed4:arm64 (3.4.1-1+deb10u1) over (3.4.1-1) ...
Setting up libhogweed4:arm64 (3.4.1-1+deb10u1) ...
(Reading database ... 56766 files and directories currently installed.)
Preparing to unpack .../libgnutls30_3.6.7-4+deb10u7_arm64.deb ...
Unpacking libgnutls30:arm64 (3.6.7-4+deb10u7) over (3.6.7-4+deb10u6) ...
Setting up libgnutls30:arm64 (3.6.7-4+deb10u7) ...
(Reading database ... 56766 files and directories currently installed.)
Preparing to unpack .../mariadb-server-core-10.3_1%3a10.3.29-0+deb10u1_arm64.deb ...
Unpacking mariadb-server-core-10.3 (1:10.3.29-0+deb10u1) over (1:10.3.27-0+deb10u1) ...
Preparing to unpack .../mariadb-client-core-10.3_1%3a10.3.29-0+deb10u1_arm64.deb ...
Unpacking mariadb-client-core-10.3 (1:10.3.29-0+deb10u1) over (1:10.3.27-0+deb10u1) ...
Preparing to unpack .../mariadb-client-10.3_1%3a10.3.29-0+deb10u1_arm64.deb ...
Unpacking mariadb-client-10.3 (1:10.3.29-0+deb10u1) over (1:10.3.27-0+deb10u1) ...
Setting up mariadb-common (1:10.3.29-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
E: Write error - ~LZMAFILE (28: No space left on device)
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2021-06-22:
Hi,

Looks like your file system is full

`E: Write error - ~LZMAFILE (28: No space left on device)`

Can you check available disk space `df -h`

tanandry, 2021-06-22:
Here it is:
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       118G   26G   88G  23% /
devtmpfs        3.7G     0  3.7G   0% /dev
tmpfs           3.7G   40K  3.7G   1% /dev/shm
tmpfs           3.7G   12M  3.7G   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.7G     0  3.7G   0% /sys/fs/cgroup
tmpfs           3.9G  9.1M  3.9G   1% /tmp
tmpfs            50M  332K   50M   1% /var/log
/dev/sda1       253M   33M  220M  13% /boot
/dev/sdb1       3.7T  111G  3.6T   3% /mnt/DATACENTER1
/dev/sdd1       1.8T  496G  1.4T  28% /mnt/MEDIACENTER1
tmpfs           757M     0  757M   0% /run/user/1000
```

MichaIng, 2021-06-22:
Lol, first match on DuckDuckGo for this error message is another report on our repo: #2597
There it was a filesystem error it seems. Can you check:
```sh
dmesg -l emerg,alert,crit,err
```

tanandry, 2021-06-22:
Yeah I search for that too before posting.

```
[    0.576370] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    2.698908] sd 2:0:0:0: [sdd] No Caching mode page found
[    2.698923] sd 2:0:0:0: [sdd] Assuming drive cache: write through
```

MichaIng, 2021-06-22:
Okay, no corruption is the issue here. I found other cases where indeed the filesystem was full, in another case the RAMlog, which was not cleared hourly due to a bug in DietPi v6.11 that time. In your case `/var/log` is pretty empty, as expected, so just to rule out that it was a temporary full RAMlog (e.g. it fills up just shortly before it is cleared by the hourly cron job), could you retry:
```sh
df /var/log
apt update
apt upgrade
df /var/log
```

tanandry, 2021-06-22:
df /var/log
```
Filesystem     1K-blocks  Used Available Use% Mounted on
tmpfs              51200  9444     41756  19% /var/log
```
apt update
```
Hit:1 https://download.mono-project.com/repo/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster InRelease                                                                                                                                     
Hit:3 https://deb.debian.org/debian buster-updates InRelease                                                                                                                             
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease                                                 
Hit:5 https://deb.debian.org/debian buster-backports InRelease                                  
Hit:6 https://apt.sonarr.tv/debian buster InRelease                                             
Hit:7 https://archive.raspberrypi.org/debian buster InRelease       
Hit:8 https://repo.jellyfin.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
13 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
apt upgrade
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libreoffice-common
The following packages will be upgraded:
  isc-dhcp-client libgcrypt20 libgl1-mesa-dri libglapi-mesa libglib2.0-0 libglx-mesa0 libmariadb3 libxml2 mariadb-server mariadb-server-10.3 python-dnspython rpi-eeprom
12 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
4 not fully installed or removed.
Need to get 0 B/13.8 MB of archives.
After this operation, 172 kB disk space will be freed.
Do you want to continue? [Y/n] Y
debconf: delaying package configuration, since apt-utils is not installed
Setting up mariadb-common (1:10.3.29-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
df /var/log
```
Filesystem     1K-blocks  Used Available Use% Mounted on
tmpfs              51200  9444     41756  19% /var/log
```

Joulinar, 2021-06-22:
ok the space issue is gone however MariaDB still has issues. It seems config file is missing. Can you share `ls -la /etc/mysql/`

tanandry, 2021-06-22:
```
total 20
drwxr-xr-x   3 root root 4096 Jun 22 19:57 .
drwxr-xr-x 103 root root 4096 Jun 23 05:18 ..
-rwxr-xr-x   1 root root 1620 Oct 13  2020 debian-start
-rw-------   1 root root  277 Apr  7 13:27 debian.cnf
drwxr-xr-x   2 root root 4096 Jun 22 19:56 mariadb.conf.d
```

Joulinar, 2021-06-22:
Uhh you are missing some more file/directory. Do you have a backup of your system?

MichaIng, 2021-06-23:
Strange is:
```
# dpkg -S /etc/mysql/mariadb.cnf
mariadb-common: /etc/mysql/mariadb.cnf
```
This file is served by `mariadb-common` itself. Something must have messed up when it failed to be installed with this space error and now it things it is installed already and needs to be configured only. The following might solve it:
```
cd /tmp
curl -sSfLO http://deb.debian.org/debian/pool/main/m/mariadb-10.3/mariadb-common_10.3.29-0+deb10u1_all.deb
dpkg -i mariadb-common_10.3.29-0+deb10u1_all.deb
rm mariadb-common_10.3.29-0+deb10u1_all.deb
apt upgrade
```

Joulinar, 2021-06-23:
@MichaIng 
MariaDB is already installed and `apt` is trying to update the package. But this is failing as the conf file is missing. I could replicate the behaviour. I simply installed MariaDB, removed the conf file and tried to `apt install --reinstall`, which failed with same error. The question is, why are the configuration files are lost on an already running systems? 🤔

tanandry, 2021-06-23:
Still error:
```
root@DietPi-RASPI:/tmp# cd /tmp
root@DietPi-RASPI:/tmp# curl -sSfLO http://deb.debian.org/debian/pool/main/m/mysql-defaults/mysql-common_5.8+1.0.5_all.deb
root@DietPi-RASPI:/tmp# dpkg -i mysql-common_5.8+1.0.5_all.deb
(Reading database ... 56507 files and directories currently installed.)
Preparing to unpack mysql-common_5.8+1.0.5_all.deb ...
Unpacking mysql-common (5.8+1.0.5) over (5.8+1.0.5) ...
Setting up mysql-common (5.8+1.0.5) ...
root@DietPi-RASPI:/tmp# rm mysql-common_5.8+1.0.5_all.deb
root@DietPi-RASPI:/tmp# curl -sSfLO http://deb.debian.org/debian/pool/main/m/mariadb-10.3/mariadb-common_10.3.29-0+deb10u1_all.deb
root@DietPi-RASPI:/tmp# dpkg -i mariadb-common_10.3.29-0+deb10u1_all.deb
(Reading database ... 56507 files and directories currently installed.)
Preparing to unpack mariadb-common_10.3.29-0+deb10u1_all.deb ...
Unpacking mariadb-common (1:10.3.29-0+deb10u1) over (1:10.3.29-0+deb10u1) ...
Setting up mariadb-common (1:10.3.29-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--install):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
root@DietPi-RASPI:/tmp# rm mariadb-common_10.3.29-0+deb10u1_all.deb
root@DietPi-RASPI:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libreoffice-common
The following packages will be upgraded:
  isc-dhcp-client libgcrypt20 libgl1-mesa-dri libglapi-mesa libglib2.0-0 libglx-mesa0 libmariadb3 libxml2 mariadb-server mariadb-server-10.3 python-dnspython rpi-eeprom
12 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
4 not fully installed or removed.
Need to get 0 B/13.8 MB of archives.
After this operation, 172 kB disk space will be freed.
Do you want to continue? [Y/n] Y
debconf: delaying package configuration, since apt-utils is not installed
Setting up mariadb-common (1:10.3.29-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

@Joulinar I'm not sure at all. Everything else works without any problem though. My Nextcloud works fine. The problem of "No space left on device" happen quite sometimes ago during an update too. But when I redo the command the update proceeded without any problem and the update was installed. I forgot during what update the problem occurred. The difference right now it stuck.

Joulinar, 2021-06-23:
yes it stuck because of missing configuration file `/etc/mysql/mariadb.cnf`. It seems to be mandatory for the `apt upgrade`, even if the database themselves is working without. Usually `/etc/mysql/` should contain following

```
drwxr-xr-x  2 root root 4096 Jun 23 09:30 conf.d
-rwxr-xr-x  1 root root 1620 May 10 05:06 debian-start
-rw-------  1 root root  277 Jun 23 09:30 debian.cnf
-rw-r--r--  1 root root  869 May 10 05:06 mariadb.cnf
drwxr-xr-x  2 root root 4096 Jun 23 09:30 mariadb.conf.d
lrwxrwxrwx  1 root root   24 Jun 23 09:30 my.cnf -> /etc/alternatives/my.cnf
-rw-r--r--  1 root root  839 Aug  3  2016 my.cnf.fallback
```

As a workaround we could manually create `mariadb.cnf`

1. `nano /etc/mysql/mariadb.cnf`
2. copy following inside
```
# The MariaDB configuration file
#
# The MariaDB/MySQL tools read configuration files in the following order:
# 1. "/etc/mysql/mariadb.cnf" (this file) to set global defaults,
# 2. "/etc/mysql/conf.d/*.cnf" to set global options.
# 3. "/etc/mysql/mariadb.conf.d/*.cnf" to set MariaDB-only options.
# 4. "~/.my.cnf" to set user-specific options.
#
# If the same option is defined multiple times, the last one will apply.
#
# One can use all long options that the program supports.
# Run program with --help to get a list of available options and with
# --print-defaults to see which it would actually understand and use.

#
# This group is read both both by the client and the server
# use it for options that affect everything
#
[client-server]

# Import all .cnf files from configuration directory
!includedir /etc/mysql/conf.d/
!includedir /etc/mysql/mariadb.conf.d/
```
3. try to finish package installation `apt install -f`
4. check if all packages are updated `apt update && apt upgrade -y`

MichaIng, 2021-06-23:
Ah, of course, all files in `/etc` are "conffiles" and hence won't be automatically re-created on package upgrades. Please try:
```
cd /tmp
curl -sSfLO http://deb.debian.org/debian/pool/main/m/mysql-defaults/mysql-common_5.8+1.0.5_all.deb
dpkg --force-confmiss -i mysql-common_5.8+1.0.5_all.deb
rm mysql-common_5.8+1.0.5_all.deb
curl -sSfLO http://deb.debian.org/debian/pool/main/m/mariadb-10.3/mariadb-common_10.3.29-0+deb10u1_all.deb
dpkg --force-confmiss -i mariadb-common_10.3.29-0+deb10u1_all.deb
rm mariadb-common_10.3.29-0+deb10u1_all.deb
apt upgrade
```

tanandry, 2021-06-24:
Problem solved. Thanks guys!

Joulinar, 2021-06-24:
How did you solved it at the end?

tanandry, 2021-06-24:
Using `dpkg --force-confmiss` the mariadb.cnf was created automatically and the update proceeded nicely.

Joulinar, 2021-06-24:
thx for sharing, I'm going to close the issue. Feel free to reopen if needed.

closed 2021-06-24 by Joulinar

tanandry, 2021-06-27:
Hmm.. Another issue of the same type poped up when updating to 7.3.2. But this time after retrying the update, it is updated just fine.
```
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 56382 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_12.7-1_arm64.deb ...
Unpacking rpi-eeprom (12.7-1) over (12.6-1) ...
Setting up rpi-eeprom (12.7-1) ...
E: Write error - ~LZMAFILE (28: No space left on device)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-Bugreport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-Bugreport | Packing upload archive
[  OK  ] DietPi-Bugreport | Sending bug report
[ INFO ] DietPi-Bugreport | Reference code: c5d234c9-c845-4aeb-b795-8408eb50c6e0
```

Joulinar, 2021-06-27:
again the error with the `E: Write error - ~LZMAFILE (28: No space left on device)`

Probably still a side effect from the moment you had the full file systems. Maybe there are temp files left that will be cleaned once you perform the `apt update` again