MichaIng/DietPi #4503: Issue installing PiVPN

Kingkobra813, 2021-06-22:
Hello,

I'm a complete noob when it comes to linux and things. I'm just trying to install PiVPN and I keep getting this error. I'm not sure how to fix and the googling I did do didn't give me any helpful results 

#### Details:
- Date           | Wed Jun 23 00:30:03 BST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | curl --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT ed796366-423d-428c-96dc-36b8ca8fd71c.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/bugreport/
- Exit code      | 28
- Software title | DietPi-Bugreport
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
* Expire in 0 ms for 6 (transfer 0x82c8b0)
* Expire in 8000 ms for 2 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 1 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 2 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
* Expire in 0 ms for 1 (transfer 0x82c8b0)
[... 156 lines cut ...]
* Expire in 8 ms for 1 (transfer 0x82c8b0)
* Expire in 10 ms for 1 (transfer 0x82c8b0)
* Expire in 10 ms for 1 (transfer 0x82c8b0)
* Expire in 16 ms for 1 (transfer 0x82c8b0)
* Expire in 14 ms for 1 (transfer 0x82c8b0)
* Expire in 14 ms for 1 (transfer 0x82c8b0)
* Expire in 16 ms for 1 (transfer 0x82c8b0)
* Expire in 14 ms for 1 (transfer 0x82c8b0)
* Expire in 14 ms for 1 (transfer 0x82c8b0)
* Expire in 50 ms for 1 (transfer 0x82c8b0)
*   Trying 185.101.93.93...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x82c8b0)
* Connection timed out after 8000 milliseconds
* Closing connection 0
* Expire in 0 ms for 6 (transfer 0x82c8b0)
* Expire in 8000 ms for 2 (transfer 0x82c8b0)
* Hostname ssh.dietpi.com was found in DNS cache
*   Trying 185.101.93.93...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x82c8b0)
* Connection timed out after 8001 milliseconds
* Closing connection 1
curl: (28) Connection timed out after 8001 milliseconds
```

Joulinar, 2021-06-22:
Hi, 

many thanks for your message. We had a similar case in past where the issue was slow DNS resolution. At least your error message looks similar to https://github.com/MichaIng/DietPi/issues/4061

Can you do a simple test `time wget --spider https://dietpi.com`

Kingkobra813, 2021-06-22:
here are the results

```
dietpi@DietPi:~$ time wget --spider https://dietpi.com
Spider mode enabled. Check if remote file exists.
--2021-06-23 00:48:21--  https://dietpi.com/
Resolving dietpi.com (dietpi.com)... 104.21.96.47, 172.67.173.4, 2606:4700:3034::6815:602f, ...
Connecting to dietpi.com (dietpi.com)|104.21.96.47|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.

real	0m0.672s
user	0m0.089s
sys	0m0.020s
```

MichaIng, 2021-06-23:
That worked well. The log above already shows that the correct IP was found in the first attempt, and in the retry it was found in cache immediately, so the connection itself timed out, not DNS. Let's check if it is somehow curl explicitly:
```
curl -IL https://dietpi.com/
```
I just tested a survey upload, and our SSH/SFTP server responds quickly here, so that is not the issue.

Btw, you said you tried to install PiVPN but the error our posted is about the bug report upload. Do you still have the log or remember which connection initially failed, like downloading the PiVPN installer or so?

Kingkobra813, 2021-06-23:
here are the results
```
dietpi@DietPi:~$ curl -IL https://dietpi.com/
HTTP/2 200 
date: Wed, 23 Jun 2021 00:44:16 GMT
content-type: text/html; charset=utf-8
strict-transport-security: max-age=31536000; includeSubDomains; preload
last-modified: Sat, 19 Jun 2021 21:57:46 GMT
vary: Accept-Encoding
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 1; mode=block
x-robots-tag: all
x-download-options: noopen
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer-when-downgrade
expect-ct: enforce, max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
content-security-policy: upgrade-insecure-requests; frame-ancestors 'self'; form-action 'self'; base-uri https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline' https://fonts.googleapis.com; font-src 'self' https://fonts.gstatic.com data:; img-src * data:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://unpkg.com https://google.com/recaptcha/ https://www.google.com/recaptcha/ https://www.gstatic.com/recaptcha/; frame-src 'self' https://www.youtube-nocookie.com https://google.com/recaptcha/ https://www.google.com/recaptcha/; manifest-src 'self'; connect-src 'self' https://api.github.com
permissions-policy: accelerometer=(), autoplay=(), camera=(), document-domain=(), encrypted-media=(), fullscreen=(self "https://www.youtube-nocookie.com"), geolocation=(), gyroscope=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
cache-control: public, max-age=86399
content-language: en
cf-cache-status: DYNAMIC
cf-request-id: 0ad7ec3b8f000027f450a1c000000001
report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v2?s=95vvxFbV6S7uGlYo6yIeSmjEZwz113X9B40haGWmnZ51bLFNn16KcYMjj2TtqYc9sNjwTjPVRcmr%2Fblo5ZvUJ4bVIsS%2FdwHMgFadkC8hYqIDBOmTfN0xCA%3D%3D"}],"group":"cf-nel","max_age":604800}
nel: {"report_to":"cf-nel","max_age":604800}
server: cloudflare
cf-ray: 6639afd8e80f27f4-SLC
alt-svc: h3-27=":443"; ma=86400, h3-28=":443"; ma=86400, h3-29=":443"; ma=86400, h3=":443"; ma=86400```

I don't have the log and I don't remember which connection failed. I can try again if that would help

MichaIng, 2021-06-23:
Ah I forgot to add the `time` command to measure the time it took. But I guess you would have recognised if it took more than 8 seconds.

> I can try again if that would help

Yes, please do that, as it might have been a temporary network issue.

Kingkobra813, 2021-06-23:
I'm not sure how to get the logs but this what I got when I retried it. 
```
sudo dietpi-software 
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Free space check: path=/ | available=25257 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : nginx
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | Git will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Git: Clone and manage Git repositories locally

[ INFO ] DietPi-Software | APT install for: git, please wait...
[  OK  ] DietPi-Software | APT install for: git

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing PiVPN: openvpn/wireguard server install & management tool

[FAILED] DietPi-Software | Checking URL: https://install.pivpn.io
 - Command: curl -ILfvm 10 https://install.pivpn.io

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Jun 23 02:18:34 BST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | curl -ILfvm 10 https://install.pivpn.io
- Exit code      | 7
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
[... 133 lines cut ...]
* Expire in 1 ms for 1 (transfer 0xd448b0)
* Expire in 2 ms for 1 (transfer 0xd448b0)
* Expire in 2 ms for 1 (transfer 0xd448b0)
* Expire in 2 ms for 1 (transfer 0xd448b0)
* Expire in 2 ms for 1 (transfer 0xd448b0)
*   Trying 88.198.194.194...
* TCP_NODELAY set
* Expire in 4996 ms for 3 (transfer 0xd448b0)
* Expire in 200 ms for 4 (transfer 0xd448b0)
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0* After 4996ms connect time, move on!
* connect to 88.198.194.194 port 443 failed: Connection timed out
*   Trying 88.198.194.195...
* TCP_NODELAY set
* Expire in 2498 ms for 3 (transfer 0xd448b0)
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0* After 2498ms connect time, move on!
* connect to 88.198.194.195 port 443 failed: Connection timed out
* Failed to connect to install.pivpn.io port 443: Connection timed out
* Closing connection 0
curl: (7) Failed to connect to install.pivpn.io port 443: Connection timed out
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
```

I can also try that curl request again with the time command. I just don't know how

ravenclaw900, 2021-06-23:
Just prefix it with `time`:
```
time curl -IL https://dietpi.com/
```

Kingkobra813, 2021-06-23:
> Just prefix it with `time`:
> 
> ```
> time curl -IL https://dietpi.com/
> ```

```
dietpi@DietPi:~$ time curl -IL https://dietpi.com/
HTTP/2 200 
date: Wed, 23 Jun 2021 01:31:44 GMT
content-type: text/html; charset=utf-8
strict-transport-security: max-age=31536000; includeSubDomains; preload
last-modified: Sat, 19 Jun 2021 21:57:46 GMT
vary: Accept-Encoding
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 1; mode=block
x-robots-tag: all
x-download-options: noopen
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer-when-downgrade
expect-ct: enforce, max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
content-security-policy: upgrade-insecure-requests; frame-ancestors 'self'; form-action 'self'; base-uri https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline' https://fonts.googleapis.com; font-src 'self' https://fonts.gstatic.com data:; img-src * data:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://unpkg.com https://google.com/recaptcha/ https://www.google.com/recaptcha/ https://www.gstatic.com/recaptcha/; frame-src 'self' https://www.youtube-nocookie.com https://google.com/recaptcha/ https://www.google.com/recaptcha/; manifest-src 'self'; connect-src 'self' https://api.github.com
permissions-policy: accelerometer=(), autoplay=(), camera=(), document-domain=(), encrypted-media=(), fullscreen=(self "https://www.youtube-nocookie.com"), geolocation=(), gyroscope=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
cache-control: public, max-age=86399
content-language: en
cf-cache-status: DYNAMIC
cf-request-id: 0ad817af790000a05cb8818000000001
report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v2?s=VBNEOHZg1OvWoDrimauenZZqIj6rwJgKQgNG7aJMqumU9OVLOxTWkT5cPueAtlbFly1upaAFjspkTItGBtIMqq7Dk%2BZYMHKw%2B%2BDoedkNxnPBiEDHFJ2HOQ%3D%3D"}],"group":"cf-nel","max_age":604800}
nel: {"report_to":"cf-nel","max_age":604800}
server: cloudflare
cf-ray: 6639f55f29d4a05c-SLC
alt-svc: h3-27=":443"; ma=86400, h3-28=":443"; ma=86400, h3-29=":443"; ma=86400, h3=":443"; ma=86400

real	0m0.638s
user	0m0.054s
sys	0m0.018s
```

MichaIng, 2021-06-23:
Stange, so indeed it was when downloading the PiVPN installer, respectively when connecting to its server. We can retry the exact same `curl` connection from console:
```sh
time curl -IL https://install.pivpn.io
```
but I have a feeling that it succeeds (in below 5 seconds) like the dietpi.com connection does, so that it is somehow in the context of the script.

The timeout of the URL check can be increased:
```sh
G_CONFIG_INJECT 'CONFIG_G_CHECK_URL_TIMEOUT=' 'CONFIG_G_CHECK_URL_TIMEOUT=30' /boot/dietpi.txt
```
This may work around the issue, but leaves the open question why the connection takes so long or fails during the `dietpi-software` execution.

Kingkobra813, 2021-06-23:
It looks like it took over 3 mins

```
dietpi@DietPi:~$ time curl -IL https://install.pivpn.io
curl: (7) Failed to connect to install.pivpn.io port 443: Connection timed out

real	3m35.151s
user	0m0.029s
sys	0m0.046s
```

MichaIng, 2021-06-23:
Strange, here it takes 0.5 seconds, so not a general server issue. Does it work normally to open this URL in your browser (showing the shell script code)?

The URL actually is redirected to the raw file from the master GitHub branch. Does this work better?
```
curl -IL https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh
```
We could pull the installer from there directly and skip the prior `install.pivpn.io` connection and redirect.

Kingkobra813, 2021-06-23:
I'm able to go to install.pivpn.io in my browser just fine.

```
dietpi@DietPi:~$ time curl -IL https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh
curl: (6) Could not resolve host: raw.githubusercontent.com

real	0m10.130s
user	0m0.038s
sys	0m0.022s
```

Joulinar, 2021-06-23:
`Could not resolve host: raw.githubusercontent.com`

Do you use something like PiHole or AdGuard?

MichaIng, 2021-06-23:
If so, please try to add `raw.githubusercontent.com` to the whitelist. Would be great to find out which blocklist is responsible for this as this is not the first case we see this and it breaks a large number of downloads and there is really zero reason to block the number one open source platform URL for ads.

Kingkobra813, 2021-06-23:
ahh, yes. I use PiHole. I'll try and disable it and try again.

```
dietpi@DietPi:~$ time curl -IL https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh
HTTP/2 200 
cache-control: max-age=300
content-security-policy: default-src 'none'; style-src 'unsafe-inline'; sandbox
content-type: text/plain; charset=utf-8
etag: "5df55bdd0292faffa2b74f4dfc32fe86f870bd5a4f76b100e6b78ad54a35d554"
strict-transport-security: max-age=31536000
x-content-type-options: nosniff
x-frame-options: deny
x-xss-protection: 1; mode=block
x-github-request-id: 0D7A:353A:246EA9:390892:60D34030
accept-ranges: bytes
date: Wed, 23 Jun 2021 14:33:44 GMT
via: 1.1 varnish
x-served-by: cache-den8267-DEN
x-cache: HIT
x-cache-hits: 1
x-timer: S1624458825.604226,VS0,VE274
vary: Authorization,Accept-Encoding
access-control-allow-origin: *
x-fastly-request-id: 6dfe602c55085fca8d8e14a5e1e07aaf351f4805
expires: Wed, 23 Jun 2021 14:38:44 GMT
source-age: 0
content-length: 84313

real	0m0.511s
user	0m0.049s
sys	0m0.019s
```

but this is still taking a while

```
dietpi@DietPi:~$ time curl -IL https://install.pivpn.io
curl: (7) Failed to connect to install.pivpn.io port 443: Connection timed out

real	3m35.226s
user	0m0.021s
sys	0m0.050s
```

MichaIng, 2021-06-23:
Could you check which blocklist is responsible for `raw.githubusercontent.com`?

While I'm not sure about `install.pivpn.io`, a workaround when the above has been sorted is:
```sh
sed -i 's|install.pivpn.io|raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh|' /boot/dietpi/dietpi-software
```

Kingkobra813, 2021-06-23:
I tried searching the web GUI and the command line and both resulted in 
```
[i] No results found for raw.githubusercontent.com within the block lists
```

Joulinar, 2021-06-23:
Maybe it took time to resolve the host name first, which is cached now. Maybe you can have a look to the query log for the url

MichaIng, 2021-06-23:
Strange. Probably not a blocklist is the issue but Pi-hole or, Unbound behind it, if you have both installed? On the other hand, for `install.pivpn.io` DNS seems to have succeeded and only the connection to the server failed. To verify:
```sh
getent hosts install.pivpn.io
```

Does Pi-hole and in case Unbound report any error?
```sh
journalctl -u pihole-FTL -u unbound | tail -20
tail -20 /var/log/pihole.log
tail -20 /var/log/pihole-FTL.log
```

Kingkobra813, 2021-06-23:
```
dietpi@DietPi:~$ getent hosts install.pivpn.io
88.198.194.195  install.pivpn.io
88.198.194.194  install.pivpn.io
```
```
root@DietPi:/home/dietpi# journalctl -u pihole-FTL -u unbound | tail -20
-- Logs begin at Tue 2021-06-22 22:33:15 BST, end at Wed 2021-06-23 16:01:51 BST. --
Jun 22 22:33:27 DietPi systemd[1]: Starting LSB: pihole-FTL daemon...
Jun 22 22:33:27 DietPi systemd[1]: Starting Unbound DNS server...
Jun 22 22:33:27 DietPi pihole-FTL[296]: Not running
Jun 22 22:33:28 DietPi su[324]: (to pihole) root on none
Jun 22 22:33:28 DietPi su[324]: pam_unix(su:session): session opened for user pihole by (uid=0)
Jun 22 22:33:28 DietPi package-helper[305]: /var/lib/unbound/root.key has content
Jun 22 22:33:28 DietPi package-helper[305]: success: the anchor is ok
Jun 22 22:33:29 DietPi unbound[330]: [330:0] info: start of service (unbound 1.9.0).
Jun 22 22:33:29 DietPi systemd[1]: Started Unbound DNS server.
Jun 22 22:33:33 DietPi pihole-FTL[296]: FTL started!
Jun 22 22:33:33 DietPi su[324]: pam_unix(su:session): session closed for user pihole
Jun 22 22:33:33 DietPi systemd[1]: Started LSB: pihole-FTL daemon.
root@DietPi:/home/dietpi# tail -20 /var/log/pihole.log
Jun 23 15:33:39 dnsmasq[344]: read /etc/hosts - 5 addresses
Jun 23 15:33:39 dnsmasq[344]: read /etc/pihole/custom.list - 0 addresses
Jun 23 15:33:39 dnsmasq[344]: read /etc/pihole/local.list - 4 addresses
root@DietPi:/home/dietpi# tail -20 /var/log/pihole-FTL.log
[2021-06-23 15:33:39.895 344M] Reloading DNS cache
[2021-06-23 15:33:39.903 344M] Blocking status is disabled
[2021-06-23 15:33:40.025 344/T348] Compiled 0 whitelist and 0 blacklist regex filters for 19 clients in 2.8 msec
[2021-06-23 15:53:24.369 344M] Received: Real-time signal 0 (34 -> 0)
[2021-06-23 15:53:24.480 344/T348] Compiled 0 whitelist and 0 blacklist regex filters for 19 clients in 2.6 msec
[2021-06-23 15:58:22.053 344M] Resizing "FTL-strings" from 86016 to (90112 * 1) == 90112 (/dev/shm: 3.3MB used, 250.4MB total, FTL uses 3.2MB)
```

MichaIng, 2021-06-23:
Okay that looks all fine. Let's also have a look into the kernel logs, just to rule it out:
```sh
dmesg | tail -20
```

Btw, if Pi-hole is running on the same server, I'd generally not have that server using Pi-hole as well, so it cannot break its own DNS resolver when Pi-hole or Unbound fails, is stopped or upgraded and such. As you usually do not use a browser on that system (if it is a pure server), there is not really a reason to block ads, also since Linux does not do various in-transparent connections home 🙂. E.g.:
```sh
sed -i 's/dns-nameservers 127.0.0.1/dns-nameservers 9.9.9.9/' /etc/network/interfaces
```
to switch to Quad9 DNS and by this rule out Pi-hole and Unbound for this issue.

And to not forget the workaround to allow PiVPN install:
```sh
sed -i 's|install.pivpn.io|raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh|' /boot/dietpi/dietpi-software
```

Kingkobra813, 2021-06-23:
I do have pihole running on a different raspberry pi but the pi i'm currently trying to install pivpn on is also running a web server. Do you think that could be what is causing the issue. Should I use a dedicated pi just for the pivpn?

I put all those inputs into the raspberry pi running pihole

```
root@DietPi:/home/dietpi# dmesg | tail -20
[    3.858437] Segment Routing with IPv6
[    3.963973] systemd[1]: systemd 241 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[    3.966103] systemd[1]: Detected architecture arm.
[    3.976766] systemd[1]: Set hostname to <DietPi>.
[    5.468858] random: crng init done
[    6.027196] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[    6.029522] systemd[1]: Listening on Journal Socket.
[    6.042352] systemd[1]: Starting Set the console keyboard layout...
[    6.055192] systemd[1]: Condition check resulted in Huge Pages File System being skipped.
[    6.057876] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    6.058268] systemd[1]: Reached target Slices.
[    6.059394] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    8.903455] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    9.077277] Adding 1606652k swap on /var/swap.  Priority:-2 extents:8 across:1868796k SSFS
[   10.037097] systemd-journald[84]: Received request to flush runtime journal from PID 1
[   12.198648] asix 1-1:1.0 eth0: register 'asix' at usb-20980000.usb-1, ASIX AX88772B USB 2.0 Ethernet, 00:00:10:02:8d:58
[   12.199092] usbcore: registered new interface driver asix
[   18.005286] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   18.008471] asix 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   18.029759] asix 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
root@DietPi:/home/dietpi# sed -i 's/dns-nameservers 127.0.0.1/dns-nameservers 9.9.9.9/' /etc/network/interfaces
root@DietPi:/home/dietpi# sed -i 's|install.pivpn.io|raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh|' /boot/dietpi/dietpi-software
root@DietPi:/home/dietpi# 
```

Joulinar, 2021-06-23:
usually there should be no issue running both on one system. Can you try to install PiVPN again pls as you changed the install link now.

Kingkobra813, 2021-06-23:
```
APT install for: dnsutils net-tools bsdmainutils iptables-persistent                                                 │ 
        │  - Command: apt-get -qq --allow-change-held-packages install dnsutils net-tools bsdmainutils iptables-persistent     │ 
        │  - Exit code: 100                                                                                                    │ 
        │  - DietPi version: v7.2.3 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5                                   │ 
        │  - Image creator: DietPi Core Team                                                                                   │ 
        │  - Pre-image: Raspberry Pi OS (32-bit) Lite                                                                          │ 
        │  - Error log:                                                                                                        │ 
        │ E: Failed to fetch                                                                                                   │ 
        │ http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u5_armhf.deb  Could not  │ 
        │ connect to raspbian.raspberrypi.org:80 (93.93.128.193), connection timed out [IP: 93.93.128.193 80]                  │ 
        │ E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?                                │ 
        │                                                                                                                      │ 
        │                         Retry          : Re-run the last command that failed                                         │ 
        │                         DietPi-Config  : Edit network, APT/NTP mirror settings etc                                   │ 
        │                         Open subshell  : Open a subshell to investigate or solve the issue                           │ 
        │                         Send report    : Uploads bugreport containing system info to DietPi                          │ 
        │                                        ●─ Devs only ──────────────────────────────────────●                          │ 
        │                         Change command : Adjust and rerun the command                                                │ 
        │                                                                                                                      │ 
        │                                                                                                                      │ 
        │                                  <Ok>                                      <Exit>       
```

Kingkobra813, 2021-06-23:
```
dietpi@DietPi:~$ sudo dietpi-software 
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Free space check: path=/ | available=25265 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : nginx
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Err:1 https://archive.raspberrypi.org/debian buster InRelease
  Could not resolve 'archive.raspberrypi.org'
Err:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Could not resolve 'raspbian.raspberrypi.org'
Err:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  Could not resolve 'raspbian.raspberrypi.org'
Reading package lists...
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Could not resolve 'raspbian.raspberrypi.org'
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Could not resolve 'raspbian.raspberrypi.org'
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Could not resolve 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | Git will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Git: Clone and manage Git repositories locally

[ INFO ] DietPi-Software | APT install for: git, please wait...
[  OK  ] DietPi-Software | APT install for: git

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing PiVPN: openvpn/wireguard server install & management tool

[FAILED] DietPi-Software | Checking URL: https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh
 - Command: curl -ILfvm 10 https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Jun 23 17:42:38 BST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | curl -ILfvm 10 https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh
[... 137 lines cut ...]
* connect to 185.199.109.133 port 443 failed: Connection timed out
*   Trying 185.199.111.133...
* TCP_NODELAY set
* Expire in 2498 ms for 3 (transfer 0xe498b0)
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0* After 2498ms connect time, move on!
* connect to 185.199.111.133 port 443 failed: Connection timed out
*   Trying 185.199.110.133...
* TCP_NODELAY set
* Expire in 1248 ms for 3 (transfer 0xe498b0)
  0     0    0     0    0     0      0      0 --:--:--  0:00:08 --:--:--     0* After 1248ms connect time, move on!
* connect to 185.199.110.133 port 443 failed: Connection timed out
*   Trying 185.199.108.133...
* TCP_NODELAY set
* Expire in 623 ms for 3 (transfer 0xe498b0)
  0     0    0     0    0     0      0      0 --:--:--  0:00:08 --:--:--     0* After 623ms connect time, move on!
* connect to 185.199.108.133 port 443 failed: Connection timed out
* Failed to connect to raw.githubusercontent.com port 443: Connection timed out
* Closing connection 0
curl: (7) Failed to connect to raw.githubusercontent.com port 443: Connection timed out
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

dietpi@DietPi:~$ 
```

Joulinar, 2021-06-23:
your device have quite some issues to connect to the web

`Could not resolve 'archive.raspberrypi.org'`
`curl: (7) Failed to connect to raw.githubusercontent.com port 443: Connection timed out`

Are you using WiFi or Ethernet?

Kingkobra813, 2021-06-23:
ethernet. 
But all my computers are having issues at the moment with the internet

Joulinar, 2021-06-23:
maybe restart your router?

Kingkobra813, 2021-06-23:
I was able to get it installed!!!

Thank you so much for your help. You friends are doing awesome work and I appreciate everything you do

Joulinar, 2021-06-23:
Maybe you like to share the final solution 😉

Kingkobra813, 2021-06-23:
> Okay that looks all fine. Let's also have a look into the kernel logs, just to rule it out:
> 
> ```shell
> dmesg | tail -20
> ```
> 
> Btw, if Pi-hole is running on the same server, I'd generally not have that server using Pi-hole as well, so it cannot break its own DNS resolver when Pi-hole or Unbound fails, is stopped or upgraded and such. As you usually do not use a browser on that system (if it is a pure server), there is not really a reason to block ads, also since Linux does not do various in-transparent connections home 🙂. E.g.:
> 
> ```shell
> sed -i 's/dns-nameservers 127.0.0.1/dns-nameservers 9.9.9.9/' /etc/network/interfaces
> ```
> 
> to switch to Quad9 DNS and by this rule out Pi-hole and Unbound for this issue.
> 
> And to not forget the workaround to allow PiVPN install:
> 
> ```shell
> sed -i 's|install.pivpn.io|raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh|' /boot/dietpi/dietpi-software
> ```

This seemed to resolve the issue

Joulinar, 2021-06-23:
ok, I'm going to close the issue as it seems to be solved. Feel free to reopen if needed.

closed 2021-06-23 by Joulinar