MichaIng/DietPi #4506: Wifi Hitspot on RPI4B failing to enable

psi5asp, 2021-06-23:
#### Details:
- Date           | Wed Jun 23 17:06:24 ACST 2021
- Bug report     | 6271d1ac-b9c9-400f-80c2-21024b72dc4f
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux Pi-2A 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config
#### Steps to reproduce:

1. No idea on current state of device settings - have no default model to compare with on this version, other than ETH already on and working.
2. Just launch (root) dietpi-config -> Network Adapters -> Wifi (Available [On] Disconnected) -> ToggleState from Disabled to Enable
3. Reboot makes no difference.

#### Expected behaviour:

- Wifi Hotspot should enable.
#### Actual behaviour:

- Wifi Hotspot is not being enabled
#### Extra details:

- ...
#### Additional logs:
```
# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: activating (start) since Wed 2021-06-23 17:14:26 ACST; 6ms ago
Cntrl PID: 1949 (hostapd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hostapd.service
           └─1949 /usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Jun 23 17:14:26 Pi-2A systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...

Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

```
Jun 23 17:15:49 Pi-2A systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
-- Subject: A start job for unit hostapd.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit hostapd.service has begun execution.
--
-- The job identifier is 8403.
Jun 23 17:15:49 Pi-2A hostapd[2006]: Configuration file: /etc/hostapd/hostapd.conf
Jun 23 17:15:49 Pi-2A hostapd[2006]: Line 10: invalid WPA passphrase length 6 (expected 8..63)
Jun 23 17:15:49 Pi-2A hostapd[2006]: WPA-PSK enabled, but PSK or passphrase is not configured.
Jun 23 17:15:49 Pi-2A hostapd[2006]: 2 errors found in configuration file '/etc/hostapd/hostapd.conf'
Jun 23 17:15:49 Pi-2A hostapd[2006]: Failed to set up interface with /etc/hostapd/hostapd.conf
Jun 23 17:15:49 Pi-2A hostapd[2006]: Failed to initialize interface
Jun 23 17:15:49 Pi-2A systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit hostapd.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jun 23 17:15:49 Pi-2A systemd[1]: hostapd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Jun 23 17:15:49 Pi-2A systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
-- Subject: A start job for unit hostapd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit hostapd.service has finished with a failure.
--
-- The job identifier is 8403 and the job result is failed.

```

Joulinar, 2021-06-23:
Hi,

many thanks for your report. There is an error message about the WPA passphrase length.

`Jun 23 17:15:49 Pi-2A hostapd[2006]: Line 10: invalid WPA passphrase length 6 (expected 8..63)`

can you have a look into `/etc/hostapd/hostapd.conf` is this is something you could extend?

psi5asp, 2021-06-23:
Hi Joulinar, Opps, I change the passphrase to 8+ characters, but still having the same issue. 

When installing the WiFi Hotspot software, I get this:

......

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing WiFi Hotspot: turn your device into a wifi hotspot

[ INFO ] DietPi-Software | APT install for: hostapd isc-dhcp-server iptables libnl-3-200, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package hostapd.
(Reading database ... 100211 files and directories currently installed.)
Preparing to unpack .../hostapd_2%3a2.7+git20190128+0c1e29f-6+deb10u3_arm64.deb ...
Unpacking hostapd (2:2.7+git20190128+0c1e29f-6+deb10u3) ...
Selecting previously unselected package libisccfg-export163.
Preparing to unpack .../libisccfg-export163_1%3a9.11.5.P4+dfsg-5.1+deb10u5_arm64.deb ...
Unpacking libisccfg-export163 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
Selecting previously unselected package libirs-export161.
Preparing to unpack .../libirs-export161_1%3a9.11.5.P4+dfsg-5.1+deb10u5_arm64.deb ...
Unpacking libirs-export161 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
Selecting previously unselected package isc-dhcp-server.
Preparing to unpack .../isc-dhcp-server_4.4.1-2+deb10u1_arm64.deb ...
Unpacking isc-dhcp-server (4.4.1-2+deb10u1) ...
Setting up hostapd (2:2.7+git20190128+0c1e29f-6+deb10u3) ...
Created symlink /etc/systemd/system/multi-user.target.wants/hostapd.service → /lib/systemd/system/hostapd.service.
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
Created symlink /etc/systemd/system/hostapd.service → /dev/null.
Setting up libisccfg-export163 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
Setting up libirs-export161 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
Setting up isc-dhcp-server (4.4.1-2+deb10u1) ...

Generating /etc/default/isc-dhcp-server...
Job for isc-dhcp-server.service failed because the control process exited with error code.
See "systemctl status isc-dhcp-server.service" and "journalctl -xe" for details.

invoke-rc.d: initscript isc-dhcp-server, action "start" failed.
● isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
   Active: failed (Result: exit-code) since Wed 2021-06-23 17:56:58 ACST; 26ms ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2990 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)

Jun 23 17:56:56 Pi-2A dhcpd[3002]: bugs on either our web page at www.isc.org or in the README file
Jun 23 17:56:56 Pi-2A dhcpd[3002]: before submitting a bug.  These pages explain the proper
Jun 23 17:56:56 Pi-2A dhcpd[3002]: process and the information we find helpful for debugging.
Jun 23 17:56:56 Pi-2A dhcpd[3002]:
Jun 23 17:56:56 Pi-2A dhcpd[3002]: exiting.
Jun 23 17:56:58 Pi-2A isc-dhcp-server[2990]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ... failed!
Jun 23 17:56:58 Pi-2A isc-dhcp-server[2990]:  failed!
Jun 23 17:56:58 Pi-2A systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Jun 23 17:56:58 Pi-2A systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Jun 23 17:56:58 Pi-2A systemd[1]: Failed to start LSB: DHCP server.

Processing triggers for libc-bin (2.28-10) ...
Processing triggers for systemd (241-7~deb10u7) ...
[  OK  ] DietPi-Software | APT install for: hostapd isc-dhcp-server iptables libnl-3-200
[ SUB1 ] DietPi-Set_hardware > wifimodules (enable)
[ INFO ] DietPi-Set_hardware | Please wait, enabling WiFi modules...
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: iw wireless-tools crda wpasupplicant
[  OK  ] DietPi-Set_hardware | All required APT packages are already installed.
[  OK  ] wifimodules enable | Completed
[  OK  ] DietPi-Software | systemctl unmask hostapd
[ SUB1 ] DietPi-Services > stop
[... 19 lines cut ...]
[  OK  ] DietPi-Software | Added setting ieee80211n=1 to end of file /etc/hostapd/hostapd.conf
[  OK  ] DietPi-Software | systemctl daemon-reload
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : isc-dhcp-server
[  OK  ] DietPi-Services | dietpi_controlled : nmbd
[  OK  ] DietPi-Services | dietpi_controlled : smbd
[  OK  ] DietPi-Services | dietpi_controlled : syncthing
[  OK  ] DietPi-Services | dietpi_controlled : hostapd
[  OK  ] DietPi-Services | dietpi_controlled : webmin
[  OK  ] DietPi-Services | dietpi_controlled : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : isc-dhcp-server
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : syncthing
[FAILED] DietPi-Services | restart : hostapd
[  OK  ] DietPi-Services | restart : webmin
[  OK  ] DietPi-Services | restart : cron
`

Joulinar, 2021-06-23:
pls restart your system and check status of your services afterwards `dietpi-services status`

psi5asp, 2021-06-23:
I re-installed and the issue went away - so it was just some config conflict/issue. All ok now. Thanks

closed 2021-06-23 by psi5asp