MichaIng/DietPi #4507: DietPi-Software | Add support for Time Machine backups via Samba

Camry2731, 2021-06-24:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Adding "native" support for macOS Time Machine backups to Samba via Avahia and vfs objects. Looking at the current solutions, none of them seem to be able to hook into Time Machine (unless I'm missing something).

#### Describe the solution you'd like:

- I would imagine it should be it's own share in order to prevent conflicts with the main share. Since not everyone would need it out of the gate, potentially having it be an option that prompts you if you want that feature enabled or not.

#### Describe alternatives you've considered:

- Upon some general searching, Netatalk seems to be another option as well, but looks like it's not the recommended option and hasn't been updated since December 20, 2018.

MichaIng, 2021-06-24:
Many thanks for your request.

The Avahi daemon is already available as install option, the VFS plugins can be installed via:
```sh
apt install samba-vfs-modules
```
Based on the file list, all modules you listed are included, also Netatalk: https://packages.debian.org/buster/amd64/samba-vfs-modules/filelist

Those can be enabled via `/etc/smb/smb.conf`:
```
[<share>]
vfs objects = fruit catia streams_xattr
```

I'm currently not sure where/how to implement this. Currently for the Samba server a default share for `/mnt/dietpi_userdata` is setup, which should definitely not be used as time machine target. Adding a whiptail dialogue to `dietpi-software` to enable an additional time machine share e.g. `/mnt/time_machine` seems too much, taking into account the uncountable alternative uses and plugins that exist. That would be better placed in a separate script then, like a file server management script or so.

But to address it quickly, we can add instruction to our docs: https://dietpi.com/docs/software/file_servers/#samba
An additional tab "Setup for time machine backups" or so?

Camry2731, 2021-06-24:
I do agree with what your saying, and maybe more input just to see if there are any other opinions. Having it on the wiki would be helpful, as just when I was searching I found 10 different configurations for the smb.conf file that is supposed to make it work, so having a definitive answer would be nice. 

In that case then, is there an alternative software out there where it could be standalone to have as an option added to DietPi Software for those who don't want to mess with the smb.conf, or do you feel like this would be enough?

MichaIng, 2021-06-24:
You mean some kind of dedicated time machine backup target/share? Not sure. When searching the web, I find indeed often this Avahi daemon + Samba combination or Netatalk. Hmm Netatalk is dedicated for macOS features, so in this case setting up a Time Machine backup seems reasonable:
```sh
dietpi-software install 152 # Avahi daemon
apt install netatalk
useradd -rmU -d /mnt/dietpi_userdata/timemachine -p '<yourPassword>' timemachine
cat << '_EOF_' >> /etc/netatalk/afp.conf
[TimeMachine]
# is this machine a time machine?
time machine = yes
# directory for time machine data on server
path = /mnt/dietpi_userdata/timemachine
# the max size of the data folder (in Mb)
vol size limit = 980000
# users with access to time machine
valid users = timemachine
_EOF_
```
and then on macOS enter username "timemachine" and the chosen password on the prompt. The server should be broadcasted via Avahi as `DietPi.local`.

Not sure if there are other useful features to setup by default, but a backup store on a different machine sounds reasonable to me.

bt61, 2021-07-27:
@Camry2731    I am not a very savvy user, so I don't know if there are pros or cons with Avahia and vfs, but I found this guide to setting up TimeMachine over samba share and it has been working well over the network with my DietPi home server for months: https://manjaro.site/how-to-configure-time-machine-to-backup-to-samba-shared-folder/