MichaIng/DietPi #4509: DietPi-Software | Komga: Startup fails on ARMv6

MichaIng, 2021-06-24:
Due to missing included SQLite library compiled for this architecture: https://github.com/gotson/komga/issues/398

Joulinar, 2021-06-24:
nope, installing SQLite `apt install sqlite3`, didn't solve anything. It still behave same.

MichaIng, 2021-06-24:
Then we need to wait for the external fix. I offered to compile the missing library. Should work when being done on any Raspbian system.

MichaIng, 2021-07-08:
Would be great if someone could test this one on an RPi 1 or Zero:
```sh
curl -sSfL https://dietpi.com/downloads/binaries/komga-0.106.0-withArmV6.jar -o /mnt/dietpi_userdata/komga/komga.jar
systemctl restart komga
```

Joulinar, 2021-07-08:
hmm I tried to install komga again but JAVA installation is failing atm

```
[ INFO ] DietPi-Software | APT install for: ca-certificates-java openjdk-11-jre-headless openjdk-11-jdk-headless, please wait... (1/2)
Setting up ca-certificates-java (20190405) ...
head: cannot open '/etc/ssl/certs/java/cacerts' for reading: No such file or directory
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
dpkg: error processing package ca-certificates-java (--configure):
 installed ca-certificates-java package post-installation script subprocess returned error exit status 1
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
done.
Errors were encountered while processing:
 ca-certificates-java
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2021-07-08:
Ayy, this sounds like #3788 is not solved yet... The error matches the initial one #3182. Strange that it worked fine the last time you tested it 🤔. But the workaround as definitely never applied correctly, so in your previous test, Java 11 must have been installed already: https://github.com/MichaIng/DietPi/commit/0d54f6ca5334ca30ebfc0b2ac660fdcc4b6e0e5d#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5R6471-R6472
> ```sh
> local version=11
> (( $G_DISTRO < 5 || G_HW_ARCH == 1 )) && version=8
> ```

The missing `$` so that `G_HW_ARCH` was literally checked against `1`.

Joulinar, 2021-07-08:
Yes it Java installation was working last time. Not sure what was the difference. Usually I reset all my systems before testing. 🤔

MichaIng, 2021-07-08:
Mysterious... I wonder whether it causes any issue when just removing this `jks-keystore` call on package config. `/etc/ssl/certs/java/cacerts` btw seems to be a file, not a directory and probably it's expected to not exist on first install.

MichaIng, 2021-07-08:
```sh
        if dpkg --compare-versions "$2" lt "20180516"; then
            if [ -e /etc/ssl/certs/java/cacerts \
                 -a "$(head -c4 /etc/ssl/certs/java/cacerts)" != "$(echo -en '\xfe\xed\xfe\xed')" ]; then
                check_proc
                convert_pkcs12_keystore_to_jks
            fi
        fi
```
This causes the fist error, which can be ignored:
- The `lt` operator treats a non-installed prior version always as the lower one.
- `lt-nl` would be correct here to treat a non-installed prior version a newer, and hence not enter the condition.
- Next, the test command expands the command substitution with `head` inside regardless whether the first included condition matched (the file existence). Correct would be:
    ```sh
            if [ -e /etc/ssl/certs/java/cacerts ] \
                 && [ "$(head -c4 /etc/ssl/certs/java/cacerts)" != "$(echo -en '\xfe\xed\xfe\xed')" ]; then
    ```
    So that the second condition is wrapped into an own `test`/`[` call and then only reached and expanded when the first succeeded.
- Interesting that such a blunder (two of them) survived so long in this script. I'm gonna report it to the Debian bug tracker

But that has all nothing to do with the ARMv6 install. This doesn't help, does it?
```sh
G_CONFIG_INJECT 'cacerts_updates=' 'cacerts_updates=no' /etc/default/cacerts
dpkg --configure -a
```

Joulinar, 2021-07-09:
No it doesn't change anything. 

```
root@DietPi1:/etc/ssl/certs/java# G_CONFIG_INJECT 'cacerts_updates=' 'cacerts_updates=no' /etc/default/cacerts
[  OK  ] G_CONFIG_INJECT | Setting in /etc/default/cacerts adjusted: cacerts_updates=no
root@DietPi1:/etc/ssl/certs/java# dpkg --configure -a
Setting up ca-certificates-java (20190405) ...
head: cannot open '/etc/ssl/certs/java/cacerts' for reading: No such file or directory
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
dpkg: error processing package ca-certificates-java (--configure):
 installed ca-certificates-java package post-installation script subprocess returned error exit status 1
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

updates of cacerts keystore disabled.
done.
Errors were encountered while processing:
 ca-certificates-java
root@DietPi1:/etc/ssl/certs/java
```

Still the issue with armv6 I guess. Strange that I was able to have it installed once.

```
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
```

MichaIng, 2021-07-26:
Both issues in the `postinst` script have been fixed (exactly as, but before I suggested above 😄) with: https://salsa.debian.org/java-team/ca-certificates-java/-/merge_requests/5/diffs#5355c957e917efca2b0eb20fca872e458c13e0cd_0_153
It solves a few other issues as well, e.g. a circular dependency which forced us to rerun the install two times, at least on ARM.

But the gratest news is, if I checked everything correctly, that `cacerts_updates=no` in `/etc/default/cacerts` is now respected on fresh install as well as upgrade, which allows us to easily disable it before installed the package on ARMv6. The problem is that it was not yet merged even into Bullseye.

I'll add a hacky workaround into dietpi-software now to remove the `postinst` script after the first call fails on ARMv6.

MichaIng, 2021-08-04:
Java install on ARMv6 fixed: #4593
Now trying Komga.

MichaIng, 2021-08-04:
...okay not fixed... I mean the install goes through, but it's the `java` executable itself which fails:
```sh
2021-08-04 18:54:10 root@Zero:/tmp# java
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
```
No change with Debians/Raspbians OpenJDK 11 build.

MichaIng, 2021-08-04:
Interesting: https://packages.debian.org/buster/openjdk-11-jre-zero
This works with the Java 9 and 10 versions, shipped by Raspbian: http://raspbian.raspberrypi.org/raspbian/pool/main/o/openjdk-10/
Sadly for Java 11 and 17, no zero builds are available.

But also, since the `java` executable must be called with the `-zero` command line option, to use this "additional" VM, I think it is not very applicable as a general solution for us. Many software might internally call Java and don't know about that switch. We'd need a build which has the "Zero" VM only, so that it must be used.

Luckily Java 8 is still shipped, even on Raspbian Bullseye.

MichaIng, 2021-08-04:
Back to the initial workaround: https://github.com/MichaIng/DietPi/commit/5b280e1c18c64f7ca3de5f3baf2e2d502e8ba407

Next is to try 3rd party builds: https://github.com/adoptium/temurin17-binaries/releases/
Adoptium is the rebranded AdoptOpenJDK: https://github.com/AdoptOpenJDK/openjdk17-binaries/releases
Or Zulu, but with ARM32hf only for up to Java 13: https://cdn.azul.com/zulu/bin/

For now it works with Java 8, testing Komga now.

MichaIng, 2021-08-04:
Tested the Komga jar with included SQLite library for ARMv6: https://github.com/MichaIng/DietPi/issues/4509#issuecomment-876343989
Sadly is is built against glibc v2.33 which is not even shipped on Bullseye (v2.31), so not applicable in our case. Should be however not a big deal to build it against an older version, like v2.28 to gain Buster compatibility.

MichaIng, 2021-08-06:
New binary up and now it works fine on ARMv6 with Java 8 (latest DietPi dev branch): https://github.com/MichaIng/DietPi/issues/4509#issuecomment-876343989

Joulinar, 2021-08-06:
Can I simply run a test using `dev` branch?

MichaIng, 2021-08-06:
Yes, but the new jar file still needs to be downloaded and a restart done to make it work. The Java install however should be fixed.

Joulinar, 2021-08-06:
Looking good now

```
root@DietPi1:~# systemctl status komga.service
● komga.service - Komga (DietPi)
   Loaded: loaded (/etc/systemd/system/komga.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-08-06 15:46:37 CEST; 11min ag
o
 Main PID: 3303 (java)
    Tasks: 48 (limit: 991)
   CGroup: /system.slice/komga.service
           └─3303 /usr/bin/java -Xmx200m -jar komga.jar
```

MichaIng, 2021-08-08:
New version released with ARMv6 support added: https://github.com/gotson/komga/releases/tag/v0.115.1

Joulinar, 2021-08-09:
jup it's working on `dev v7.5.-1`

MichaIng, 2021-08-09:
Jep there has been another release in the meantime 🙂.

closed 2021-08-20 by MichaIng