MichaIng/DietPi #4511: mariadb broken after update to 7.2.3

Phil1988, 2021-06-25:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version 
stretch

- Kernel version
Linux DietPi 4.9.0-16-amd64 #1 SMP Debian 4.9.272-1 (2021-06-21) x86_64 GNU/Linux

- SBC model |
Virtual Machine (x86_64)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
Update from 7.1.2 to 7.2.3

#### Actual behaviour
mariadb doesnt start.
At bootup I get:
`[FAILED] Dietpi-Services start : mariadb`

#### Extra details
Checked the ownership for mysql files but everything is ok

```
root@DietPi:~# systemctl restart mariadb.service
Job for mariadb.service failed because a fatal signal was delivered to the control process.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status mariadb.service
● mariadb.service - MariaDB 10.1.48 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: activating (start) since Thu 2021-06-24 19:58:45 CEST; 846ms ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 11498 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`
cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP
_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 11493 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (c
ode=exited, status=0/SUCCESS)
  Process: 11491 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (co
de=exited, status=0/SUCCESS)
 Main PID: 11588
   Status: "InnoDB: Error: Database page corruption on disk or a failed file read of tablespa
ce ./ibdata1 page  [page id: space=0, page number=5]. You may have to recover from a backup."
    Tasks: 13 (limit: 19660)
   CGroup: /system.slice/mariadb.service

Jun 24 19:58:45 DietPi systemd[1]: Starting MariaDB 10.1.48 database server...
Jun 24 19:58:45 DietPi mysqld[11588]: 2021-06-24 19:58:45 140479465110912 [Note] /usr/sbin/my
sqld (mysqld 10.1.48-MariaDB-0+deb9u2) starting as process 11588 ...
Jun 24 19:58:45 DietPi systemd[1]: mariadb.service: Main process exited, code=killed,
 status=6/ABRT
```

I hope you have ideas :)

Joulinar, 2021-06-25:
Hi,

pls provide following information

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

MichaIng, 2021-06-25:
Additionally to verify whether there is actual filesystem corruption:
```
dmesg -l emerg,alert,crit,err
```

Phil1988, 2021-06-25:
hey,

the systemctl restart mariadb.service is already there, for the rest:

```
root@DietPi:~# journalctl -u mariadb
-- Logs begin at Fri 2021-06-25 09:29:44 CEST, end at Fri 2021-06-25 14:04:49 CE
ST. --
Jun 25 09:29:55 DietPi systemd[1]: Starting MariaDB 10.1.48 database server...
Jun 25 09:29:56 DietPi mysqld[1157]: 2021-06-25  9:29:56 140653996760448 [Note]
/usr/sbin/mysqld (mysqld 10.1.48-MariaDB-0+deb9u2) starting as process 1157 ...
Jun 25 09:29:56 DietPi systemd[1]: mariadb.service: Main process exited,
 code=killed, status=6/ABRT
Jun 25 09:29:56 DietPi systemd[1]: Failed to start MariaDB 10.1.48 datab
ase server.
Jun 25 09:29:56 DietPi systemd[1]: mariadb.service: Unit entered failed
state.
Jun 25 09:29:56 DietPi systemd[1]: mariadb.service: Failed with result '
signal'.
Jun 25 09:30:01 DietPi systemd[1]: mariadb.service: Service hold-off time over,
scheduling restart.
Jun 25 09:30:01 DietPi systemd[1]: Stopped MariaDB 10.1.48 database server.
Jun 25 09:30:01 DietPi systemd[1]: Starting MariaDB 10.1.48 database server...
Jun 25 09:30:02 DietPi mysqld[1557]: 2021-06-25  9:30:02 139890696887680 [Note]
/usr/sbin/mysqld (mysqld 10.1.48-MariaDB-0+deb9u2) starting as process 1557 ...
Jun 25 09:30:02 DietPi systemd[1]: mariadb.service: Main process exited,
 code=killed, status=6/ABRT
Jun 25 09:30:02 DietPi systemd[1]: Failed to start MariaDB 10.1.48 datab
ase server.
Jun 25 09:30:02 DietPi systemd[1]: mariadb.service: Unit entered failed state.
Jun 25 09:30:02 DietPi systemd[1]: mariadb.service: Failed with result 'signal'.
Jun 25 09:30:07 DietPi systemd[1]: mariadb.service: Service hold-off time over, scheduling restart.
Jun 25 09:30:07 DietPi systemd[1]: Stopped MariaDB 10.1.48 database server.
Jun 25 09:30:07 DietPi systemd[1]: Starting MariaDB 10.1.48 database server...
Jun 25 09:30:08 DietPi mysqld[1678]: 2021-06-25  9:30:08 140481425399168 [Note] /usr/sbin/mysqld (mysqld 10.1.48-MariaDB-0+deb9u2) starting as process 1678 ...
Jun 25 09:30:08 DietPi systemd[1]: mariadb.service: Main process exited, code=killed, status=6/ABRT
Jun 25 09:30:08 DietPi systemd[1]: Failed to start MariaDB 10.1.48 database server.
Jun 25 09:30:08 DietPi systemd[1]: mariadb.service: Unit entered failed state.
Jun 25 09:30:08 DietPi systemd[1]: mariadb.service: Failed with result 'signal'.
Jun 25 09:30:13 DietPi systemd[1]: mariadb.service: Service hold-off time over, scheduling restart.
Jun 25 09:30:13 DietPi systemd[1]: Stopped MariaDB 10.1.48 database server.
Jun 25 09:30:13 DietPi systemd[1]: Starting MariaDB 10.1.48 database server...
Jun 25 09:30:14 DietPi mysqld[1796]: 2021-06-25  9:30:14 140077151415680 [Note] /usr/sbin/mysqld (mysqld 10.1.48-MariaDB-0+deb9u2) starting as process 1796 ...
Jun 25 09:30:14 DietPi systemd[1]: mariadb.service: Main process exited, code=killed, status=6/ABRT
Jun 25 09:30:14 DietPi systemd[1]: Failed to start MariaDB 10.1.48 database server.
[...]
```

```
cat /var/log/mysql/error.log
InnoDB: End of page dump
2021-06-25 14:06:28 7ff8b21b7d80 InnoDB: uncompressed page, stored checksum in field1 1420198940, calculated checksums for field1: crc32 3306546772, innodb 1420198940, none 3735928559, stored checksum in field2 2279621922, calculated checksums for field2: crc32 3306546772, innodb 1848301123, none 3735928559, page LSN 13 2577993519, low 4 bytes of LSN at page end 2577993499, page number (if stored to page already) 5, space id (if created with >= MySQL-4.1.1 and stored already) 0
InnoDB: page type 7 meaning TRX_SYS
InnoDB: Page may be a transaction system page
2021-06-25 14:06:28 140706116763008 [Note] InnoDB: It is also possible that your operating system has corrupted its own file cache and rebooting your computer removes the error. If the corrupt page is an index page. You can also try to fix the corruption by dumping, dropping, and reimporting the corrupt table. You can use CHECK TABLE to scan your table for corruption. Please refer to http://dev.mysql.com/doc/refman/5.6/en/forcing-innodb-recovery.html for information about forcing recovery.
2021-06-25 14:06:28 140706116763008 [ERROR] InnoDB: Ending processing because of a corrupt database page.
2021-06-25 14:06:28 7ff8b21b7d80  InnoDB: Assertion failure in thread 140706116763008 in file ha_innodb.cc line 21944
InnoDB: We intentionally generate a memory trap.
InnoDB: Submit a detailed bug report to https://jira.mariadb.org/
InnoDB: If you get repeated assertion failures or crashes, even
InnoDB: immediately after the mysqld startup, there may be
InnoDB: corruption in the InnoDB tablespace. Please refer to
InnoDB: http://dev.mysql.com/doc/refman/5.6/en/forcing-innodb-recovery.html
InnoDB: about forcing recovery.
210625 14:06:28 [ERROR] mysqld got signal 6 ;
This could be because you hit a bug. It is also possible that this binary
or one of the libraries it was linked against is corrupt, improperly built,
or misconfigured. This error can also be caused by malfunctioning hardware.

To report this bug, see https://mariadb.com/kb/en/reporting-bugs
[... 33 lines cut ...]
/usr/sbin/mysqld(+0x834f41)[0x55afb9f1ff41]
/usr/sbin/mysqld(_Z24ha_initialize_handlertonP13st_plugin_int+0x62)[0x55afb9cabdc2]
/usr/sbin/mysqld(+0x43bba5)[0x55afb9b26ba5]
/usr/sbin/mysqld(_Z11plugin_initPiPPci+0x7da)[0x55afb9b2820a]
/usr/sbin/mysqld(+0x386733)[0x55afb9a71733]
/usr/sbin/mysqld(_Z11mysqld_mainiPPc+0x4fb)[0x55afb9a7608b]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7ff8b09832e1]
/usr/sbin/mysqld(_start+0x2a)[0x55afb9a6a06a]
[... cut ...] at https://mariadb.com/kb/en/how-to-produce-a-full-stack-trace-for-mysqld/ contains
information that should help you find out what is causing the crash.

```

```
root@DietPi:~# readlink /var/lib/mysql
/mnt/dietpi_userdata/mysql
root@DietPi:~# readlink -f /var/lib/mysql
/mnt/dietpi_userdata/mysql
```

```
root@DietPi:~# dmesg -l emerg,alert,crit,err
[    0.939447] sd 0:0:0:0: [sda] Assuming drive cache: write through
[    1.531565] piix4_smbus 0000:00:07.3: SMBus Host Controller not enabled!
```

MichaIng, 2021-06-25:
Okay please try the following:
1. Assure that you have a backup, or otherwise copy `/mnt/dietpi_userdata/mysql` to somewhere else as backup.
2. Enable first level database recovery, then try to start MariaDB:
    ```sh
    echo -e '[mysqld]\ninnodb_force_recovery=1' > /etc/mysql/mariadb.conf.d/recovery.cnf
    systemctl restart mariadb
    ```
3. If this did not work, increment the recovery level by one with each iteration, until MariaDB starts up:
    ```sh
    echo -e '[mysqld]\ninnodb_force_recovery=2' > /etc/mysql/mariadb.conf.d/recovery.cnf
    systemctl restart mariadb
    ```
4. As fast as it worked, try to do a full database dump:
    ```sh
    mysqldump -A > /mnt/dietpi_userdata/mysqldump.sql
    ```
5. Then check and try to fix the database:
    ```sh
    mysqlcheck -A # check only
    mysqlcheck -A -r # check and repair
    ```

Which software uses MariaDB in your case?

Phil1988, 2021-06-25:
I tired step 3 untill iteration 20... I think this is going nowhere :(

```
root@DietPi:~# echo -e '[mysqld]\ninnodb_force_recovery=20' > /etc/mysql/mariadb.conf.d/99-recovery.cnf
root@DietPi:~# systemctl restart mariadb
Job for mariadb.service failed because a fatal signal was delivered to the control process.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
```

So I stoped and didnt do step 4 or 5.

I am using nextcloud.

As I am using a VM I did a snapshot before the update.

So if I can help you finding the reason to prevent this for others and make the update more stable, please let me know what to do :)

Joulinar, 2021-06-25:
To be honest, this has nothing to do with DietPi update function. You got a corrupted database file. This could multiple reasons. At least on a VM we probably could exclude file system issue. But maybe unclean poweroff could lead to this. 

What you could try is to restore the snapshot, perform an `apt update && apt upgrade`, check if all is working fine and if this is the case perform DietPi update.

Phil1988, 2021-06-25:
OK, I will restore the snapshot, test and feedback :)

Phil1988, 2021-06-25:
You are right Joulinar!

I did restore the snapshot, performed an `apt update && apt upgrade`, checked if all is working fine and performed a DietPi update.

Everything working now.

And you were right with your idea. There really was a poweroff here in my region so that could be the cause.

Sorry guys for taking your time and thanks for helping me! :)

Joulinar, 2021-06-25:
you are welcome, better to check on such issues. 😃

MichaIng, 2021-06-25:
> `'[mysqld]\ninnodb_force_recovery=20'`

Valid levels are only 1-6 😉. Good idea to recover a backup/snapshort, I totally forgot to ask about that, great that it worked 👍.

Phil1988, 2021-06-26:
> 
> 
> you are welcome, better to check on such issues. 😃
> 
> Btw: you should think of migrating to Debian Buster as you still running Stretch. Sooner or later Nextcloud might gonna drop support for old MariaDB version 10.1. As well next Debian version Bullseye will be released soon and Stretch become oldold.

"Migration" sounds so nice and easy :D 

I think this would mean to setup everything from ~~stretch~~ scratch right?
So start over with a new VM and do all the settings again.

For me that would mean 2 weeks of work ;D
I will give this a try, also I think that it will most likely fail:
https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

But having snapshots as a backup makes it possible to give it a try

MichaIng, 2021-06-26:
> I will give this a try, also I think that it will most likely fail

TL;DR: Definitely worth to give it a try!

Based on my Wheezy => Jessie => Stretch distro upgrade experiences, I would have assumed to same, that it most likely fails at some point, but I am surprised that in every case where we got a feedback about the upgrade Stretch => Buster, it went well. I personally upgrade my systems directly to the new testing suite, after a release, when there is not difference between release and testing yet, so cannot say something about Stretch => Buster => Bullseye, but it seems to have improved a lot, hence became a quite reliable alternative to setup from scratch 🙂.

Joulinar, 2021-06-26:
Yes you could update from Stretch to Buster without performing a reinstall. Just have a look to following forum post https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

But we can't give a guarantee 😉 
Therefore do a backup before

Phil1988, 2021-07-01:
Just as a feedback:

MichaIng, 2021-07-01:
Awesome, many thanks and great to know. I feel quite confident that we can guide Stretch users through a distro upgrade when dropping Stretch support. Plan is to migrate Stretch systems to a "stretch" branch first, to block DietPi updates to newest incompatible version, which was one purpose of the updater change with DietPi v7.0. There we add a script which guides or performs an interactive distro upgrade (interactive, so that all APT questions need to be answered/confirmed interactively) and, when succeeded, migrates them back to the master branch to update to latest DietPi version.

Joulinar, 2021-07-01:
thx for sharing. So basically you are ready to update to Bullseye that might be released end of July 2021 🤣

MichaIng, 2021-07-01:
One thing we should take care of: When PHP was installed, Ondrej's repo should be removed revert to the PHP version from the main Debian repository. There are too many reports of users, when installing the PHP meta packages, ending up with PHP 8 (unintentionally), two parallel PHP instances, and will then get additionally PHP 8.1 once release 😄.

@Phil1988 
That might be reasonable for you as well. If you have MariaDB installed, I guess PHP as well? It's not necessary, as long as it's taken care to keep using/installing packages like `php7.3-fpm` or upgrading (and changing the webserver config accordingly) intentionally to PHP 7.4 or PHP 8.0. But when installing packages like `php-fpm`, it will always pull the newest version, like `php8.0-fpm` currently and all related PHP 8.0 dependencies. Again okay when migrating PHP applications and webserver, but it's not yet 100% supported by all software titles, our phpBB instance is still running into the one or the other PHP 8 related error, especially when extensions are used 😉.
```sh
rm /etc/apt/sources.list.d/dietpi-php.list
rm /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm /etc/apt/preferences.d/dietpi-php
rm /etc/apt/preferences.d/dietpi-openssl
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```
It is possible that the `apt upgrade` asks whether a package downgrade should be done. This should be confirmed then to avoid failures of G_AGUG calls which are cancelled on such APT questions.

Phil1988, 2021-07-02:
currently I am using:
redis-server
apache2
mariadb
nextcloud
mosquitto
nodered
pihole
blynkserver
home-assistant

I have not seen a problem since the upgrade so far.

I already did a `apt upgrade` before without the commands you posted.
The php versions pulled have been 7.3.

Should I perform 
```
rm /etc/apt/sources.list.d/dietpi-php.list
rm /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm /etc/apt/preferences.d/dietpi-php
rm /etc/apt/preferences.d/dietpi-openssl
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```
anyways?

and to @Joulinar:
I am happily here to test any upgrade processes to bullseye on my system for you :D

Joulinar, 2021-07-02:
let's wait for @MichaIng before doing any further steps. But in theory you could use same procedure to update to Bullseye. It's already possible and there shouldn't be any surprises as Bullseye is already in a freeze mode and there is no major development happen anymore.

MichaIng, 2021-07-05:
Jep, upgrade to Bullseye works the same way. If you do so, Ondrej's repository needs to be removed and PHP and the webserver need to be reinstalled, as Bullseye does not ship PHP7.3 anymore but PHP7.4 instead. Also on Buster, just as a pre-caution against unintended PHP upgrades, I'd do the steps to remove Ondrej's repository: This was only added to maintain support for Debian Stretch, else most PHP application would have needed to be disabled, as most of them dropped support for PHP7.2 and below for a while.

Phil1988, 2021-07-09:
Well I am not able to do it as I am not confident enough to try and error :D 

1. I am still not sure if I should perform
```
rm /etc/apt/sources.list.d/dietpi-php.list
rm /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm /etc/apt/preferences.d/dietpi-php
rm /etc/apt/preferences.d/dietpi-openssl
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

2. I am not sure, where I added ondrejs php repo (but I am pretty sure that I did that).
In theory I should do something like `ppa-purge ppa:ondrej/php7-7.3` right?
As I am not sure, I tried to search the used ondrejs repositorys with `apt-cache policy | grep ondrej` whit 0 results.

It might also work to just `sudo add-apt-repository --remove ppa:ondrej/php` but I am not sure.
So it would be really kind if you can tell me, what would be the right way to do so.

3. For the update to Bulleye:
My limited knowledge, tells me, I should do something like this ( :D ) :
```
sed -i 's|buster|bullseye|' /etc/apt/sources.list
sed -i 's|bullseye/updates|bullseye-security|' /etc/apt/sources.list
apt update
apt upgrade
apt full-upgrade
apt autopurge
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
```

4. To install PHP and the webserver, should I just 
apt-get -s install php7.4

and install the webserver from the dietpi-software?

Sorry for those questions and many thanks for your help!

Joulinar, 2021-07-09:
You already have Apache2 installed. Therefore no need to install any other web server. For the update to Bullseye, it should looks like this

```
sed -i 's/buster/bullseye/' /etc/apt/sources.list
sed -i 's/buster/bullseye/' /etc/apt/sources.list.d/*.list
# Also check other /etc/apt/sources.list.d/*.list files that might need to be updated
apt update
apt upgrade
apt full-upgrade
apt autopurge
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
```

it needs to be a `/` instead of `|` 😉 

But first, let's ask @MichaIng if he could help to ensure using correct PHP package 😃

MichaIng, 2021-07-09:
And yes, please do the `dietpi-php.list` removal in any case with the steps I provided. PPA stuff is more Ubuntu stuff, basically it depends on how the repository was added. Since we add it manually by creating this files, it also needs to be removed manually by removing the created files 😉.

Phil1988, 2021-07-09:
Hmmm... well the ppa stuff would be handy, as I dont know, where the created php (ondrej) files are sitting :D 

I did the following so far:
(the dietpi-php.list stuff)

```
root@DietPi:~# ncc maintenance:mode --on
Maintenance mode enabled
root@DietPi:~# rm /etc/apt/sources.list.d/dietpi-php.list
root@DietPi:~# rm /etc/apt/trusted.gpg.d/dietpi-php.gpg
root@DietPi:~# rm /etc/apt/preferences.d/dietpi-php
rm: das Entfernen von '/etc/apt/preferences.d/dietpi-php' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
root@DietPi:~# rm /etc/apt/preferences.d/dietpi-openssl
root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt-cache clean
 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi:~# apt update
Holen:1 https://deb.debian.org/debian buster InRelease [122 kB]
Holen:2 https://repo.mosquitto.org/debian buster InRelease [12,4 kB]
Holen:3 https://deb.debian.org/debian buster-updates InRelease [51,9 kB]
Holen:4 https://deb.debian.org/debian-security buster/updates InRelease [65,4 kB]
Holen:5 https://deb.debian.org/debian buster-backports InRelease [46,7 kB]
Holen:6 https://repo.mosquitto.org/debian buster/main i386 Packages [34,0 kB]
Holen:7 https://repo.mosquitto.org/debian buster/main amd64 Packages [34,0 kB]
Holen:8 https://repo.mosquitto.org/debian buster/main all Packages [3.988 B]
Holen:9 https://deb.debian.org/debian buster/non-free amd64 Packages [87,7 kB]
Holen:10 https://deb.debian.org/debian buster/non-free i386 Packages [76,2 kB]
Holen:11 https://deb.debian.org/debian buster/contrib i386 Packages [46,0 kB]
Holen:12 https://deb.debian.org/debian buster/main i386 Packages [7.863 kB]
Holen:13 https://deb.debian.org/debian buster/main amd64 Packages [7.907 kB]
Holen:14 https://deb.debian.org/debian buster/contrib amd64 Packages [50,1 kB]
Holen:15 https://deb.debian.org/debian buster-updates/main i386 Packages [15,2 kB]
Holen:16 https://deb.debian.org/debian buster-updates/main amd64 Packages [15,2 kB]
Holen:17 https://deb.debian.org/debian-security buster/updates/non-free amd64 Packages [556 B]
Holen:18 https://deb.debian.org/debian-security buster/updates/non-free i386 Packages [556 B]
Holen:19 https://deb.debian.org/debian-security buster/updates/main i386 Packages [292 kB]
Holen:20 https://deb.debian.org/debian-security buster/updates/main amd64 Packages [293 kB]
Holen:21 https://deb.debian.org/debian buster-backports/non-free amd64 Packages [34,4 kB]
Holen:22 https://deb.debian.org/debian buster-backports/main i386 Packages [480 kB]
Holen:23 https://deb.debian.org/debian buster-backports/contrib amd64 Packages [9.704 B]
Holen:24 https://deb.debian.org/debian buster-backports/non-free i386 Packages [23,5 kB]
Holen:25 https://deb.debian.org/debian buster-backports/contrib i386 Packages [9.020 B]
Holen:26 https://deb.debian.org/debian buster-backports/main amd64 Packages [486 kB]
Es wurden 18,1 MB in 4 s geholt (4.160 kB/s).
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Alle Pakete sind aktuell.
root@DietPi:~# apt upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
```

The bullseye upgrade:

```
root@DietPi:~# sed -i 's/buster/bullseye/' /etc/apt/sources.list
root@DietPi:~# sed -i 's/buster/bullseye/' /etc/apt/sources.list.d/*.list
root@DietPi:~# apt update
Holen:1 https://deb.debian.org/debian bullseye InRelease [150 kB]
Holen:2 https://deb.debian.org/debian bullseye-updates InRelease [40,1 kB]
Ign:3 https://repo.mosquitto.org/debian bullseye InRelease
Ign:4 https://deb.debian.org/debian-security bullseye/updates InRelease
Holen:5 https://deb.debian.org/debian bullseye-backports InRelease [37,0 kB]
[... 2608 lines cut ...]
Entfernen von python-talloc:amd64 (2.1.14-2) ...
Entfernen von python (2.7.16-1) ...
Entfernen von libpython-stdlib:amd64 (2.7.16-1) ...
Entfernen von libstdc++-8-dev:amd64 (8.3.0-6) ...
Entfernen von linux-image-4.9.0-16-amd64 (4.9.272-1) ...
/etc/kernel/postrm.d/initramfs-tools:
update-initramfs: Deleting /boot/initrd.img-4.9.0-16-amd64
/etc/kernel/postrm.d/zz-update-grub:
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-5.10.0-7-amd64
Found initrd image: /boot/initrd.img-5.10.0-7-amd64
Found linux image: /boot/vmlinuz-4.19.0-17-amd64
Found initrd image: /boot/initrd.img-4.19.0-17-amd64
done
Entfernen von mariadb-server (1:10.3.29-0+deb10u1) ...
Entfernen von mariadb-server-10.3 (1:10.3.29-0+deb10u1) ...
Entfernen von mariadb-client-10.3 (1:10.3.29-0+deb10u1) ...
Entfernen von mariadb-client-core-10.3 (1:10.3.29-0+deb10u1) ...
Entfernen von mariadb-server-core-10.3 (1:10.3.29-0+deb10u1) ...
Entfernen von python-minimal (2.7.16-1) ...
(Lese Datenbank ... 49202 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../00-cron_3.0pl1-137_amd64.deb ...
Entpacken von cron (3.0pl1-137) über (3.0pl1-134+deb10u1) ...
Vorbereitung zum Entpacken von .../01-dbus_1.12.20-2_amd64.deb ...
Entpacken von dbus (1.12.20-2) übe

Joulinar, 2021-07-09:
can you check if the database is running `systemctl staus mariadb.service`

What happen if you do `ncc maintenance:mode --off`

Phil1988, 2021-07-10:
MariaDB:
```
root@DietPi:~# systemctl staus mariadb.servic
Unknown command verb staus.
```

Nextcloud:
```
root@DietPi:~# ncc maintenance:mode --off
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occ                                                                                                                                                             urred in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/ne                                                                                                                                                             xtcloud/lib/private/DB/Connection.php:72
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(14                                                                                                                                                             49): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(89                                                                                                                                                             2): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(202): Doctrine\DBAL\Connecti                                                                                                                                                             on->executeQuery('SELECT * FROM `...', Array, Array, NULL)
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilde                                                                                                                                                             r.php(206): OC\DB\Connection->executeQuery('SELECT * FROM `...', Array, Array)
#4 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(217): Doctrin                                                                                                                                                             e\DBAL\Query\QueryBuilder->execute()
#5 /var/www/nextcloud/lib/private/AppConfig.php(345): OC\DB\QueryBuilder\QueryBu                                                                                                                                                             ilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(110): OC\AppConfig->loadConfigVa                                                                                                                                                             lues()
#7 /var/www/nextcloud/lib/private/AppConfig.php(301): OC\AppConfig->getApps()
#8 /var/www/nextcloud/lib/private/legacy/OC_App.php(949): OC\AppConfig->getValue                                                                                                                                                             s(false, 'installed_versi...')
#9 /var/www/nextcloud/lib/private/Server.php(668): OC_App::getAppVersions()
#10 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155)                                                                                                                                                             : OC\Server->OC\{closure}(Object(OC\Server))
#11 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\                                                                                                                                                             AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(P                                                                                                                                                             imple\Container))
#12 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122)                                                                                                                                                             : Pimple\Container->offsetGet('OC\\Memcache\\Fac...')
#13 /var/www/nextcloud/lib/private/ServerContainer.php(156): OC\AppFramework\Uti                                                                                                                                                             lity\SimpleContainer->query('OC\\Memcache\\Fac...', true)
#14 /var/www/nextcloud/lib/private/Server.php(1677): OC\ServerContainer->query('                                                                                                                                                             OC\\Memcache\\Fac...')
#15 /var/www/nextcloud/lib/private/Server.php(1017): OC\Server->getMemCacheFacto                                                                                                                                                             ry()
#16 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155)                                                                                                                                                             : OC\Server->OC\{closure}(Object(OC\Server))
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\                                                                                                                                                             AppFramework\Utility\SimpleContainer [... cut ...]
#27 /var/www/nextcloud/lib/private/Server.php(1324): OC\ServerContainer->query('                                                                                                                                                             OC\\Files\\Node\\H...')
#28 /var/www/nextcloud/lib/base.php(595): OC\Server->boot()
#29 /var/www/nextcloud/lib/base.php(1091): OC::init()
#30 /var/www/nextcloud/console.php(49): require_once('/var/www/nextcl...')
#31 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
```

Joulinar, 2021-07-10:
Sorry there was a typo

`systemctl status mariadb.service`

Phil1988, 2021-07-10:
You are right.. I also didnt see the missing "T".
I did that but that is also not working
```
root@DietPi:~# systemctl status mariadb.service
Unit mariadb.service could not be found.
```

Also pihole does not work correctly after upgrading.
I get a `DNS service not running` message on the status board:
[image]

At least here is something I was able to fix on my own.
Maybe that might help you:

I tried to reconfigure it, but that didnt work:
<details>
  <summary>Didnt work</summary>

```
root@DietPi:~# pihole -r

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [✗] Update local cache of available packages
  Error: Unable to update package cache. Please try "apt-get update"root@DietPi:~#
root@DietPi:~# apt-get update
OK:1 https://deb.debian.org/debian bullseye InRelease
OK:2 https://deb.debian.org/debian bullseye-updates InRelease
OK:3 https://deb.debian.org/debian-security bullseye-security InRelease
OK:4 https://deb.debian.org/debian bullseye-backports InRelease
Ign:5 https://repo.mosquitto.org/debian bullseye InRelease
Fehl:6 https://repo.mosquitto.org/debian bullseye Release
  404  Not Found [IP: 85.119.83.194 443]
Paketlisten werden gelesen… Fertig
E: Das Depot »https://repo.mosquitto.org/debian bullseye Release« enthält keine Release-Datei.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
root@DietPi:~# pihole -r

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [✗] Update local cache of available packages
  Error: Unable to update package cache. Please try "apt-get update"root@DietPi:~# pihole -up
  [i] Checking for updates...
  [i] Pi-hole Core:     up to date
  [i] Web Interface:    up to date
  [i] FTL:              up to date

  [✓] Everything is up to date!
```

</details>

<details>
  <summary>lsof was missing</summary>

```
root@DietPi:~# pihole status
/usr/local/bin/pihole: Zeile 319: lsof: Kommando nicht gefunden.
  [✗] DNS service is NOT listening
root@DietPi:~# ^C
root@DietPi:~# ^C
root@DietPi:~# ^C
root@DietPi:~# ^C
root@DietPi:~#
root@DietPi:~#
root@DietPi:~# sudo apt install lsof
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden NEUEN Pakete werden installiert:
  lsof
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Es müssen 319 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 462 kB Plattenplatz zusätzlich benutzt.
Holen:1 https://deb.debian.org/debian bullseye/main amd64 lsof amd64 4.93.2+dfsg-1.1 [319 kB]
Es wurden 319 kB in 0 s geholt (3.377 kB/s).
Vormals nicht ausgewähltes Paket lsof wird gewählt.
(Lese Datenbank ... 43011 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../lsof_4.93.2+dfsg-1.1_amd64.deb ...
Entpacken von lsof (4.93.2+dfsg-1.1) ...
lsof (4.93.2+dfsg-1.1) wird eingerichtet ...
root@DietPi:~# pihole status
  [✓] DNS service is listening
     [✓] UDP (IPv4)
     [✓] TCP (IPv4)
     [✓] UDP (IPv6)
     [✓] TCP (IPv6)

  [✓] Pi-hole blocking is enabled
root@DietPi:~#
```
</details>

[image]

Joulinar, 2021-07-10:
Yes `isof` was removed as well during update. And in general an issue is `mosquitto`. It doesn't seem to offer Bullseye repository yet. Probably this source list would need to be changed back to Buster. Strange that there are packages removed that are still required. Interesting behaviour.

The database could be more tricky. Can you check what is still installed.
`dpkg -l | grep maria`

MichaIng, 2021-07-10:
MariaDB was remove during full-upgrade, for some reason. The 10.3 server and client packages should have been, but not the meta packages:
```sh
apt install mariadb-server
```

Also the Mosquitto repository does not have a Bullseye suite yet:
```sh
sed -i 's/bullseye/buster/' /etc/apt/sources.list.d/*mosquitto*.list
apt update
```

Phil1988, 2021-07-10:
```
root@DietPi:~# dpkg -l | grep maria
rc  mariadb-client-10.1           10.1.48-0+deb9u2                                   amd64        MariaDB database client binaries
rc  mariadb-client-10.3           1:10.3.29-0+deb10u1                                amd64        MariaDB database client binaries
rc  mariadb-server-10.1           10.1.48-0+deb9u2                                   amd64        MariaDB database server binaries
rc  mariadb-server-10.3           1:10.3.29-0+deb10u1                                amd64        MariaDB database server binaries
```

<details>
  <summary>Installing mariadb</summary>
```
root@DietPi:~# apt install mariadb-server
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  galera-4 libaio1 libconfig-inifiles-perl libdbi-perl libmariadb3 mariadb-client-10.5 mariadb-client-core-10.5 mariadb-common mariadb-server-10.5 mariadb-server-core-10.5 mysql-common rsync socat
Vorgeschlagene Pakete:
  libclone-perl libmldbm-perl libnet-daemon-perl libsql-statement-perl mailx mariadb-test
Empfohlene Pakete:
  libdbd-mariadb-perl | libdbd-mysql-perl libterm-readkey-perl libhtml-template-perl
Die folgenden NEUEN Pakete werden installiert:
  galera-4 libaio1 libconfig-inifiles-perl libdbi-perl libmariadb3 mariadb-client-10.5 mariadb-client-core-10.5 mariadb-common mariadb-server mariadb-server-10.5 mariadb-server-core-10.5 mysql-common rsync socat
0 aktualisiert, 14 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Es müssen 15,7 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 152 MB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Holen:1 https://deb.debian.org/debian bullseye/main amd64 mysql-common all 5.8+1.0.7 [7.464 B]
Holen:2 https://deb.debian.org/debian bullseye/main amd64 mariadb-common all 1:10.5.10-2 [36,0 kB]
Holen:3 https://deb.debian.org/debian bullseye/main amd64 galera-4 amd64 26.4.8-1 [794 kB]
Holen:4 https://deb.debian.org/debian bullseye/main amd64 libdbi-perl amd64 1.643-3+b1 [780 kB]
Holen:5 https://deb.debian.org/debian bullseye/main amd64 libconfig-inifiles-perl all 3.000003-1 [52,1 kB]
Holen:6 https://deb.debian.org/debian bullseye/main amd64 libmariadb3 amd64 1:10.5.10-2 [174 kB]
Holen:7 https://deb.debian.org/debian bullseye/main amd64 mariadb-client-core-10.5 amd64 1:10.5.10-2 [778 kB]
Holen:8 https://deb.debian.org/debian bullseye/main amd64 mariadb-client-10.5 amd64 1:10.5.10-2 [1.493 kB]
Holen:9 https://deb.debian.org/debian bullseye/main amd64 libaio1 amd64 0.3.112-9 [12,5 kB]
Holen:10 https://deb.debian.org/debian bullseye/main amd64 mariadb-server-core-10.5 amd64 1:10.5.10-2 [6.649 kB]
Holen:11 https://deb.debian.org/debian bullseye/main amd64 rsync amd64 3.2.3-4 [333 kB]
Holen:12 https://deb.debian.org/debian bullseye/main amd64 socat amd64 1.7.4.1-3 [370 kB]
Holen:13 https://deb.debian.org/debian bullseye/main amd64 mariadb-server-10.5 amd64 1:10.5.10-2 [4.219 kB]
Holen:14 https://deb.debian.org/debian bullseye/main amd64 mariadb-server all 1:10.5.10-2 [34,5 kB]
Es wurden 15,7 MB in 2 s geholt (7.934 kB/s).
Preconfiguring packages ...
Vormals nicht ausgewähltes Paket mysql-common wird gewählt.
(Lese Datenbank ... 43037 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../00-mysql-common_5.8+1.0.7_all.deb ...
Entpacken von mysql-common (5.8+1.0.7) ...
Vormals nicht ausgewähltes Paket mariadb-common wird gewählt.
Vorbereitung zum Entpacken von .../01-mariadb-common_1%3a10.5.10-2_all.deb ...
Entpacken von mariadb-common (1:10.5.10-2) ...
Vormals nicht ausgewähltes Paket galera-4 wird gewählt.
Vorbereitung zum Entpacken von .../02-galera-4_26.4.8-1_amd64.deb ...
Entpacken von galera-4 (26.4.8-1) ...
Vormals nicht ausgewähltes Paket libdbi-perl:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../03-libdbi-perl_1.643-3+b1_amd64.deb ...
Entpacken von libdbi-perl:amd64 (1.643-3+b1) ...
Vormals nicht ausgewähltes Paket libconfig-inifiles-perl wird gewählt.
Vorbereitung zum Entpacken von .../04-libconfig-inifiles-perl_3.000003-1_all.deb ...
Entpacken von libconfig-inifiles-perl (3.000003-1) ...
Vormals nicht ausgewähltes Paket libmariadb3:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../05-libmariadb3_1%3a10.5.10-2_amd64.deb ...
Entpacken von libmariadb3:amd64 (1:10.5.10-2) ...
Vormals nicht ausgewähltes Paket mariadb-client-core-10.5 wird gewählt.
Vorbereitung zum Entpacken von .../06-mariadb-client-core-10.5_1%3a10.5.10-2_amd64.deb ...
Entpacken von mariadb-client-core-10.5 (1:10.5.10-2) ...
Vormals nicht ausgewähltes Paket mariadb-client-10.5 wird gewählt.
Vorbereitung zum Entpacken von .../07-mariadb-client-10.5_1%3a10.5.10-2_amd64.deb ...
Entpacken von mariadb-client-10.5 (1:10.5.10-2) ...
Vormals nicht ausgewähltes Paket libaio1:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../08-libaio1_0.3.112-9_amd64.deb ...
[... 67 lines cut ...]
Vorbereitung zum Entpacken von .../libfdisk1_2.36.1-7_amd64.deb ...
Entpacken von libfdisk1:amd64 (2.36.1-7) ...
Vormals nicht ausgewähltes Paket fdisk wird gewählt.
Vorbereitung zum Entpacken von .../fdisk_2.36.1-7_amd64.deb ...
Entpacken von fdisk (2.36.1-7) ...
libfdisk1:amd64 (2.36.1-7) wird eingerichtet ...
fdisk (2.36.1-7) wird eingerichtet ...
Trigger für libc-bin (2.31-12) werden verarbeitet ...
```
</details>

mariaDB seems to work now:
```
root@DietPi:~# systemctl start mariadb.servic
Failed to start mariadb.servic.service: Unit mariadb.servic.service not found.
root@DietPi:~# systemctl status mariadb.service
● mariadb.service - MariaDB 10.5.10 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
       Docs: man:mariadbd(8)
             https://mariadb.com/kb/en/l
```

But Nextcloud is still not working.
Do I have to restart/reinstall it?

Joulinar, 2021-07-10:
MariaDB is core, otherwise Nextcloud will not be working at all. 

Let's have a look

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

Phil1988, 2021-07-10:
I just started mariadb now:

```
root@DietPi:~# systemctl start mariadb.service
root@DietPi:~# systemctl status mariadb.service
● mariadb.service - MariaDB 10.5.10 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-07-10 10:18:02 CEST; 1s ago
       Docs: man:mariadbd(8)
             https://mariadb.com/kb/en/library/systemd/
    Process: 29533 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
    Process: 29534 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
    Process: 29537 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
    Process: 29616 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
    Process: 29618 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
   Main PID: 29601 (mariadbd)
     Status: "Taking your SQL requests now..."
      Tasks: 20 (limit: 4666)
     Memory: 126.1M
        CPU: 468ms
     CGroup: /system.slice/mariadb.service
             ├─29601 /usr/sbin/mariadbd
             ├─29619 /bin/bash /etc/mysql/debian-start
             ├─29621 /usr/bin/mysql_upgrade --defaults-extra-file=/etc/mysql/debian.cnf --version-check
             ├─29622 grep -E -v ^(1|@had|ERROR (1051|1054|1060|1061|1146|1347|1348))
             ├─29623 logger -p daemon warn -i -t/etc/mysql/debian-start
             ├─29637 sh -c '/usr/bin/mysql' --defaults-file=/tmp/mysql_upgrade-z6FvG0 --database=mysql --batch --force --silent < /tmp/sql2gQtB0 2>&1
             └─29638 /usr/bin/mysql --defaults-file=/tmp/mysql_upgrade-z6FvG0 --database=mysql --batch --force --silent

Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] InnoDB: 10.5.10 started; log sequence number 59587254717; transaction id 99858462
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] Plugin 'FEEDBACK' is disabled.
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] Server socket created on IP: '127.0.0.1'.
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] Reading of all Master_info entries succeeded
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] Added new Master_info '' to hash table
Jul 10 10:18:02 DietPi mariadbd[29601]: 2021-07-10 10:18:02 0 [Note] /usr/sbin/mariadbd: ready for connections.
Jul 10 10:18:02 DietPi mariadbd[29601]: Version: '10.5.10-MariaDB-2'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 11
Jul 10 10:18:02 DietPi systemd[1]: Started MariaDB 10.5.10 database server.
root@DietPi:~#
```

And NC is working 👍 

I will do your things @Joulinar and edit afterwards

Joulinar, 2021-07-10:
No need if MariaDB is working now. Just do a reboot test

Phil1988, 2021-07-10:
It looking good to me:

[image]

The only thing not good right now is, that I get this message:

[image]

cron will probably be done in a short time... and the imagick its an seperate problem that came up with upgrading to buster.

Joulinar, 2021-07-10:
`imagick` can be ignored. Usually it is not needed.

Cron can be checked as follow `systemctl status cron.service`

Phil1988, 2021-07-10:
cron is running 👍 
<details>
  <summary>Works</summary>

```
root@DietPi:~# systemctl status cron.service
● cron.service - Regular background program processing daemon
     Loaded: loaded (/lib/systemd/system/cron.service; disabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-07-10 10:22:12 CEST; 10min ago
       Docs: man:cron(8)
   Main PID: 1020 (cron)
      Tasks: 1 (limit: 4666)
     Memory: 11.2M
        CPU: 351ms
     CGroup: /system.slice/cron.service
             └─1020 /usr/sbin/cron -f

Jul 10 10:30:01 DietPi CRON[8250]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 10 10:30:01 DietPi CRON[8251]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jul 10 10:30:01 DietPi CRON[8252]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jul 10 10:30:01 DietPi CRON[8253]: (www-data) CMD (php -f /var/www/nextcloud/occ preview:pre-generate)
Jul 10 10:30:01 DietPi CRON[8254]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jul 10 10:30:01 DietPi CRON[8255]: (root) CMD (   PATH="$PATH:/usr/sbin:/usr/local/bin/" pihole updatechecker local)
Jul 10 10:30:01 DietPi CRON[8250]: pam_unix(cron:session): session closed for user root
Jul 10 10:30:02 DietPi CRON[8252]: pam_unix(cron:session): session closed for user www-data
Jul 10 10:30:02 DietPi CRON[8251]: (CRON) info (No MTA installed, discarding output)
Jul 10 10:30:02 DietPi CRON[8251]: pam_unix(cron:session): session closed for user www-data

```
  
</details>

Phil1988, 2021-07-10:
**EDIT:**
The maintenance mode seems to not work:
```
root@DietPi:~# ncc maintenance:mode --on
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occurred in driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:72
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(1449): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(892): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(202): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilder.php(206): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(217): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /var/www/nextcloud/lib/private/AppConfig.php(345): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(110): OC\AppConfig->loadConfigValues()
#7 /var/www/nextcloud/lib/private/AppConfig.php(301): OC\AppConfig->getApps()
#8 /var/www/nextcloud/lib/private/legacy/OC_App.php(949): OC\AppConfig->getValues()
#9 /var/www/nextcloud/lib/private/Server.php(668): OC_App::getAppVersions()
#10 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}()
#11 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet()
#13 /var/www/nextcloud/lib/private/ServerContainer.php(156): OC\AppFramework\Utility\SimpleContainer->query()
#14 /var/www/nextcloud/lib/private/Server.php(1677): OC\ServerContainer->query()
#15 /var/www/nextcloud/lib/private/Server.php(1017): OC\Server->getMemCacheFactory()
#16 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet()
#19 /var/www/nextcloud/lib/private/ServerContainer.php(156): OC\AppFramework\Utility\SimpleContainer->query()
#20 /var/www/nextcloud/lib/private/Server.php(1977): OC\ServerContainer->query()
#21 /var/www/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#22 /var/www/nextcloud/lib/private/Server.php(395): OC\Files\View->__construct()
#23 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}()
#24 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#25 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet()
#26 /var/www/nextcloud/lib/private/ServerContainer.php(156): OC\AppFramework\Utility\SimpleContainer->query()
#27 /var/www/nextcloud/lib/private/Server.php(1324): OC\ServerContainer->query()
#28 /var/www/nextcloud/lib/base.php(595): OC\Server->boot()
#29 /var/www/nextcloud/lib/base.php(1091): OC::init()
#30 /var/www/nextcloud/console.php(49): require_once('/var/www/nextcl...')
#31 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
```

As I assume the command didnt changed, there might be still an issue

Joulinar, 2021-07-10:
Issue is still the connection to MariaDB as you have following error

`Failed to connect to the database`

Phil1988, 2021-07-10:
Hmm... 

Ok so mariaDB is still kind of broken and also the cron job is not working (are these problems maybe linked?)

MichaIng, 2021-07-10:
Due to the migration to Bullseye, the migration from PHP7.3 to PHP7.4 is required, including changing the webserver configs. In this case, a reinstall should be easiest, after purging PHP7.3 packages:
```sh
G_AGP 'php7.3-'
dietpi-software reinstall 114
```

If you don't use theming in Nextcloud, you can get rid of the Imagick warning by disabling the theming app. A long discussion is going on at Nextcloud for years about whether Imagick is enough of a security issue to remove it completely, but while many alternative suggestions have been done, it is not any priority at dev side, so we need to live with this warning, or indeed install the Imagick module, while most participants of the discussion see it as unnecessary overhead in most cases and a security issue as well (as theoretically one can pass faulty code through e.g. SVGs which is then blindly executed by ImageMagick.

Phil1988, 2021-07-10:
For knowledge purposes I revert back to the state before I manually installed the php modules and try your suggestion (and of course report back).

I also found [your comment](https://help.nextcloud.com/t/imagick-missing-version-php-7-3/47765/13) regarding imagick from a couple of years ago ;)

MichaIng, 2021-07-10:
Ah, nice find. I just checked, at least the [php-imagick package on Raspbian Buster](http://raspbian.raspberrypi.org/raspbian/pool/main/p/php-imagick/) (updated 11-Jun-2019, after my post) is now the correct PHP7.3 one, so that has been solved in the meantime 😄.

Phil1988, 2021-07-10:
Yeah... well... I may be bad at this linux stuff, but at least I am decent in searching to partially overcome my shortcomes :D 

Here we go:

```
root@DietPi:~# G_AGP 'php7.3-'
[ INFO ] APT purge for: php7.3-, please wait...
[ INFO ] None of the requested packages are currently installed. Aborting...
[  OK  ] APT purge for: php7.3-
```
(so no 7.3 installed anymore)

```
root@DietPi:~# dietpi-software reinstall 114
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[  OK  ] DietPi-Software | Reinstalling Nextcloud: File sync, sharing and collab                                     oration platform
[  OK  ] DietPi-Software | Free space check: path=/ | available=673515 MiB | req                                     uired=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : redis-server
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : apache2
[  OK  ] DietPi-Services | unmask : mosquitto
[  OK  ] DietPi-Services | unmask : node-red
[  OK  ] DietPi-Services | unmask : blynkserver
[  OK  ] DietPi-Services | unmask : home-assistant
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[FAILED] DietPi-Services | ncc maintenance:mode --on
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : home-assistant
[  OK  ] DietPi-Services | stop : blynkserver
[  OK  ] DietPi-Services | stop : node-red
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : apache2
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userd                                     ata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www                                      /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/d                                     ietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/download                                     s
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_user                                     data/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Get:1 https://deb.debian.org/debian bullseye InRelease [150 kB]
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [40.1 kB]
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Get:4 https://deb.debian.org/debian bullseye-backports InRelease [37.0 kB]
Hit:5 https://repo.mosquitto.org/debian buster InRelease
Fetched 227 kB in 1s (396 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | MariaDB will be reinstalled
[ INFO ] DietPi-Software | PHP will be reinstalled
[ INFO ] DietPi-Software | Redis will be reinstalled
bad decrypt
140090161325376:error:06065064:digital envelope routines:EVP_DecryptFinal_ex:bad                                      decrypt:../crypto/evp/evp_enc.c:610:
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
[... 265 lines cut ...]
`[ INFO ] DietPi-Software | APT install for: libapache2-mod-php7.4 php7.4-apcu ph                                     p7.4-curl php7.4-gd php7.4-mbstring php7.4-xml php7.4-zip php7.4-mysql php7.4-sq                                     lite3 php7.4-redis, please wait...
`

during that, i got this message

[image]

After this, I again got the Server error via browser
```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```

```
root@DietPi:~# ncc maintenance:mode --off
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Maintenance mode already disabled
root@DietPi:~
```

This is weird...
Im going to reboot now and report back

Joulinar, 2021-07-10:
The issue seems with Redis now

`PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0`

Phil1988, 2021-07-10:
Yes, the important part during the resinstall is posted here (for faster finding for you):

```
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
.htaccess has been updated
[  OK  ] DietPi-Software | Current setting in /var/www/nextcloud/config/config.php will be preserved:   'memcache.local' => '\\OC\\Memcache\\APCu',
[ INFO ] DietPi-Software | Enabling Redis for transactional file locking.
[  OK  ] DietPi-Software | Current setting in /etc/redis/redis.conf will be preserved: unixsocket /var/run/redis/redis-server.sock
[  OK  ] DietPi-Software | Desired setting in /etc/redis/redis.conf was already set: unixsocketperm 770
[  OK  ] DietPi-Software | usermod -aG redis www-data
[  OK  ] DietPi-Software | systemctl restart redis-server
[  OK  ] DietPi-Software | Desired setting in /var/www/nextcloud/config/config.php was already set:   'filelocking.enabled' => true,
[  OK  ] DietPi-Software | Current setting in /var/www/nextcloud/config/config.php will be preserved:   'memcache.locking' => '\\OC\\Memcache\\Redis',
[  OK  ] DietPi-Software | Added setting 'hashingThreads' => 4, to /var/www/nextcloud/config/config.php after line   'version' => '20.0.5.2',
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Set mode for background jobs to 'cron'
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
All tables already up to date!
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Check columns of the comments table.
Done.
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Done.
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Check primary keys.
Done.
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Maintenance mode enabled
```

MichaIng, 2021-07-10:
Ah sorry, I forgot the wildcard:
```sh
G_AGP 'php7.3-*'
```

The issue is the following:
```
libapache2-mod-php7.4: php7.3 module already enabled, not enabling PHP 7.4
```
to resolve:
```sh
a2dismod php7.3
a2enmod php7.4
systemctl restart apache2
```
And the Apache2 PHP7.3 module should then be purged as well:
```sh
G_AGP libapache2-mod-php7.3
```

Phil1988, 2021-07-10:
It asks me something I dont even understand :D 

`root@DietPi:~# a2dismod
Your choices are: access_compat alias auth_basic authn_core authn_file authz_core authz_host authz_user autoindex deflate dir env filter headers mime mpm_prefork negotiation php7.3 reqtimeout rewrite setenvif socache_shmcb ssl status
Which module(s) do you want to disable (wildcards ok)?
`

but for the sake of knowledge (and future easier upgrades) would you like me to revert back to my snapshot and perform the 
`G_AGP 'php7.3-*'` before the reinstall?

Joulinar, 2021-07-10:
```
a2dismod php7.3
a2enmod php7.4
systemctl restart apache2
G_AGP libapache2-mod-php7.3
```

Phil1988, 2021-07-10:
done.

working now!

(testing further)

MichaIng, 2021-07-10:
The gmp and bcmath modules are two other cases which you can ignore. Those are only required for a specific password-less login method (WebAuthn).
If you don't use it, you don't require those. I wish they wouldn't add general recommendations/warnings for modules that are only required for specific features. Better to show the warning/info on the settings page where/when this feature is enabled: https://github.com/nextcloud/server/issues/22849

Probably you get the NC21 offer on next update check. Of force it from console: `ncc update:check`
_____________
Okay to sum it up, on Buster => Bullseye upgrade, prior to doing it:
```
dietpi-services stop
command -v a2dismod > /dev/null && a2dismod php7.3
# In case backup custom PHP configurations from /etc/php/7.3
G_AGP '*php7.3*'
G_EXEC rm -Rf /etc/php/7.3
dietpi-software reinstall 83 84 85 38 40 47 48 56 93 114 143
```
This reinstalls all (installed) software titles which require additional PHP modules and/or configs.

Phil1988, 2021-07-10:
I also noted everything down for the summery.

This is what I had and you better on deciding what was important and what not:
buster to bullseye

```
rm /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm /etc/apt/preferences.d/dietpi-php
rm /etc/apt/preferences.d/dietpi-openssl
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

```
sed -i 's/buster/bullseye/' /etc/apt/sources.list
sed -i 's/buster/bullseye/' /etc/apt/sources.list.d/*.list
apt update
apt upgrade
apt full-upgrade
apt autopurge
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
```

install lsof:
`sudo apt install lsof`

install mariadb-server:
```
apt install mariadb-server
dietpi-services dietpi_controlled mariadb
```

Mosquitto repo:
```
sed -i 's/bullseye/buster/' /etc/apt/sources.list.d/*mosquitto*.list
apt update
```

Installed fdisk:
`apt install fdisk`

removing php 7.3 and reinstall NC:
```
G_AGP 'php7.3-*'
dietpi-software reinstall 114
a2dismod php7.3
a2enmod php7.4
systemctl restart apache2
G_AGP libapache2-mod-php7.3
```

I tried also to install bcmath and gmp to make the message dissapear, but that didnt worked:
```
apt install php7.4-bcmath
apt install php7.4-gmp
```

MichaIng, 2021-07-10:
The Ondrej repository stuff is actually required for Stretch => Buster. We should keep those (Stretch => Buster and Buster => Bullseye) separate. But also good to not forget about it.

Same with the `fdisk` package, which became a dedicated package with Buster. That one however is not strictly required. `fdisk` is used to view and manipulate partitions, but e.g. `parted` can do the same (and is installed on all distro versions) and none of our scripts uses `fdisk`, aside of the partition/filesytem resize service on very first boot.

I'm not yet sure why the `mariadb-server` got removed as part of the Bullseye upgrade, instead of `mariadb-server-10.5` being installed, i.e. `mariadb-server-10.3` => `mariadb-server-10.5` migration. During the upgrade to Buster (10.1 => 10.3) this obviously worked fine. I need to replicate this, else jep the additional step doesn't hurt anyway, when MariaDB was installed before, of course. In your case the Nextcloud reinstall would have (re)installed MariaDB as well. `lsof` is then also installed as MariaDB dependency, so that is not strictly required as dedicated step.

I hope, when Bullseye is release at the end of the month, Mosquitto will provide a suite for it as well, so then reverting that repo to Buster won't be required.

I think we have the important parts around PHP for both upgrade steps, so no need to redo the upgrade from my end. Many thanks for doing this and reporting back how it went. A valuable documentation 👍😃!!

Joulinar, 2021-07-10:
@MichaIng 
I guess we would need to review all the repositories we add during installation?

MichaIng, 2021-07-10:
I've done this already and added Bullseye => Buster fallback when this was not found. So checking the `dietpi-software` script for `${G_DISTRO_NAME/bullseye/buster}` can be done to find affected repos. Only for Mosquitto we do not add it on Bullseye at all (as also it does not support ARMv8/arm64). I have to change that so we have a single identifier and as any next Mosquitto release won't make it into Bullseye anymore. Currently the Debian repo still ships the newest Mosquitto 2.0.11.

But I guess many repos will start supporting Bullseye soon, so this needs to be reviewed.

Phil1988, 2021-07-10:
Ok great!

So I hope this have not been too much information/command line copys, but I rather prefer to have it documented a little too much then not enough :)

A and the 
`apt install php7.4-bcmath
apt install php7.4-gmp`
did in the end work... 

[image]

maybe considerable of you install it right away if NC is installed.
I know dietpi is lightweight and we all love it that way (at least I know nobody that tested it and was not satisfied), but its also great for beginners with a great install management (dietpi-software).
So maybe these few MB prevents you from getting time consuming questions in the future :D 

Btw: I installed the brave browser just for your project and I hope I can tip you by that.
As I am a poor student, I will spend again a few bucks but not as much as I wished, so this is at least something :)
You guys and this support is really awesome!
Thank you a lot!

The only thing that is still questionable from my side is, why there is no NC update visible.
Any ideas?

Joulinar, 2021-07-10:
What is the version of NC you are running at the moment?

Phil1988, 2021-07-10:
I'm on Nextcloud 20.0.11

Joulinar, 2021-07-10:
Let it run over night. Hope it's gonna change 😃

Phil1988, 2021-07-10:
Haha ok!

[16 more replies]

closed 2021-08-29 by MichaIng