MichaIng/DietPi #4512: RPI3b+ new dietpi image + enable bluetooth FAIL.

BelgarionNL, 2021-06-25:
Details:

    Date | Sun 28 Jul 22:13:46 CEST 2019
    Bug report | N/A
    DietPi version | v6.25.3 (MichaIng/master)
    Img creator | DietPi Core Team
    Pre-image | Raspbian Lite
    SBC device | RPi 3 Model B+ (armv7l) (index=3)
    Kernel version | 

    Letsencrypt supports Free Noip.com Dynamic DNS #1159 SMP Sun Nov 4 17:50:20 GMT 2018
    Distro | stretch (index=4)
    Command | systemctl start hciuart
    Exit code | 1
    Software title | DietPi-Set_hardware

Steps to reproduce:

    Install dietpi image enable bluetooth in dietpi-config and then nada.

Expected behaviour:

    Bluetooth should be working?

Actual behaviour:

    Bluetooth does not work

Extra details:

    Running the command systemctl start hciuart fails
    Tried to disable/enable Bluetooth from the dietpi-software menu, no change

Additional logs:

root@dietpi-woonkamer:~# journalctl -u hciuart
-- Logs begin at Fri 2021-06-25 09:57:19 CEST, end at Fri 2021-06-25 09:58:02 CEST. --
Jun 25 09:57:22 dietpi-woonkamer systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jun 25 09:58:02 dietpi-woonkamer btuart[209]: Initialization timed out.
Jun 25 09:58:02 dietpi-woonkamer btuart[209]: bcm43xx_init
Jun 25 09:58:02 dietpi-woonkamer systemd[1]: hciuart.service: Control process ex
ited, code=exited, status=1/FAILURE
Jun 25 09:58:02 dietpi-woonkamer systemd[1]: hciuart.service: Failed with result
 'exit-code'.
Jun 25 09:58:02 dietpi-woonkamer systemd[1]: Failed to start Configure Bluetooth
 Modems connected by UART.

BelgarionNL, 2021-06-25:
similar to this:

https://github.com/MichaIng/DietPi/issues/3018
https://github.com/MichaIng/DietPi/issues/4060

BelgarionNL, 2021-06-25:
UPDATE: seems to work on the 7.3 beta branch

MichaIng, 2021-06-25:
Many thanks for your report.

Bluetooth on Raspberry Pi can fail for various reasons, so you'll always find many of those reports on all distros 😉.

You had a very old DietPi version when doing the report and the image is a Stretch image (not a new one as stated in the header) with a very strange kernel signature:
```
Letsencrypt supports Free Noip.com Dynamic DNS #1159 SMP Sun Nov 4 17:50:20 GMT 2018
Distro | stretch (index=4)
```

Where did you get this image from, and is the first line above really the output of `uname -a`?

BelgarionNL, 2021-06-25:
I got the latest version from your website. tried both the stable and beta 7.3

I copied the layout from another similar bug report so perhaps I forgot to change it :)

Joulinar, 2021-06-25:
Can you share `uname -a` pls. So we know what you are running

BelgarionNL, 2021-06-25:
sorry its now running raspbian os lite.

pi@rpi3-woonkamer:~ $ Linux rpi3-woonkamer 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l G                                                                         NU/Linux

Do you absolutely need it? I can flash another sd card if you want.

BelgarionNL, 2021-06-25:
[image]

I think something is going wrong here because before the reboot I had 3 options :D

MichaIng, 2021-06-25:
Ah, there it is: The serial console on ttyAMA0 needs to be disabled, as this is used for Bluetooth. If you need a serial console/device for other purpose, use ttyS0 (mini UART) for this. Toggle the device selector to [On], then reboot, and it becomes available to toggle the console on it.

BelgarionNL, 2021-06-26:
[image]
[image]
[image]

BelgarionNL, 2021-06-26:
Just going to document all my steps so you guys can point to my mistake :)

[image]
[image]
[image]
[image]
[image]

Just checking here, not changing anything:
[image]
[image]
[image]
[image]

BelgarionNL, 2021-06-26:
[image]
[image]

Joulinar, 2021-06-26:
Pla if possible don't do screen prints. Try to copy the output from SSH terminal

BelgarionNL, 2021-06-26:
updated post with text. how do I spoiler tag it or something so its not so long?

MichaIng, 2021-06-26:
Can you try the following:
```sh
sed -i '/dtoverlay=disable-wifi/d' /boot/config.txt
reboot
```
I remember a case where, if I'm not mistaken, the Bluetooth module needed to be enabled for WiFi to work, and I guess in these cases it's true as well the other way round. It's rare RPi 3 revisions where both seem to be on the same lane and always toggled together, instead of independently, as expected. I cannot find the issue currently, let's first see if this is the case for you. As it fails only after reboot, it seems to be one of the boot/firmware configs, done during firstrun setup, at least.

BelgarionNL, 2021-06-26:
yep! thats it

its pure luck that both of these rpi3b+ have this then since I got one ecently and the other one is a couple years old.

so using the GUI simply enabling wifi will also work? or should I keep using this command for new versions?

MichaIng, 2021-06-26:
Ah great that we found it.

> so using the GUI simply enabling wifi will also work?

Better keep using that command or remove the related line manually form `config.txt`, so the kernel modules and the network interface stays disabled, but only the hardware device part is enabled.

Could you check the revisions of those two RPis?
```sh
echo $G_HW_REVISION
```

BelgarionNL, 2021-06-26:
will do thanks!

a020d3

and the other one has raspbian on it that does not give me any. Once I have moved my devices over to the now working dietpi then I will give you the other revision :)

MichaIng, 2021-06-26:
On Raspbian, the following shows the revision:
```sh
mawk '/Revision/{print $3}' /proc/cpuinfo
```

BelgarionNL, 2021-06-26:
well they are the same. good to know.

MichaIng, 2021-06-26:
Oh you are right. Hmm, it is a little late before the release, but I would want to check this back with the RPi devs. Probably there is a possibility to fix this firmware-wise.

MichaIng, 2021-07-12:
Found the two related threads in the Raspberry Pi forum, and indeed it's model 3B+:
- https://www.raspberrypi.org/forums/viewtopic.php?t=276582
- https://www.raspberrypi.org/forums/viewtopic.php?t=284221

And indeed it's when disabling WiFi, that Bluetooth is disabled as well while I couldn't find a way that it's true the other way round as well, at least.

Joulinar, 2021-07-12:
@MichaIng 

this is how it looks on my system after reboot. WiFi disabled +  BT enabled

```
root@DietPi3:~# cat /boot/config.txt | grep dtoverlay
dtoverlay=dietpi-disable_vcsm
dtoverlay=disable-wifi
root@DietPi3:~# systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-07-12 23:30:32 CEST; 2min 30s ago
  Process: 224 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
 Main PID: 416 (hciattach)
    Tasks: 1 (limit: 2181)
   CGroup: /system.slice/hciuart.service
           └─416 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - b8:27:eb:7c:cb:e0

Jul 12 23:29:57 DietPi3 systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jul 12 23:30:32 DietPi3 btuart[224]: bcm43xx_init
Jul 12 23:30:32 DietPi3 btuart[224]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Jul 12 23:30:32 DietPi3 btuart[224]: Set BDADDR UART: b8:27:eb:7c:cb:e0
Jul 12 23:30:32 DietPi3 btuart[224]: Set Controller UART speed to 3000000 bit/s
Jul 12 23:30:32 DietPi3 btuart[224]: Device setup complete
Jul 12 23:30:32 DietPi3 systemd[1]: Started Configure Bluetooth Modems connected by UART.
root@DietPi3:~#
```

MichaIng, 2021-07-12:
Many thanks for testing, and strange that it works in your case. The revision is `a020d3`, right?
```sh
mawk '/Revision/{print $3}' /proc/cpuinfo
```

Joulinar, 2021-07-12:
yep

```
root@DietPi3:~# mawk '/Revision/{print $3}' /proc/cpuinfo
a020d3
root@DietPi3:~#
```

BelgarionNL, 2021-07-23:
gave this another fresh go on 1 of my 2 rpi3b+ and yea for me it fails every time :) 

weird yours is working. anyway its working with the workaround so thanks for getting this sorted 👍

MichaIng, 2021-07-24:
Nasty, as I'm not sure then how to know whether the workaround is required or not (for an implementation in our scripts) 🤔. I'll keep the issue open and see whether I can find further info about this, if it's a specific charge of the RPi 3+ builds that are affected or so, and how to detect.

MichaIng, 2021-08-14:
@BelgarionNL 
When you disable Bluetooth, but enable onboard WiFi, does the WiFi adapter/interface work? Short test:
```sh
sed -i '/dtoverlay=disable-wifi/d' /boot/config.txt
G_CONFIG_INJECT 'dtoverlay=' 'dtoverlay=disable-bt' /boot/config.txt
rm -f /etc/modprobe.d/dietpi-disable_wifi.conf
reboot
# after reboot
ip l
```

MichaIng, 2021-08-16:
Okay, indeed there seem to have been a change on the RPi 3B+ models without a revision code increment. On RPi 3A+ it behaves the same. I'm checking back another detail, probably we can implement a probe on RPi 3B+ and then, when user requests to disable WiFi while Bluetooth is enabled or the other way round, we can add an additional continue/abort prompt, informing user that at least the `dtoverlay=disable-wifi` needs to be/stay absent for Bluetooth to function, which means at some level, WiFi is enabled. Not a real issue IMO, also since we blacklist the Broadcom WiFi kernel modules, but some like to assure that certain communication channel/hardware is disabled lowest possible level.

MichaIng, 2021-08-16:
@BelgarionNL 
Could you try to install these two files into your `/boot` directory (overwriting the existing ones): https://drive.google.com/file/d/10gBNYX6UcFFwjZircjbfgOd7uKzBSnzA/view

To have those loaded, you need to at least have 32 MiB GPU memory applied, i.e. check via `dietpi-config` > `Display Options` > `GPU/RAM Memory Split` and raise it to 32 MiB, if 16 MiB are currently applied. Else the cutdown firmware is loaded, hence `start_cd.elf` and `fixup_cd.dat`. Then disable again WiFi, reboot and see if Bluetooth is still working:
```sh
G_CONFIG_INJECT 'dtoverlay=disable-wifi' 'dtoverlay=disable-wifi' /boot/config.txt
reboot
systemctl status hciuart
```

MichaIng, 2021-08-19:
Now even easier to apply the fix without manual file download:
```sh
apt install rpi-update
rpi-update
G_CONFIG_INJECT 'dtoverlay=disable-wifi' 'dtoverlay=disable-wifi' /boot/config.txt
reboot
systemctl status hciuart
```

MichaIng, 2021-12-12:
@BelgarionNL 
Could you test with recent stable kernel/firmware (`apt update && apt upgrade`)? It should be fixed now, i.e. you should be able to disable WiFi now without breaking Bluetooth.

MichaIng, 2022-07-31:
I'll mark this as closed now, AFAIK it has been solved, though I have no RPi 3+ to test.

closed 2022-07-31 by MichaIng