MichaIng/DietPi #4519: i2c bus not properly enabling

psi5asp, 2021-06-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux Pi-2A 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | Offical PSU 5V 3A 
- SDcard used | SanDisk Extreme 64

#### Additional Information (if applicable)
- Software title | i2c
- Was the software title installed freshly or updated/migrated? n/a
- Can this issue be replicated on a fresh installation of DietPi? Unknown

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. dietpi-config
2. Advanced Options
3. Change "I2C state" to [On] 
4. Reboot when prompted
5. lsmod shows i2c devices (correctly):
i2c_dev                20480  0
i2c_bcm2708            16384  0

6. however /boot/config.txt still shows (incorrectly):
dtparam=i2c1=off

7. and errors on: i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

#### Expected behaviour

- when enabling i2c1, /boot/config.txt should be set to:
dtparam=i2c1=on

#### Actual behaviour

- appears that /boot/config.txt does get changed from 
dtparam=i2c1=off 
- to 
dtparam=i2c1=on

#### Extra details

- currently when i2c is enabled in dietpi-config -> Advanced Options, running i2cdetect gives an error
# i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory 
# ls /dev/i*
ls: cannot access '/dev/i*': No such file or directory

- once _/boot/config.txt_ is changed to _dtparam=i2c1=on_, i2cdetect works fine.
-

MichaIng, 2021-06-26:
Many thanks for your report.

That is a transitional issue since we do not set the `i2c1` parameter anymore, as it is deprecated: https://github.com/MichaIng/DietPi/commit/5bf7a3ca8c4c7133c88a5de4a25edf7c6eb2365b

So if one disabled I2C once and enables it after this update, the option is left. You can fix it by disabling and enabling I2C again, then it is commented. I would have added a little change to remove this deprecated entry when enabling I2C, but I just merged the last beta for this release 🤔:
```sh
sed -i '/dtparam=i2c1=/d' /boot/config.txt
```

MichaIng, 2021-06-26:
Okay just fixed: https://github.com/MichaIng/DietPi/commit/8fe72b18f381647d19a8b41dfe02dc322d1ca519

closed 2021-06-26 by MichaIng