MichaIng/DietPi #4524: Request: qemu based VM image

dbasinge, 2021-06-27:
Can a qemu-based DietPi VM image be created for download? Have try converting the VMware based one with little luck.

ravenclaw900, 2021-06-27:
I've tried using QEMU with DietPi, but DietPi uses the `ping` command for most all of its connectivity checks, and QEMU doesn't support it without workarounds, which I haven't been able to get working. See https://wiki.qemu.org/Documentation/Networking#Enabling_ping_in_the_guest.2C_on_Linux_hosts.

MichaIng, 2021-06-27:
On a DietPi host, a QEMU DietPi guest would work with ping, as we apply this ping permission by default: #1012

I've only really played with QEMU when trying to test Raspberry Pi images, but the problem was that it could not load bootloader and kernel from an RPi image, but needed to load that externally and then only used a root filesystem (userspace). So the question is whether with x86_64 it is able to load grub and a kernel from a regular bootable disk image, or whether it loads that externally and then only loads the userspace from the disk image.

What was the exact issue you ran into with the converted VMware vmdk?

dbasinge, 2021-06-27:
> What was the exact issue you ran into with the converted VMware vmdk?

It converted, but VM got stuck at boot looking a device.

MichaIng, 2021-06-28:
Could you show the exact error message/console output of QEMU?

MichaIng, 2022-02-20:
We have a raw VM image available for download now: https://dietpi.com/downloads/images/DietPi_VM-x86_64-Bullseye.7z
This should actually work with QEMU on an x86_64 host, with emulation on any other host with performance downsides.

I'll try it as well the next days.

MichaIng, 2022-02-28:
The QCOW2 image for Proxmox supports the VirtIO SCSI controller, probably it makes sense to provide it as generic QEMU image (Proxmox uses QEMU internally): https://dietpi.com/downloads/images/testing/DietPi_Proxmox-x86_64-Bullseye.7z

MichaIng, 2022-07-31:
Both images generally work, but the initramfs fails to detect the root filesystem due to a missing controller driver. We use `tiny-initramfs` with a specific set of controller drivers which work with all virtualizers we support. Need to find out which controller is used by default for hard driver by QEMU and how to properly define the controller otherwise, e.g. `virtio-scsi` preferably.