MichaIng/DietPi #4527: All external drives wiped after update to 7.3

UNiXMIT, 2021-06-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.3.2 master
- Distro version | Buster
- Kernel version | `Linux DietPi 5.10.34-sunxi64 #21.05.1 SMP Fri May 7 15:50:19 UTC 2021 aarch64 GNU/Linux
- SBC model | PINE A64 (aarch64)
- SDcard used | SanDisk

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? Updated

#### Steps to reproduce
After updating to 7.3 I noticed that the machine was sloooow and running very hot. When checking later I noticed that everything on external drives had been completely removed! Nothing left. 3 drives wiped completely.

MichaIng, 2021-06-27:
Many thanks for your report.

> 3 drives wiped completely

Such does actually not happen. I guess it's a voltage issue and the drives were unmounted. Please check for kernel errors:
```sh
dmesg -l emerg,alert,crit,err
```

Are those drives powered via dedicated PSUs or via USB only?

UNiXMIT, 2021-06-27:
OK panic over.... They weren't wiped just unmounted after the update for some reason. I just saw the empty mounted directories and panicked.

closed 2021-06-27 by UNiXMIT