MichaIng/DietPi #4528: Option to use Dietpi-VPN as a VPN gateway

Metamba, 2021-06-27:
It will be nice to have an option to use dietpi-vpn as a VPN gateway. This will make it easy to use Dietpi to connect multiple devices over ethernet or wifi to a remote VPN server.
I think the option should whitelist devices connected over ethernet/wifi such as adding "-A INPUT -s 192.168.0.0/24 -j ACCEPT" to the killswitch.rules file.
The option should also enable ipV4 net forwarding and configure the iptables so that the Dietpi setup acts as a VPN gateway.
These are my thoughts.

MichaIng, 2021-06-28:
Many thanks for your request. I think we'll implement a general gateway toggle independently, similar how it is setup when installing WiFi Hotspot and for the WireGuard server. DietPi-VPN will then respect it an incorporate the related INPUT and FORWARD rules into the killswitch.

Metamba, 2021-06-28:
Thanks a bunch! Looking forward to having this feature in the near-future. You rock!!!