MichaIng/DietPi #4529: Unauthenticated packages called when updating

Fireman1224, 2021-06-28:
#### Details:
- Date           | Mon Jun 28 06:49:34 EDT 2021
- Bug report     | 646ee8bb-b6f9-43e5-9571-b6524520055d
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux PiHole-1 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update fails because packages cannot be authenticated
2. ...
#### Expected behaviour:

- Updates should be installed correctly prior to update release
#### Actual behaviour:

- Updates fail to install
#### Extra details:

- ...
#### Additional logs:
```
WARNING: The following packages cannot be authenticated!
  libxml2 php-common php-intl php-json
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```

Joulinar, 2021-06-28:
Hi,

can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

MichaIng, 2021-06-28:
Many thanks for your report.

Please run the following commands and paste the full output:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

It appears that you have the PHP meta packages installed, likely in combination with Ondrej's PHP repository (which DietPi-Software uses on Stretch systems to install PHP 7.3). In case this is true:
The intention was to install PHP 7.3 packages (the default shipped on Stretch PHP 7.0 is not supported anymore by most PHP applications). But installing the meta packages will lead to a rolling-release setup where newest PHP versions (currently PHP 8.0.7) are installed, which breaks webserver configurations and/or leads to multiple parallel installed PHP instances. Did you install those (namely `php-intl` and `php-json`) manually and intentionally, instead of keep using `php7.3-intl` and `php7.3-json`?

Fireman1224, 2021-06-28:
````
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jun 28 06:40:16 2021 from xxx.xxx.xxx.xxx
 ─────────────────────────────────────────────────────
 DietPi v7.2.3 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)
 - CPU temp : 35'C : 95'F (Cool runnings)
 - LAN IP : 192.168.10.10 (eth0)
 - MOTD : DietPi v7.3 has been released:
          https://dietpi.com/docs/releases/
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v7.2.3 to v7.3.2.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@PiHole-1:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@PiHole-1:~# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 https://packages.sury.org/php stretch InRelease [6,839 B]
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [                                            98.9 kB]
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG                                             Automatic Signing Key <deb@sury.org>
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [5                                            6.9 kB]
Get:6 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7                                             MB]
Get:8 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB                                            ]
Get:9 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360 B]
Reading package lists... Done
W: GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
E: The repository 'https://packages.sury.org/php stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@PiHole-1:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libgcrypt20 libtiff5
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 682 kB of archives.
After this operation, 3,072 B disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 http://mirror.umd.edu/raspbian/raspbian stretch/main armhf libgcrypt20 armhf 1.7.6-2+deb9u4 [462 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch/main armhf libtiff5 armhf 4.0.8-2+deb9u6 [220 kB]
Fetched 682 kB in 1s (658 kB/s)
(Reading database ... 54959 files and directories currently installed.)
Preparing to unpack .../libgcrypt20_1.7.6-2+deb9u4_armhf.deb ...
Unpacking libgcrypt20:armhf (1.7.6-2+deb9u4) over (1.7.6-2+deb9u3) ...
Setting up libgcrypt20:armhf (1.7.6-2+deb9u4) ...
(Reading database ... 54959 files and directories currently installed.)
Preparing to unpack .../libtiff5_4.0.8-2+deb9u6_armhf.deb ...
Unpacking libtiff5:armhf (4.0.8-2+deb9u6) over (4.0.8-2+deb9u5) ...
Setting up libtiff5:armhf (4.0.8-2+deb9u6) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
root@PiHole-1:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Reading package lists... Done
W: GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
E: The repository 'https://packages.sury.org/php stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@PiHole-1:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
````

Joulinar, 2021-06-28:
Hi Ondrej PHP repository key expired. Similar to this one https://github.com/MichaIng/DietPi/issues/4219

Pla do following

```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
/boot/dietpi/func/dietpi-set_software apt-cache clean # clean the apt cache
apt update # update again the package cache
```

Fireman1224, 2021-06-28:
@Joulinar 
This solved the dietpi-update issue.  I am having an issue with updating pi-hole, and I think that may have been the root cause of this issue.  PiHole now requires :
````
  [i] Checking for php8.0-cgi (will be installed)
  [i] Checking for php8.0-sqlite3 (will be installed)
  [i] Checking for php8.0-xml (will be installed)
  [i] Checking for php8.0-json (will be installed)
````
I may have to contact them as I'm not sure you can help with this issue...

Joulinar, 2021-06-28:
No this is a side effect of the Ondrej PHP repository. Usually @MichaIng knows how to correct this. I guess you have multiple PHP version installed at the moment.

Fireman1224, 2021-06-28:
@MichaIng 
````
root@PiHole-1:~# dpkg -l | grep php
ii  php-common                           2:84+0~20210621.36+debian9~1.gbp28513e                  all          Common files for PHP packages
ii  php-intl                             2:8.0+84+0~20210621.36+debian9~1.gbp28513e              all          Internationalisation module for PHP [default]
ii  php-json                             2:8.0+84+0~20210621.36+debian9~1.gbp28513e              all          JSON module for PHP [default]
ii  php8.0-apcu                          5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d       armhf        APC User Cache for PHP
ii  php8.0-cli                           8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        command-line interpreter for the PHP scripting language
ii  php8.0-common                        8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        documentation, examples and common module for PHP
ii  php8.0-gmagick                       2.0.6~rc1+1.1.7~rc3-5+0~20210228.21+debian9~1.gbpd1e77a armhf        Provides a wrapper to the GraphicsMagick library
ii  php8.0-intl                          8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        Internationalisation module for PHP
ii  php8.0-opcache                       8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        Zend OpCache module for PHP
ii  php8.0-phpdbg                        8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        server-side, HTML-embedded scripting language (PHPDBG binary)
ii  php8.0-readline                      8.0.7-1+0~20210604.20+debian9~1.gbp7cd340               armhf        readline module for PHP
root@PiHole-1:~#
````
maybe yes, multiple versions?

MichaIng, 2021-06-28:
Ah nope, PHP 8.0 only, so then it's fine. Did you install Ondrej's repository manually? Or probably you uninstalled PHP some time in the past and the repository was left.

However, the Pi-hole installer does not yet support PHP 8.0: The JSON module does not exist as separate package, but has been merged into the PHP core). I fixed it already, but it has not yet been released: https://github.com/pi-hole/pi-hole/pull/4135
Workaround:
```sh
sed -i 's/" \${phpVer}-json"//' '/etc/.pihole/automated install/basic-install.sh'
pihole -up
```

You'll need to apply this once again for the next Pi-hole release.

Fireman1224, 2021-06-28:
@MichaIng 
Originally I had tried to update the pihole, but when it failed. I did a little google-foo without resolution (and maybe made it worse), then I auto-removed and purged all php, then re-installed the non-8.0 version.   I'm a noob, so I don't have a lot of experience other than error messages and google...LOL

Below is the result of your code... I'm a noob so I don't have a lot of experience other than error messages and google...LOL

````
root@PiHole-1:~# pihole -up
  [i] Checking for updates...
  [i] Pi-hole Core:     up to date
  [i] Web Interface:    up to date
  [i] FTL:              update available

  [i] FTL out of date, it will be updated by the installer.

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 8.0.7
  [i] Performing unattended setup, no whiptail dialogs will be displayed

  [✓] Checking apt-get for upgraded packages... up to date!

  [i] Installer Dependency checks...
  [✓] Checking for dhcpcd5
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for whiptail
  [✓] Checking for dnsutils

  [✓] Supported OS detected
  [i] SELinux not detected
  [i] Performing reconfiguration, skipping download of local repos
  [✓] Resetting repository within /etc/.pihole...
  [✓] Resetting repository within /var/www/html/admin...
  [i] Main Dependency checks...
  [✓] Checking for cron
  [✓] Checking for curl
  [✓] Checking for iputils-ping
  [✓] Checking for lsof
  [✓] Checking for netcat
  [✓] Checking for psmisc
  [✓] Checking for sudo
  [✓] Checking for unzip
  [✓] Checking for wget
  [✓] Checking for idn2
  [✓] Checking for sqlite3
  [✓] Checking for libcap2-bin
  [✓] Checking for dns-root-data
  [✓] Checking for libcap2
  [✓] Checking for lighttpd
  [✓] Checking for php8.0-common
  [i] Checking for php8.0-cgi (will be installed)
  [i] Checking for php8.0-sqlite3 (will be installed)
  [i] Checking for php8.0-xml (will be installed)
  [i] Checking for php8.0-json (will be installed)
  [✓] Checking for php8.0-intl
  [i] Processing apt-get install(s) for: php8.0-cgi php8.0-sqlite3 php8.0-xml php8.0-json, please wait...
--------------------------------------------------------------------------------
E: Package 'php8.0-json' has no installation candidate

  Unable to complete update, please contact Pi-hole Support
````

Joulinar, 2021-06-28:
Maybe better to go back to PHP 7.4 @MichaIng ?

Fireman1224, 2021-06-28:
@Joulinar 
I thought that's what I tried before, but I still couldn't upgrade PiHole.  Either way it didn't work, so I really don't know the correct solution because both ways fail at the same place, *-json...

MichaIng, 2021-06-28:
Ah sorry space at the wrong place. Please try again:
```sh
sed -i 's/ "\${phpVer}-json"//' '/etc/.pihole/automated install/basic-install.sh'
pihole -up
```

Otherwise remove that manually from the installer  script, should be easy to find.

Fireman1224, 2021-06-29:
@MichaIng 
That worked!  But now the web interface gives me
````
503 - Service Not Available
````
LOL

MichaIng, 2021-06-29:
Let me do a quick test, probably there is something in the AdminLTE code not PHP 8.0 compatible yet.

MichaIng, 2021-06-29:
Works fine here with PHP 8.0 on Stretch. Can you check webserver logs after you tried to access?
```sh
journalctl -u lighttpd
cat /var/log/lighttpd/error.log
```

Fireman1224, 2021-06-29:
@MichaIng 
Here are the log results:
````
-- Logs begin at Thu 2016-11-03 13:16:43 EDT, end at Tue 2021-06-29 08:50:01 EDT. --
Jun 28 20:53:13 PiHole-1 systemd[1]: Starting Lighttpd Daemon...
Jun 28 20:53:15 PiHole-1 systemd[1]: Started Lighttpd Daemon.
root@PiHole-1:~# cat /var/log/lighttpd/error.log
2021-06-29 08:48:58: (mod_fastcgi.c.1926) connect failed: No such file or directory on unix:/run/php/php7.3-fpm.sock
2021-06-29 08:48:58: (mod_fastcgi.c.2842) backend died; we'll disable it for 1 seconds and send the request to another backend instead: reconnects: 0 load: 1
2021-06-29 08:48:58: (mod_fastcgi.c.1689) all handlers for /admin/index.php? on .php are down.
2021-06-29 08:49:00: (mod_fastcgi.c.2686) fcgi-server re-enabled: unix:/run/php/php7.3-fpm.sock
2021-06-29 08:49:02: (mod_fastcgi.c.1926) connect failed: No such file or directory on unix:/run/php/php7.3-fpm.sock
2021-06-29 08:49:02: (mod_fastcgi.c.2842) backend died; we'll disable it for 1 seconds and send the request to another backend instead: reconnects: 0 load: 1
2021-06-29 08:49:02: (mod_fastcgi.c.1689) all handlers for /admin/index.php? on .php are down.
2021-06-29 08:49:04: (mod_fastcgi.c.2686) fcgi-server re-enabled: unix:/run/php/php7.3-fpm.sock
````

MichaIng, 2021-06-29:
Ah, the FPM config is still in place. This needs to be overwritten with the CGI config:
```sh
cat << '_EOF_' > /etc/lighttpd/conf-available/15-fastcgi-php.conf
# -*- depends: fastcgi -*-
# /usr/share/doc/lighttpd/fastcgi.txt.gz
# http://redmine.lighttpd.net/projects/lighttpd/wiki/Docs:ConfigurationOptions#mod_fastcgi-fastcgi

## Start an FastCGI server for php (needs the php5-cgi package)
fastcgi.server += ( ".php" => 
	((
		"bin-path" => "/usr/bin/php-cgi",
		"socket" => "/var/run/lighttpd/php.socket",
		"max-procs" => 1,
		"bin-environment" => ( 
			"PHP_FCGI_CHILDREN" => "4",
			"PHP_FCGI_MAX_REQUESTS" => "10000"
		),
		"bin-copy-environment" => (
			"PATH", "SHELL", "USER"
		),
		"broken-scriptfilename" => "enable"
	))
)
_EOF_
systemctl restart lighttpd
```

Fireman1224, 2021-06-29:
@MichaIng 
That worked!  Appears everything is updated and working as it should as far as I can tell. :)

Thanks for your quick responses!!

MichaIng, 2021-06-29:
Great that it works now. I'll mark this issue as closed then. Feel free to reopen if required.

closed 2021-06-29 by MichaIng

Joulinar, 2021-06-29:
@Btw: you should think of migrating to Debian Buster as you still running Stretch. Next Debian version Bullseye will be released soon and Stretch become oldold resulting in limited support 😉

Fireman1224, 2021-06-29:
> 
> 
> @BTW: you should think of migrating to Debian Buster as you still running Stretch. Next Debian version Bullseye will be released soon and Stretch become oldold resulting in limited support 😉

Yes, thinking about moving to a Pi4 w/ SSD to run my Unifi Controller, VPN, and PiHole all on one device.  I'm investigating if I want to have everything on one or keep them separate as they are currently.

Joulinar, 2021-06-29:
VPN and Pihole did not require that much resources and should be fine to have it running on a RPi3B+. Not sure about Unifi but usually it should fit as well. As long as you did not plan to use a torrent client or anything like Nextcloud, a SSD might be oversized. Personally I'm using a 16GB USB3 stick to host my Pihole + Wireguard server.

Fireman1224, 2021-06-29:
@Joulinar 
I have each running on separate RPi3B+'s.  I have an extra and might set it up like that but with Docker to make it easier to migrate, but I'm a noob and want to learn some more before jumping off.   I'm just afraid my SD cards are going to fail before I'm ready...

Fireman1224, 2021-06-29:
Could I update the DietPi OS without having to reinstall everything or is it best to do a clean install?

MichaIng, 2021-06-29:
An upgrade worked well in all cases where we got feedback. Which software titles do you have installed?

Upgrade instructions: https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

Fireman1224, 2021-06-29:
I have WireGuard on a regular Raspbian distro, and 2 PiHoles with DietPi.  I was going to add Wireguard to one of the DietPi's and use it as both VPN and PiHole on one and free up the other two for other projects.  I'm curious when talking though as to if I could transfer the settings, keys, etc from one pi to the next to keep from having to reconfigure all my devices that can access the vpn...

Joulinar, 2021-06-29:
Wireguard can be copied quite easily. 

1. Install Wireguard using `dietpi-software`
2. Copy whole `/etc/wireguard` from old to new system
3. Change port forwarding on your router to new system
4. Restart Wireguard or PI
5. Done

Fireman1224, 2021-06-29:
Well, that does sound easy!!