MichaIng/DietPi #4532: Can't install MongoDB 4.x raspberry pi 4 (DietPi 64bit)

tindecken, 2021-06-29:
Hi, I've tried many ways to install MongoDB (Community Edition) 4.x on raspberry, but found the error as below
I followed the guide from: https://developer.mongodb.com/how-to/mongodb-on-raspberry-pi/
```
root@192.168.1.17's password:
 ─────────────────────────────────────────────────────
 DietPi v7.3.2 : 22:30 - Tue 06/29/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 51'C : 123'F (Running warm, but safe)
 - LAN IP : 192.168.1.17 (wlan0)
sucurl: (28) Resolving timed out after 3000 milliseconds
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Raspberry Pi OS Lite (64-bit))
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.
                                                                                                                                                                                                                                            root@DietPi:~# sudo apt update
Hit:1 https://download.docker.com/linux/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease
Hit:5 https://deb.debian.org/debian buster-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@DietPi:~# sudo apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~# wget -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -
OK
root@DietPi:~# echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 multiverse" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list
deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 multiverse
root@DietPi:~# sudo apt-get update
Hit:1 https://download.docker.com/linux/debian buster InRelease
Ign:2 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 InRelease
Get:3 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 Release [5378 B]
Get:4 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 Release.gpg [801 B]
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Get:6 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 Packages [9381 B]
Get:7 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse amd64 Packages [12.6 kB]
Hit:8 https://deb.debian.org/debian buster InRelease
Hit:9 https://deb.debian.org/debian buster-updates InRelease
Hit:10 https://deb.debian.org/debian-security buster/updates InRelease
Hit:11 https://deb.debian.org/debian buster-backports InRelease
Fetched 28.2 kB in 22s (1258 B/s)
Reading package lists... Done
root@DietPi:~# sudo apt-get install -y mongodb-org
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mongodb-org : Depends: mongodb-org-shell but it is not going to be installed
               Depends: mongodb-org-server but it is not going to be installed
               Depends: mongodb-org-mongos but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@DietPi:~#
```
I also tried to install from guide here: https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/, but it throws error: E: Unable to locate package mongodb-org

Could anyone please help to take a look at it? Much appreciated your help.
Regards,

MichaIng, 2021-06-29:
Many thanks for your question.

TL;DR: The MongoDB repository does not support ARM.

Debian Buster is not Ubuntu Focal, so it makes sense that there are unresolved dependencies when trying what you did. The guide for Debian you linked is of course what you need to follow: The repository for Debian Buster, not the one for Ubuntu Focal.

Now, checking the repository, indeed the one for aarch64 somehow only serves the `mongodb-mongosh` package, nothing else: http://repo.mongodb.org/apt/debian/dists/buster/mongodb-org/4.4/main/binary-arm64/Packages
Not sure why, as without an actual database server, that seems to be nonsense?
_____________________
Now back to the Ubuntu repository, which does support aarch64. You can try to identify the missing/conflicting package by repeatedly trying to install the listed packages which are "not going to be installed":
```sh
apt install mongodb-org-server
```

tindecken, 2021-06-30:
Thanks you so much for the help,

I'm trying to install it, will let you know the result as soon as possible

Joulinar, 2021-07-25:
Going to close this issue. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar