MichaIng/DietPi #4533: cannot update , out of space

KenMoreira, 2021-06-29:
#### Additional logs:
```
cp: error writing '/boot/dietpi/dietpi-ddns': No space left on device
cp: error writing '/boot/dietpi/dietpi-drive_manager': No space left on device
cp: error writing '/boot/dietpi/dietpi-login': No space left on device
cp: error writing '/boot/dietpi/dietpi-software': No space left on device
cp: error writing '/boot/dietpi/dietpi-vpn': No space left on device
cp: error writing '/boot/dietpi/func/dietpi-set_hardware': No space left on device
cp: error writing '/boot/dietpi/patch_file': No space left on device
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

root@USBridge:~# ls -Al /boot
total 40293
-rwxr-xr-x 1 root root   23219 Aug 16  2018 bcm2708-rpi-0-w.dtb
-rwxr-xr-x 1 root root   23966 Sep 18  2019 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   24229 Sep 18  2019 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   23747 Sep 18  2019 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   23671 Sep 18  2019 bcm2708-rpi-zero.dtb
-rwxr-xr-x 1 root root   24407 Sep 18  2019 bcm2708-rpi-zero-w.dtb
-rwxr-xr-x 1 root root   25293 Sep 18  2019 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   26463 Sep 18  2019 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   27082 Sep 18  2019 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root   25277 Sep 18  2019 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   40559 Sep 18  2019 bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root   52296 Sep 18  2019 bootcode.bin
-rwxr-xr-x 1 root root     142 Dec  1  2018 cmdline.txt
-rwxr-xr-x 1 root root    2370 Apr 25 01:03 config.txt
-rwxr-xr-x 1 root root   18693 Jul  2  2019 COPYING.linux
drwxr-xr-x 4 root root    3584 May  5 00:08 dietpi
-rwxr-xr-x 1 root root    6129 Aug 14  2018 dietpi-README.md
-rwxr-xr-x 1 root root   10674 Apr 24 03:41 dietpi.txt
-rwxr-xr-x 1 root root      41 Aug 16  2018 .firmware_revision
-rwxr-xr-x 1 root root    3073 Sep 18  2019 fixup4cd.dat
-rwxr-xr-x 1 root root    6169 Sep 18  2019 fixup4.dat
-rwxr-xr-x 1 root root    9249 Sep 18  2019 fixup4db.dat
-rwxr-xr-x 1 root root    9247 Sep 18  2019 fixup4x.dat
-rwxr-xr-x 1 root root    2657 Sep 18  2019 fixup_cd.dat
-rwxr-xr-x 1 root root    6736 Sep 18  2019 fixup.dat
-rwxr-xr-x 1 root root    9806 Sep 18  2019 fixup_db.dat
-rwxr-xr-x 1 root root    9808 Sep 18  2019 fixup_x.dat
-rwxr-xr-x 1 root root    1024 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x 1 root root  621568 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x 1 root root   10752 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x 1 root root     145 Jun 20  2019 issue.txt
-rwxr-xr-x 1 root root 5310552 Sep 18  2019 kernel7.img
-rwxr-xr-x 1 root root 5628360 Sep 18  2019 kernel7l.img
-rwxr-xr-x 1 root root 5027744 Sep 18  2019 kernel.img
-rwxr-xr-x 1 root root    1494 Jul  2  2019 LICENCE.broadcom
-rwxr-xr-x 1 root root   18974 Mar 13  2018 LICENSE.oracle
drwxr-xr-x 2 root root   15360 Apr 19 23:43 overlays
-rwxr-xr-x 1 root root    6129 Aug 14  2018 README.md
-rwxr-xr-x 1 root root  770560 Sep 18  2019 start4cd.elf
-rwxr-xr-x 1 root root 4732968 Sep 18  2019 start4db.elf
-rwxr-xr-x 1 root root 2769380 Sep 18  2019 start4.elf
-rwxr-xr-x 1 root root 3683656 Sep 18  2019 start4x.elf
-rwxr-xr-x 1 root root  685604 Sep 18  2019 start_cd.elf
-rwxr-xr-x 1 root root 4854696 Sep 18  2019 start_db.elf
-rwxr-xr-x 1 root root 2877956 Sep 18  2019 start.elf
-rwxr-xr-x 1 root root 3792200 Sep 18  2019 start_x.elf
drwxr-xr-x 2 root root     512 Jul  2  2019 System Volume Information

need help identifying what i can delete. and how to find out my boot partition size

Joulinar, 2021-06-29:
Hi,

you can have a look to your file system using `df -h`

As well you can search for the biggest files `du -a /boot | sort -n -r | head -n 20`

KenMoreira, 2021-06-29:
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
/boot/dietpi/dietpi-login: line 253: syntax error: unexpected end of file
root@USBridge:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  4.8G  9.2G  35% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           488M   12K  488M   1% /dev/shm
tmpfs           488M  7.1M  481M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           488M     0  488M   0% /sys/fs/cgroup
tmpfs            50M   36K   50M   1% /var/log
tmpfs           2.0G  4.0K  2.0G   1% /tmp
/dev/mmcblk0p1   42M   42M  2.5K 100% /boot

Joulinar, 2021-06-29:
what device you are running?

KenMoreira, 2021-06-29:
> what device you are running?

allo usbridge signature 

retyped it , 42128   /boot
1358    /boot/dietpi
492     /boot/overlays
250     /boot/dietpi/func
43      /boot/dietpi/misc
2       /boot/System Volume Information

Joulinar, 2021-06-29:
@MichaIng 

I guess some of the unused RPi4 kernel files can be removed?

KenMoreira, 2021-06-29:
no clue , bought with dietpi installed because i've never used ssh stuff before.

Joulinar, 2021-06-29:
the question was for the DietPi developer not for you 😉

MichaIng, 2021-06-30:
The RPi4 files are removed automatically with a kernel upgrade, so I wonder why this did not happen yet, before our scripts are being installed.

Can you share some image details, please:
```sh
cat /boot/dietpi/.hw_model
cat /etc/os-release
```

Try to remove some files and upgrade the packages manually first:
```sh
rm /boot/start4*
rm /boot/fixup4*
rm /boot/kernel7l.img
rm /boot/FSCK*
rm -R '/boot/System Volume Information'
apt update
apt full-upgrade
dietpi-update 1
```

KenMoreira, 2021-06-30:
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
/boot/dietpi/dietpi-login: line 253: syntax error: unexpected end of file
root@USBridge:~# cat /boot/dietpi/.hw_model
G_HW_MODEL=3
G_HW_MODEL_NAME='RPi CM 3+ (armv7l)'
G_HW_ARCH=2
G_HW_ARCH_NAME='armv7l'
G_HW_CPUID=0
G_HW_CPU_CORES=4
G_DISTRO=4
G_DISTRO_NAME='stretch'
G_ROOTFS_DEV='/dev/mmcblk0p2'
G_HW_UUID='ed987da7-8247-4ac6-ac0e-f58a33c89e02'
G_RASPBIAN=1
G_HW_ONBOARD_WIFI=0
G_HW_REVISION='a02100'
G_HW_PCB_REVISION=0
G_HW_MEMORY_SIZE=1024
G_HW_MANUFACTURER='Sony UK'
root@USBridge:~# http://192.168.0.37/allo/public/index.php

root@USBridge:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@USBridge:~#

KenMoreira, 2021-06-30:
root@USBridge:~# cat /boot/dietpi/.hw_model
G_HW_MODEL=3
G_HW_MODEL_NAME='RPi CM 3+ (armv7l)'
G_HW_ARCH=2
G_HW_ARCH_NAME='armv7l'
G_HW_CPUID=0
G_HW_CPU_CORES=4
G_DISTRO=4
G_DISTRO_NAME='stretch'
G_ROOTFS_DEV='/dev/mmcblk0p2'
G_HW_UUID='ed987da7-8247-4ac6-ac0e-f58a33c89e02'
G_RASPBIAN=1
G_HW_ONBOARD_WIFI=0
G_HW_REVISION='a02100'
G_HW_PCB_REVISION=0
G_HW_MEMORY_SIZE=1024
G_HW_MANUFACTURER='Sony UK'
root@USBridge:~#

KenMoreira, 2021-06-30:
> The RPi4 files are removed automatically with a kernel upgrade, so I wonder why this did not happen yet, before our scripts are being installed.
> 
> Can you share some image details, please:
> 
> ```shell
> cat /boot/dietpi/.hw_model
> cat /etc/os-release
> ```
> 
> Try to remove some files and upgrade the packages manually first:
> 
> ```shell
> rm /boot/start4*
> rm /boot/fixup4*
> rm /boot/kernel7l.img
> rm /boot/FSCK*
> rm -R '/boot/System Volume Information'
> apt update
> apt full-upgrade
> dietpi-update 1
> ```

[image]

KenMoreira, 2021-06-30:
[image]

Joulinar, 2021-06-30:
did you removed the files as indicated by @MichaIng ?

```
rm /boot/start4*
rm /boot/fixup4*
rm /boot/kernel7l.img
rm /boot/FSCK*
rm -R '/boot/System Volume Information'
apt update
apt full-upgrade
dietpi-update 1
```

KenMoreira, 2021-06-30:
They were not found.

Joulinar, 2021-07-01:
you don't have the `rm` command available anymore? Does the `apt` command is still working?

MichaIng, 2021-07-01:
Oh lol, it's a Stretch system. Strange that the RPi 4 files are present, as Stretch images were never supposed to support Raspberry Pi 4 🤔.

@KenMoreira 
Is it a new system or did you run and set it up since a while? If it's new, I suggest to download our new Raspberry Pi 32-bit image: https://dietpi.com/#download
If you have the Allo web UI installed, there is an image available with it as well.
Those are shipped with Debian Buster, a larger /boot partition that have no problem to hold multiple modern kernel versions etc.

Reminds me to ask Allo to update the DietPi images they ship.

KenMoreira, 2021-07-01:
I got it with the ui etc all pre installed. I tried to add some kernel or
whatever to run upmp but gave up so I know theirs that sitting there taking
up space.   I mainly use GMrender for audirvana

KenMoreira, 2021-07-04:
> Oh lol, it's a Stretch system. Strange that the RPi 4 files are present, as Stretch images were never supposed to support Raspberry Pi 4 🤔.
> 
> @KenMoreira
> Is it a new system or did you run and set it up since a while? If it's new, I suggest to download our new Raspberry Pi 32-bit image: https://dietpi.com/#download
> If you have the Allo web UI installed, there is an image available with it as well.
> Those are shipped with Debian Buster, a larger /boot partition that have no problem to hold multiple modern kernel versions etc.
> 
> * [x]  Reminds me to ask Allo to update the DietPi images they ship.

i have the allo web UI and i Have Debian Buster  but no space  :(

KenMoreira, 2021-07-04:
i was able to clear enough space to do the dietpi update. next ill try n figure out updating GMRender lol

Joulinar, 2021-07-04:
Before continuing, you should follow the suggestion of @MichaIng and download new Allo image with Debian Buster, because you are running Stretch. This is an old Debian version and will become oldold soon once Bullseye will be released end of the month.

KenMoreira, 2021-07-04:
I don't know how to do the imagine. Is it just a command line in ssh?

I don't have a micro SD card reader to flash my existing card.

KenMoreira, 2021-07-04:
> Before continuing, you should follow the suggestion of @MichaIng and download new Allo image with Debian Buster, because you are running Stretch. This is an old Debian version and will become oldold soon once Bullseye will be released end of the month.

[image]

Is this the correct file?

Joulinar, 2021-07-04:
Yes if you like to use the Allo Gui, you would need to download the Allo image. But you need a MicroSD card reader to be able to flash the image

KenMoreira, 2021-07-04:
Worth just waiting for a month for that new version you mentioned?  Or?

KenMoreira, 2021-07-04:
after doing some updates this happened

root@KensBridge:~# apt full-upgrade
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

Joulinar, 2021-07-05:
usually it' should not be an issue to switch from Buster to Bullseye later on. You are ok to go with current Allo Buster image

Joulinar, 2021-07-25:
Going to close the issue. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar