MichaIng/DietPi #4534: VM Image for VMware aarch64

marshallmassengill, 2021-07-01:
### Creating an image request

#### Formal device information
- Device name | aarch64 VM
- Official product URL | https://flings.vmware.com/esxi-arm-edition
- GitHub resource URL (if available) | N/A
- Image download URL | https://flings.vmware.com/esxi-arm-edition

#### Is the SBC officially supported by the Debian installer?
- Yes, Debian will work as a VM on ESXi-Arm

#### If not, are there install instructions for Debian available?
- There are a few blogs out there:
- https://www.virten.net/2020/10/vmware-tools-for-debian-10-arm64-on-esxi-arm/
- https://www.servethehome.com/getting-started-with-vmware-esxi-on-arm-with-a-raspberry-pi/

marshallmassengill, 2021-07-01:
This might also be another reason to create this: https://blogs.vmware.com/teamfusion/2021/04/fusion-on-apple-silicon-progress-update.html

MichaIng, 2021-07-05:
I mark this as closed in favour of: #3831

closed 2021-07-05 by MichaIng