MichaIng/DietPi #4536: logging /dev/console in a ram tmpfs drive

timoteicampian, 2021-07-02:
Hello

  We are using right now a headless setup of DietPi on a raspberry pi 4b for our project and we would really need to "see" what is on the screen via ssh.
  The screendump application does a pretty good job at reading the screen, but it always outputs 30 lines (even if only 2-3 lines are displayed on the screen) and it becomes confusing as - if you loop display the screendump - then the same lines are displayed more once.
  Another possible workaround would be to use syslog to log everything to a log file in some tmpfs ram drive. That would be a more direct approach, so it would be better.
  How would you propose us to approach this?... Does DietPi have some already built-in tool for this?...

MichaIng, 2021-07-05:
Many thanks for your question.

Let me try to understand first better what you want to do:
- Via SSH you cannot see the content of the "screen", respectively the local console `/dev/tty1` or `/dev/console`. SSH sessions use their own console device at `/dev/pts/0`. And by default nothing prints to a console outside of an interactive shell, aside of boot messages, kernel errors (which you can both check better via `journalctl` or `dmesg`) and the login prompt.
- Is there something specific which you expect to log to a certain console device? Any service or process can of course be started a way that it writes to a log file instead, by redirecting STDOUT and STDERR.
- If you need to record an interactive shell session, [script](https://manpages.debian.org/script) might be worth to have a look at, which is available OOTB.
- An ugly workaround to skip empty lines with screendump could be `screendump 1 | grep -v '^$'`, but I'm pretty sure, depending on what you actually want to view/catch, there are better methods, like the above ones.

timoteicampian, 2021-07-07:
Dear DietPi Support,

Many thanks for your detailed explanations.

- we were testing some [overlay](https://gist.github.com/paul-ridgway/d39cbb30530442dca416734c3ee70162) scenarios, that run before `init` start, using `pivot_root` + `chroot` in **[initramfs](https://blockdev.io/mini-guide-read-only-raspberry-pi-without-nfs/)** and call `exec /sbin/init` just after all overlays are correctly mounted, but after `exec chroot . bash -c /mnt/mnt/sd/usr/bin/ro-chroot.sh` the output of `ro-chroot.sh` was not visible in the ssh console anymore except in the `/dev/console`, therefore the need of `screendump`...

- such as we started to script a filter for `screendump` (bash/perl) to displays only the new content...

- just a couple of minutes ago, just by chance, I noticed a one-liner solution proposed by someone for a _live_ `dmesg` output, but that solution applies much better for our **live `scrrendump`** - there is it!: 

**`watch -n 0.1 screendump`**

MichaIng, 2021-07-07:
Ah yes, as said, such pre-init (and also post-init) output to any console cannot be seen on SSH session, as it runs on an own new pseudo-console device, created as part of the SSH login.

Is the script continue to run after boot, so that you need to `watch` ongoing output? While watching the console that way works, to me it sounds easier to redirect it's output into a log file. E.g. in `ro-chroot.sh` wrap the content into:
```sh
{
... script
} > /path/to/log 2>&1
```
This groups all commands and redirects all their STDOUT and STDERR into a file. That file then can be read or parsed more reliably and is assured to not miss content due to buffer size reached or console cleared for other reasons.

Probably
```sh
exec chroot . bash /mnt/mnt/sd/usr/bin/ro-chroot.sh > /path/to/log 2>&1
```
works as well, but not sure whether exec or chroot messes with the redirects somehow. Btw, omit the `-c` option, which is to execute commands, while you can skip it to execute scripts. What you do is executing the script as a command, which works as well, but is not 100% what you want 😉.

timoteicampian, 2021-07-16:
Many thanks for your advices, MichaIng! 

Both, redirection and omitting `-c` option works very fine.
it was necessary to close the standard input and output before calling the _init_ process
```sh
<&-
>&-
exec /sbin/init
```
as the log file remains somehow opened by `init` process and caused filesystem errors and the system could not mount the partition (\boot) on the next boot.

closed 2021-07-16 by timoteicampian

MichaIng, 2021-07-17:
Ah yes that makes sense, when calling init within the braces. Alternatively you can call init outside/after the closing brace to have again original STDOUT/STDERR.