MichaIng/DietPi #4537: Roon Bridge and Roon Server not appearing in Software list?

chrs8519, 2021-07-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | v7.3.2
- SBC model | RP Zero W

#### Additional Information (if applicable)
Running latest DietPi version on a Raspberry Pi Zero W
Have Allo WebGUI installed, but that was never able to enable the plugin (despite being installed)
I uninstalled the Roon Bridge plugin as it was not initializing, looking to reinstall it. Rebooted. Then went back to the software catalog. 

#### Steps to reproduce

1.Go to dietpi-software
2. Search Software, type in "Roon"
3. No Results. 
4. Go to the full Software list. Look for all Roon plugins

#### Expected behaviour
- Roon Bridge and Roon Server should be visible in the list? And able to be installed?

#### Actual behaviour
- Only result when searching "Roon" is the extension manager plugin. 
- Scanning the list of all software, Roon Bridge and Roon Server are not visible. They were a few days ago?

ravenclaw900, 2021-07-02:
Hello,
This is because Roon does not provide ARMv6 packages, and the installation is such disabled on ARMv6 devices like the RPi Zero W.

Joulinar, 2021-07-02:
on the system directly, you could run following to check status of Roon

`dietpi-software list | grep roon`

chrs8519, 2021-07-02:
> on the system directly, you could run following to check status of Roon
> 
> `dietpi-software list | grep roon`

Yep, thanks for this command. It's simply not available for ARMv6!
root@DietPi:~# dietpi-software list | grep roon
ID 86 | =0 | Roon Extension Manager: manage extensions from within Roon | +Docker | https://dietpi.com/docs/software/media/#roon-extension-manager
ID 121 | =0 | Roon Bridge: Turns device into Roon capable audio player | +ALSA DISABLED for armv6l | https://dietpi.com/docs/software/media/#roon-bridge
ID 154 | =0 | Roon Server: Roon capable audio player and core | +ALSA +FFmpeg DISABLED for armv6l | https://dietpi.com/docs/software/media/#roon-server

chrs8519, 2021-07-02:
> Hello,
> This is because Roon does not provide ARMv6 packages, and the installation is such disabled on ARMv6 devices like the RPi Zero W.

You're correct, learned something new about Roon + ARMv6. 

Will close this issue! Sorry for the churn.

closed 2021-07-02 by chrs8519