MichaIng/DietPi #4538: Error running ./PREP_SYSTEM_FOR_DIETPI.sh

wsimon98, 2021-07-04:
#### Details:
- Date           | Sun Jul  4 06:09:54 BST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=0)
- Kernel version | `Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `curl -ILfvm 10 https://github.com/MichaIng/DietPi/archive/master.tar.gz`
- Exit code      | 28
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ... This is running on raspbian. The only thing extra I've done is ran this script first for the Agon case, since I can't get the fan to work on DietPi the usual way.
curl https://download.argon40.com/argon1.sh | bash 
#### Additional logs:
```
* Expire in 0 ms for 6 (transfer 0x4698b0)
* Expire in 10000 ms for 8 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Expire in 0 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 0 ms for 1 (transfer 0x4698b0)
* Expire in 0 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 0 ms for 1 (transfer 0x4698b0)
* Expire in 0 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 1 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 2 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 3 ms for 1 (transfer 0x4698b0)
* Expire in 3 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 3 ms for 1 (transfer 0x4698b0)
* Expire in 3 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 5 ms for 1 (transfer 0x4698b0)
* Expire in 5 ms for 1 (transfer 0x4698b0)
* Expire in 4 ms for 1 (transfer 0x4698b0)
* Expire in 5 ms for 1 (transfer 0x4698b0)
* Expire in 5 ms for 1 (transfer 0x4698b0)
* Expire in 8 ms for 1 (transfer 0x4698b0)
[... 559 lines cut ...]
* Operation timed out after 5147 milliseconds with 0 out of 0 bytes received
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
* Closing connection 1
curl: (28) Operation timed out after 5147 milliseconds with 0 out of 0 bytes received
HTTP/2 302
server: GitHub.com
date: Sun, 04 Jul 2021 05:09:49 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://codeload.github.com/MichaIng/DietPi/tar.gz/master
cache-control: max-age=0, private
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; connect-src 'self' uploads.github.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com; frame-ancestors 'none'; frame-src render.githubusercontent.com render-temp.githubusercontent.com viewscreen.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/socket-worker-3f088aa2.js gist.github.com/socket-worker-3f088aa2.js
content-length: 123
x-github-request-id: 830C:7ABB:2994F03:367E1AF:60E1429D

```
---------------------------------------------------------------------
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

Joulinar, 2021-07-05:
Hi,

somehow the download seems to be very small or timing out. Can you perform following change pls

`time curl -IL https://github.com/MichaIng/DietPi/archive/master.tar.gz`

andrewstech, 2021-07-07:
@wsimon98 I managed to get the argon one case working with dietpi by using the OSMC argon one setup script. Try this command. ``` curl https://download.argon40.com/argonone-setup-osmc.sh | bash ```

Joulinar, 2021-07-07:
@andrewstech 
I guess the issue is not the Argon Case script. It's more a challenge to download DietPi scripts.

Joulinar, 2021-07-25:
Going to close the issue due to missing feedback. Feel free to reopen if needed.

closed 2021-07-25 by Joulinar