MichaIng/DietPi #4539: K3S Autoinstall get wrong hostname

amalashin, 2021-07-04:
#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
buster 10.10

- Kernel version
Linux saffron 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux

- SBC model
RPi 4 Model B (aarch64)

- Power supply used
5V 10W

- SDcard used
SanDisk ultra

#### Additional Information (if applicable)
- Software title
K3S

- Was the software title installed freshly or updated/migrated?
Freshly autonstalled

- Can this issue be replicated on a fresh installation of DietPi?
yes

#### Steps to reproduce
1. Set `dietpi.txt` options:
   - AUTO_SETUP_NET_HOSTNAME=myhostname
   - AUTO_SETUP_INSTALL_SOFTWARE_ID=193
   - SOFTWARE_K3S_EXEC=server --disable servicelb --disable traefik --disable local-storage
2. Add " cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory" to the`cmdline.txt`

#### Expected behaviour
- `kubectl get nodes` should show **myhostname** at the nodes list

#### Actual behaviour
- `kubectl get nodes` shows **dietpi** at the nodes list

#### Extra details
- by default with autoinstall option of the K3S, the system should change hostname to the AUTO_SETUP_NET_HOSTNAME

Joulinar, 2021-07-04:
Hi, 

Did you restart your system after installation?

amalashin, 2021-07-04:
Sure. After the reboot I have two nodes one with _NotReady_ state.

`root@saffron:~# kubectl get nodes -o wide
NAME      STATUS     ROLES                  AGE     VERSION        INTERNAL-IP     EXTERNAL-IP   OS-IMAGE                       KERNEL-VERSION   CONTAINER-RUNTIME
myhostname   Ready      control-plane,master   80s     v1.21.2+k3s1   192.168.6.101   <none>        Debian GNU/Linux 10 (buster)   5.10.17-v8+      containerd://1.4.4-k3s2
dietpi    NotReady   control-plane,master   4m56s   v1.21.2+k3s1   192.168.6.101   <none>        Debian GNU/Linux 10 (buster)   5.10.17-v8+      containerd://1.4.4-k3s2
`

Joulinar, 2021-07-05:
@MichaIng 
can you have a look. Another case where the hostname change would need to be initiated on fist install before any other software installation continue.

MichaIng, 2021-07-05:
I'll test if `hostname <new_hostname>` from within a script is sufficient or if a reboot is required in any case. If so, we can add a forced reboot after firstrun installs when K3s was installed, just as it was before we removed the mandatory reboot some versions ago. That would be done prior to service starts, so the node is not created in the first place.

Another good point is adding the cgroups on RPi. We do that for Docker installs on Armbian-based systems, but we should as well do it on K3s installs and on RPi. There has been some back and forth whether the memory cgroup is enabled by default or not and latest master has it again disabled: https://github.com/raspberrypi/linux/commit/2b13c54592135b6fab269517ed687fa9f80bf8e5

amalashin, 2021-07-05:
I can prove that by adding `--node-name myhostname` to the `SOFTWARE_K3S_EXEC` at _dietpi.txt_ works as a workaround now. May be it will be good idea to place this option based on hostname to the config file or to the service startup  file with `ExecStart=`.

MichaIng, 2021-07-08:
The hostname is now additionally applied via `hostname` command: https://github.com/MichaIng/DietPi/commit/b5d86e4905359cde933e2d48412c854417a15752
I'll test if this is sufficient for K3s to pick up the new one.

MichaIng, 2021-07-12:
Okay, works fine:
```sh
/boot/dietpi/func/change_hostname anotherone # The script called on first boot and by dietpi-config to change the hostname
dietpi-software install 193 # Install K3s
kubectl get nodes -o wide
```
```
NAME         STATUS   ROLES                  AGE     VERSION        INTERNAL-IP    EXTERNAL-IP   OS-IMAGE                       KERNEL-VERSION    CONTAINER-RUNTIME
anotherone   Ready    control-plane,master   2m38s   v1.21.2+k3s1   192.168.1.22   <none>        Debian GNU/Linux 10 (buster)   4.19.0-17-amd64   containerd://1.4.4-k3s2
```

This theoretically makes the reboot obsolete. Running processes which use the hostname usually need to be restarted though, including the shell itself, e.g. the bash `PS1` `\h` expands to the old hostname for already running bash instances.

closed 2021-07-24 by MichaIng