MichaIng/DietPi #4540: apt update takes forever

andreicaba, 2021-07-05:
There is no need to mention first all these, but... just in case

 DietPi v7.3.2 : 03:32 - Mon 07/05/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 58'C : 136'F (Running warm, but safe)
 - LAN IP : 192.168.1.54 (eth0)
 - MOTD : DietPi v7.3 has been released:
          https://dietpi.com/docs/releases/

I actually just wanted to report that - when I'm using raspbian and I do apt update - it takes about 5 seconds to do those

when I am doing apt update under dietpi - it takes about 30 seconds...

Can this be fixed?...

PS - it stays a lot at this message:
Reading package lists... 0%

Joulinar, 2021-07-05:
Hi,

Many thanks for your question. On DietPi, this is an expected behaviour as we keep `apt` cache compressed to reduce r/w operations. On `apt update` this needs to be uncompressed first and this is taking time. Running aSd card, we recommend to keep it compressed to enlarge runtime of you card. But if you really like, you could change the behaviour as follow.

```
echo 'Acquire::GzipIndexes "false";' > /etc/apt/apt.conf.d/98dietpi-uncompressed
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

MichaIng, 2021-07-05:
Reminds me on the option we need for that.

Joulinar, 2021-07-14:
@MichaIng 
do you like to keep the issue open and have it marked as `enhancement` ?

MichaIng, 2021-07-15:
An issue for this exists already: #3926

closed 2021-07-15 by MichaIng