MichaIng/DietPi #4541: Libump capitalization error on LXDE install

exadeci, 2021-07-05:
#### Details:
- Date           | Mon Jul  5 23:07:13 AEST 2021
- Bug report     | f21c9453-88d4-49ed-8bab-e0cdf20d1a2f
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux DietPi 5.10.34-sunxi64 #21.05.1 SMP Fri May 7 15:50:19 UTC 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `mv /usr/local/lib/libUMP* /usr/lib/`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
1. Just tried to install LXDE
#### Expected behaviour:
- It install LXDE
#### Actual behaviour:
- mv: cannot stat '/usr/local/lib/libUMP*': No such file or directory 
#### Extra details:
- `/usr/local/lib/libUMP*` should be `/usr/local/lib/libump*`
---------------------------------------------------------------------

exadeci, 2021-07-05:
I've now found this https://dietpi.com/downloads/binaries/deprecation.txt

libump was removed on:
```
v6.31
all/Ubooquity.jar
all/libump_1-1_arm64.deb
```

How come my 7.3.2 install still uses it for LXDE?

MichaIng, 2021-07-05:
Many thanks for your report.

It's doubled strange:
- The old package does contain the files with capital letters: https://dietpi.com/downloads/binaries/all/libump_1-1_arm64.deb
    So not sure why lower-case letter can be required in your case. But it doesn't make sense to invest further time in this part, as indeed it's the wrong package.
- I've no idea how migrating to the intended new packages was missing while I added the old packages to the deprecated list already. Here is how it was intended: https://github.com/MichaIng/DietPi/commit/1bd646b46251aad54b00e97bda09f68fe15865a6
    Could you try it with this change applied to your `/boot/dietpi/dietpi-software`? Or alternatively with installing the packages manually:
    ```sh
    wget https://dietpi.com/downloads/binaries/buster/libump_aarch64.deb
    wget https://dietpi.com/downloads/binaries/buster/xf86-video-fbturbo_aarch64.deb
    apt install ./libump_aarch64.deb ./xf86-video-fbturbo_aarch64.deb
    ```

Changelog: https://github.com/MichaIng/DietPi/commit/0f03627db5d433e61f756a41344784434d7a6296

exadeci, 2021-07-05:
So I used the https://github.com/MichaIng/DietPi/commit/1bd646b46251aad54b00e97bda09f68fe15865a6 `dietpi-software`
- Uninstalled LXDE
- Started the install again
- Got:
`│ E: Packages were downgraded and -y was used without --allow-downgrades.`
- Added the `allow-downgrades`
- Which did a `dpkg: warning: downgrading xf86-video-fbturbo from 1.0-1 to 0.5.1-dietpi1`

But at least it works.

MichaIng, 2021-07-06:
Thanks for testing. I'll rebuild the packages with an epoch version prefix to avoid the downgrade prompt. The old package got simply version 1.0 while the new packages follow the actual upstream version number of fbturbo, which is [0.5.1](https://github.com/ssvb/xf86-video-fbturbo/blob/master/configure.ac#L26).

The other interesting question is whether KMS/DRM actually works. Just if you have time to test:
```sh
# Check if a DRI devices exist
ls -l /dev/dri/card0
# If so, check es2gears from desktop, then
mv /etc/X11/xorg.conf{,_bak}
# Restart desktop and retest es2gears
# If it's worse, revert
mv /etc/X11/xorg.conf{_bak,}
```
Without the X11 config to use fbturbo, as long as the DRI device is present, X11 will fallback to modesetting, which in theory provides some hardware acceleration via glamor. With the latest Linux kernel provided by Armbian and the Mesa packages from Debian Buster, this might be better than fbturbo, which is not more than a little ARM-optimised (unaccelerated) framebuffer DDX. With Debian Bullseye, things might become even better with default modesetting/KMS/DRM driver, as the Mesa drivers made quite some progress with version 20+.

exadeci, 2021-07-07:
@MichaIng It seems to run at a high fps without the change until it gets warm 😄

MichaIng, 2021-07-07:
> until it gets warm

I think so, as all is still CPU rendering 😉. Is the DRI device present, so that we could try to get some GPU acceleration?

exadeci, 2021-07-07:
Oh right ok!
Yes it has a DRI device 
[image]

It came with a ARM Mali 400 MP2 GPU

Thanks for the support btw!

MichaIng, 2021-07-08:
Ah that is good. Can you then check whether the modesetting driver works, probably even better?
```sh
mv /etc/X11/xorg.conf{,_bak}
cat << '_EOF_' > /etc/X11/xorg.conf
Section "Device"
    Identifier  "Mali Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
EndSection
_EOF_
```
Restart the desktop session and see whether `es2gears` has higher fps and/or lower CPU usage, or at least not worse than with fbturbo.

exadeci, 2021-07-08:
Thanks, so I get less FPS than with the default xorg.conf

I've noticed though that for both I get:
libEGL warning: DRI2: failed to authenticate

I've found a few results about it but not sure if I should do changes or not on DietPi

MichaIng, 2021-07-08:
Looks like fbturbo can cause this when only installed, somehow. Let's rule that out and I forgot whether GLESv2 libraries are installed already:
```sh
G_AGP xserver-xorg-video-fbturbo
G_AGI libgles2 libegl1
rm /etc/X11/xorg.conf
reboot # to be sure
```
The X config can then be removed since modesetting is the only available driver and used by default, as fallback at least.

exadeci, 2021-07-08:
Here's the output of the 2 last tests and the default xorg.conf
```
root@DietPi:~# cat mali.txt
libEGL warning: DRI2: failed to authenticate
EGL_VERSION = 1.4
vertex shader info:
fragment shader info:
info:
738 frames in 5.0 seconds = 147.541 FPS
792 frames in 5.0 seconds = 158.273 FPS
868 frames in 5.0 seconds = 173.461 FPS
952 frames in 5.0 seconds = 190.400 FPS
836 frames in 5.0 seconds = 167.200 FPS
954 frames in 5.0 seconds = 190.647 FPS
950 frames in 5.0 seconds = 189.848 FPS
897 frames in 5.0 seconds = 179.364 FPS
908 frames in 5.0 seconds = 181.491 FPS
912 frames in 5.0 seconds = 182.291 FPS
824 frames in 5.0 seconds = 164.635 FPS
811 frames in 5.0 seconds = 161.973 FPS
891 frames in 5.0 seconds = 178.200 FPS
720 frames in 5.0 seconds = 144.000 FPS
888 frames in 5.0 seconds = 177.423 FPS
```

```
root@DietPi:~# cat default.txt
libEGL warning: DRI2: failed to authenticate
EGL_VERSION = 1.4
vertex shader info:
fragment shader info:
info:
1044 frames in 5.0 seconds = 208.633 FPS
1047 frames in 5.0 seconds = 209.233 FPS
1047 frames in 5.0 seconds = 209.400 FPS
1044 frames in 5.0 seconds = 208.675 FPS
1001 frames in 5.0 seconds = 200.080 FPS
1102 frames in 5.0 seconds = 220.356 FPS
```

```
root@DietPi:~# cat libgles2.txt
libEGL warning: DRI2: failed to authenticate
EGL_VERSION = 1.4
vertex shader info:
fragment shader info:
info:
947 frames in 5.0 seconds = 189.400 FPS
937 frames in 5.0 seconds = 187.325 FPS
907 frames in 5.0 seconds = 181.364 FPS
914 frames in 5.0 seconds = 182.763 FPS
1085 frames in 5.0 seconds = 217.000 FPS
1123 frames in 5.0 seconds = 224.600 FPS
1128 frames in 5.0 seconds = 225.420 FPS
1122 frames in 5.0 seconds = 224.310 FPS
1061 frames in 5.0 seconds = 212.073 FPS
1004 frames in 5.0 seconds = 200.800 FPS
954 frames in 5.0 seconds = 190.762 FPS
```

I got slightly better results with https://github.com/tipam/pi3d/issues/249:

```
root@DietPi:~# export EGL_PLATFORM=x11
root@DietPi:~# glxinfo | grep "OpenGL"
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 7.0, 128 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.3.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 18.3.6
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
root@DietPi:~# es2gears
libEGL warning: DRI2: failed to authenticate
EGL_VERSION = 1.4
vertex shader info:
fragment shader info:
info:
1112 frames in 5.0 seconds = 222.267 FPS
1117 frames in 5.0 seconds = 223.221 FPS
1108 frames in 5.0 seconds = 221.600 FPS
1119 frames in 5.0 seconds = 223.755 FPS
1117 frames in 5.0 seconds = 223.355 FPS
1112 frames in 5.0 seconds = 222.311 FPS
1117 frames in 5.0 seconds = 223.311 FPS
1085 frames in 5.0 seconds = 217.000 FPS
1085 frames in 5.0 seconds = 216.957 FPS
1073 frames in 5.0 seconds = 214.471 FPS
1067 frames in 5.0 seconds = 213.357 FPS
1039 frames in 5.0 seconds = 207.758 FPS
987 frames in 5.0 seconds = 197.361 FPS
1032 frames in 5.0 seconds = 206.400 FPS
```

I'm not sure if there's much more that can be done

MichaIng, 2021-07-09:
Okay many thanks. Comparing your last three tests, fbturbo vs modesetting with libgles2 vs the one with EGL_PLATFORM variable show overlapping results within the error range. I checked several sources and Mesa drivers got major support with the version in Debian Bullseye, so I guess on Buster it is minimal: https://wiki.debian.org/PanfrostLima

However, fbturbo doesn't seem to perform much better. I'm not sure about 2D desktop/browser performance, but it all looks to me that the effort to compile and ship fbturbo over the functional modesetting driver doesn't seem to be worth the effort and I'll hence remove that special handling for PINE A64. Also the Mesa driver is being further worked on: https://www.phoronix.com/scan.php?page=news_item&px=Mesa-Lima-Shader-Cache

Many thanks for your testing here. You can always re-apply fbturbo: https://github.com/MichaIng/DietPi/issues/4541#issuecomment-874301398
If you feel without it performance decrease in other applications, that would be good to know, but to me it looks like fbdev/fbturbo was implemented when there was simply no DRI/modesetting support kernel-wise for PINE A64, so that there was no other choice than using a framebuffer driver.

MichaIng, 2021-07-09:
Ah I just see the Buster backports provide newer versions. That might fix the "libEGL warning: DRI2: failed to authenticate" error:
```sh
apt install -t buster-backports libegl1 libgles2 libdrm2
```

exadeci, 2021-07-09:
@MichaIng Yeah it seems that nothing run much better than with no settings.

I did the `apt install -t buster-backports libegl1 libgles2 libdrm2` it didn't fix the warning and messed up the modsets I'll see if I can fix them otherwise I'll just remove this.

At this low resolution I'm only getting about 230fps anyway so I don't think it would run much better at the actual screen resolution.

closed 2021-07-24 by MichaIng