MichaIng/DietPi #4542: grub-pc error during upgrade from 6.34.3 to 6.35.0

brianbloom, 2021-07-06:
#### Details:
- Date           | Mon Jul  5 22:44:43 EDT 2021
- Bug report     | da7d6539-3cce-4e61-9fee-7fc52a1bfeb7
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | Brian Bloom (bbgh@mooman.com)
- Pre-image      | Debian 10.6
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | Linux ook 4.19.0-11-amd64 #1 SMP Debian 4.19.146-1 (2020-09-17) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

- Was applying the recommended update via dietpi-upgrade script. Completes the apt update, then begins to do upgrade, and then fails with message below
#### Expected behaviour:

- Should have completed upgrade with no error
#### Actual behaviour:

- Fails with shown Grub error. Repeating upgrade attempt fails at some place immediately.
#### Extra details:

- This install is running in a VM on a Qnap nas, but has been working fine for at least a year, through multiple upgrades.
- I think the main drive volume is /dev/vda1 based on df output:
```
root@ook:~# df
Filesystem              1K-blocks     Used Available Use% Mounted on
udev                        10240        0     10240   0% /dev
/dev/vda1                28768292  1252768  26031136   5% /
tmpfs                     1021544        0   1021544   0% /dev/shm
tmpfs                     1021544    74112    947432   8% /run
tmpfs                        5120        0      5120   0% /run/lock
tmpfs                     1021544        0   1021544   0% /sys/fs/cgroup
tmpfs                       51200       96     51104   1% /var/log
tmpfs                     1020928        0   1020928   0% /tmp
```
#### Additional logs:
```
Setting up grub-pc (2.02+dfsg1-20+deb10u4) ...
/dev/sda does not exist, so cannot grub-install to it!
You must correct your GRUB install devices before proceeding:

  DEBIAN_FRONTEND=dialog dpkg --configure grub-pc
  dpkg --configure -a
dpkg: error processing package grub-pc (--configure):
 installed grub-pc package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 grub-pc
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2021-07-06:
Hi,

Could you try running `apt update && apt upgrade` manually

MichaIng, 2021-07-06:
This is the result of #3700. While it is not great, it's better than running into an unbootable system. A reinstall of the package is not required, instead following the suggestion of the error prompt will allow you to choose/verify your root drive, which should be `/dev/vda`:
```sh
DEBIAN_FRONTEND=dialog dpkg --configure grub-pc
dpkg --configure -a
```

brianbloom, 2021-07-06:
Are these statements I should run at the bash prompt?  I wasn't clear where that text was supposed to go.

> This is the result of #3700. While it is not great, it's better than running into an unbootable system. A reinstall of the package is not required, instead following the suggestion of the error prompt will allow you to choose/verify your root drive, which should be `/dev/vda`:
> 
> ```shell
> DEBIAN_FRONTEND=dialog dpkg --configure grub-pc
> dpkg --configure -a
> ```

MichaIng, 2021-07-06:
Yes you can copy&paste those two lines into ant console or terminal emulator.

brianbloom, 2021-07-06:
> 
> 
> Hi,
> 
> Could you try running `apt update && apt upgrade` manually

Manually running the apt upgrade, I got to the screen and selected the /dev/vda option, as per @Michalng.
It appeared to complete successfully so I issued a shutdown -r and the VM booted back up apparently ok.

I then ran the dietpi-update again which showed no errors, and it now shows me at v7.3.2!

[image]

MichaIng, 2021-07-06:
Great that it worked.

In case this was a question: `v7.3.2` is the correct latest version. `v6.35` is a required intermediate to migrate to v7, as the previous updater didn't allow major version upgrades.

brianbloom, 2021-07-06:
So I appear to be operational again, and can close this issue unless you want it open until a scripted fix is possible.

MichaIng, 2021-07-07:
In theory we could identify the parent of the root filesystem on first boot and adjust the grub debconf entry accordingly 🤔. This would align the behaviour on systems with regular /dev/sda and cases where the image was flashed to an SD card, eMMC module, NVMe or such.

@brianbloom 
Out of interest, since we do not recognise `/dev/vdX` as known device type currently, and hence e.g. do not auto-expand the root filesystem automatically on boot in this case, what kind of drive is it?

MichaIng, 2021-08-19:
Fixed with: https://github.com/MichaIng/DietPi/commit/d4300942479be91f4cc421b7ae7079fc19f19f36
Changelog: https://github.com/MichaIng/DietPi/commit/9c4be9bd2ecbd6d3bdcd9ca815a25c87cf3ea3da
- Lol, I needed to explain a little more why this is an enhancement.

closed 2021-08-20 by MichaIng