MichaIng/DietPi #4546: Updating node-red erases npm bin?

exadeci, 2021-07-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.3.2
- Raspberry 4

#### Steps to reproduce
1. With an existing install of node-red
2. systemctl stop node-red
3. npm up -g --unsafe-perm node-red

#### Expected behaviour
- Updates node-red

#### Actual behaviour
- Removes the NPM bin 😮 
[image]

#### Extra details
- I've reproduced it twice
- Did a `dietpi-software reinstall 9` to re-install npm.

exadeci, 2021-07-08:
Managed to reproduce it on my `PINE A64 (aarch64)`

exadeci, 2021-07-08:
While it isn't a dietpi issue pre se the recommended way to upgrade node-red may not be the best.
It seems that the bug is due to the fact that node-red isn't installed as part of the global npm install:
```
root@DietPi:~# npm list -g
/usr/local/lib
└── npm@7.19.1
```

Installing node-red through npm doesn't trigger the same odd bug:
```
root@DietPi:~# npm list -g
/usr/local/lib
├── node-red@1.3.5
└── npm@7.19.1

root@DietPi:~# npm up -g node-red
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.0: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated bcrypt@3.0.8: versions < v5.0.0 do not handle NUL in passwords properly
npm WARN deprecated bcrypt@3.0.6: versions < v5.0.0 do not handle NUL in passwords properly
npm WARN deprecated node-pre-gyp@0.14.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated node-pre-gyp@0.12.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future

changed 362 packages, and audited 363 packages in 1m

7 packages are looking for funding
  run `npm fund` for details

13 vulnerabilities (10 moderate, 3 high)

To address all issues, run:
  npm audit fix

Run `npm audit` for details.
root@DietPi:~# npm --version
7.19.1
```

It's definitely not a major issue but an interesting bug

Joulinar, 2021-07-08:
Hi,

Thx for sharing. Basically we had a similar issue reported on RPi Zero W ( ARMv6 ) https://dietpi.com/phpbb/viewtopic.php?p=34205#p34205

Interesting that this is happening on RPi4 as well know

MichaIng, 2021-07-08:
Luckily we install Node-RED as local module, not as global module 😄. To update:
```sh
systemctl stop node-red
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm up --no-audit node-red
systemctl start node-red
```
Compare with our docs: https://dietpi.com/docs/software/hardware_projects/#node-red

Strange behaviour that installing Node-RED sometimes purges everything else. In the linked case we assumed it's due to an old Node version as the issue was solved after enabling latest Node on ARMv6. Your Node is up-to-update?
```sh
node -v
```

exadeci, 2021-07-08:
Right ! I was following the forum but it's a bit old thanks!

closed 2021-07-08 by exadeci