MichaIng/DietPi #4549: Problem with space

maury77, 2021-07-09:
I try too ipdate diet-pi bu I have problem with the space

The SD card is 32GB but 

I have this situation

root@sdomotica:~# df -h
File system     Dim. Usati Dispon. Uso% Montato su
/dev/root        15G   14G       0 100% /
devtmpfs        455M     0    455M   0% /dev
tmpfs           487M     0    487M   0% /dev/shm
tmpfs           487M   15M    473M   4% /run
tmpfs           5,0M  4,0K    5,0M   1% /run/lock
tmpfs           487M     0    487M   0% /sys/fs/cgroup
tmpfs            50M   32K     50M   1% /var/log
tmpfs           488M     0    488M   0% /tmp
/dev/mmcblk0p1   42M   27M     16M  63% /boot

lug 09 11:35:27 sdomotica homeassistant[867]: OSError: [Errno 28] No space left on device

is there anything I can do ?

Joulinar, 2021-07-09:
Hi,

yes you can clean your storage 😉 

Your Root file system is completely used
`/dev/root 15G 14G 0 100% /`

Do a search for the biggest files.
`du -a / | sort -n -r | head -n 20`

maury77, 2021-07-09:
sorry hai resolved with  dietpi-drive_manager

closed 2021-07-09 by maury77

Joulinar, 2021-07-09:
How did you resolved the free space issue using drive manager 🤔

MichaIng, 2021-07-09:
Just as general addition:
```sh
dietpi-cleaner
```
Especially the `-dev` package removal can free some hundreds of MiB in case, which are usually only required once on install, to compile software, but not to run it.

In your case probably expanding the filesystem worked when you migrated the image from a 16 GiB SD card to a 32 GiB one?

maury77, 2021-07-09:
> Come hai risolto il problema dello spazio libero utilizzando Gestione unità? 🤔

I expand filesystem to all size of my SD