MichaIng/DietPi #4550: Upgrade python to 3.8.10 but some lib missing

maury77, 2021-07-10:
I update python 3.8.10 with dietpi-software reintsall 157 with success but some some libraries seem missing from version 3.8.5

#### Required Information
- DietPi version | '7.3.2'
- Distro version | `10.10'
- Kernel version | `5.10.17-v7`
- SBC model | `RPi 3 Model B+ (armv7l)'

#### Steps to reproduce
. /home/homeassistant/pyenv-activate.sh
python3 -m pip install numpy

#### Expected behaviour

 ERROR: Command errored out with exit status 1:
   command: /home/homeassistant/.pyenv/versions/3.8.10/bin/python3 /home/homeassistant/.pyenv/versions/3.8.10/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpah6771ax
       cwd: /tmp/pip-install-_nu8s75q/numpy_7c9c93fea2bc412d84e1c26a6b710259
  Complete output (1513 lines):
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/mtrand.pyx
  Processing numpy/random/_pcg64.pyx
  Processing numpy/random/bit_generator.pyx
  Processing numpy/random/_sfc64.pyx
  Processing numpy/random/_common.pyx
  Processing numpy/random/_bounded_integers.pyx.in
  Processing numpy/random/_philox.pyx
  Processing numpy/random/_generator.pyx
  Processing numpy/random/_mt19937.pyx
  Cythonizing sources
  blas_opt_info:
  blas_mkl_info:
  customize UnixCCompiler
    libraries mkl_rt not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  blis_info:
    libraries blis not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  openblas_info:
    libraries openblas not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    **NOT AVAILABLE**
  
  accelerate_info:
    **NOT AVAILABLE**
  
  atlas_3_10_blas_threads_info:
  Setting PTATLAS=ATLAS
    libraries tatlas not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf/atlas', '/usr/lib/arm-linux-gnueabihf']
    **NOT AVAILABLE**
  
  atlas_3_10_blas_info:
    libraries satlas not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf/atlas', '/usr/lib/arm-linux-gnueabihf']
    **NOT AVAILABLE**
  
  atlas_blas_threads_info:
  Setting PTATLAS=ATLAS
    libraries ptf77blas,ptcblas,atlas not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf/atlas', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  atlas_blas_info:
  get_default_fcompiler: matching types: '['gnu95', 'intel', 'lahey', 'pg', 'nv', 'absoft', 'nag', 'vast', 'compaq', 'intele', 'intelem', 'gnu', 'g95', 'pathf95', 'nagfor', 'fujitsu']'
  customize Gnu95FCompiler
  Found executable /usr/bin/gfortran
  customize Gnu95FCompiler
  customize Gnu95FCompiler using config
  compiling '_configtest.c':

.......

ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy which use PEP 517 and cannot be installed directly

MichaIng, 2021-07-10:
Strange, what worked on Python 3.8.5 should work on Python 3.8.10 just the same way.

As piwheels.org ships pre-compiled wheels for numpy, can you try if this works for pip in pyenv as well:
```sh
echo -e '[global]\nextra-index-url=https://www.piwheels.org/simple/' > /etc/pip.conf
sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh
python3 -m pip install numpy
exit
```

maury77, 2021-07-11:
I try but have same error

maury77, 2021-07-13:
any idea for resolve? Can I install manual library?

MichaIng, 2021-07-14:
Can you please paste the output of all commands from above? It worked fine here:
```
2021-07-14 15:53:25 root@VM-Buster:~# sudo -u homeassistant bash
 ─────────────────────────────────────────────────────
 DietPi v7.4.-1 (dev) : 15:54 - Wed 07/14/21
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.1.22 (eth0)
 - Freespace (RootFS) : 6.1G
 - MOTD : DietPi v7.3 has been released:
          https://dietpi.com/docs/releases/
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Debian mini.iso)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

2021-07-14 15:54:24 homeassistant@VM-Buster:/root$ . /home/homeassistant/pyenv-activate.sh
2021-07-14 15:54:24 homeassistant@VM-Buster:~$ python3 -m pip install numpy
Collecting numpy
  Downloading numpy-1.21.0-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.7 MB)
     |████████████████████████████████| 15.7 MB 16.3 MB/s
Installing collected packages: numpy
Successfully installed numpy-1.21.0
WARNING: You are using pip version 20.2.4; however, version 21.1.3 is available.
You should consider upgrading via the '/home/homeassistant/.pyenv/versions/3.8.11/bin/python3 -m pip install --upgrade pip' command.
2021-07-14 15:54:50 homeassistant@VM-Buster:~$
```
I'll try as well on RPi.

MichaIng, 2021-07-14:
Okay, the HA Python environment cannot use piwheels.org for numpy, as it only builds wheels for Python 3.7, the version shipped by the Raspbian package repository. But other wheels are built for "any" Python 3 version, e.g. [home-assistant-frontend](https://piwheels.org/project/home-assistant-frontend/), so it's totally worth it to install the pip config file. It needs to be placed into a different location:
```sh
mkdir -p /home/homeassistant/.pip
echo -e '[global]\nextra-index-url=https://www.piwheels.org/simple/' > /home/homeassistant/.pip/pip.conf
```
________
`numpy` install/built on RPi worked fine here as well:
```
2021-07-14 17:31:23 homeassistant@Testing-ARMv6-Buster:~$ python3 -m pip install numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting numpy
  Using cached numpy-1.21.0.zip (10.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: numpy
  Building wheel for numpy (PEP 517) ... done
  Created wheel for numpy: filename=numpy-1.21.0-cp38-cp38-linux_armv7l.whl size=17206279 sha256=363a277c80086120ac2fd9305ffb38cacef3d1d7c8b45c4f211b1a449f6666b6
  Stored in directory: /home/homeassistant/.cache/pip/wheels/60/65/cf/1eb6d9255266eea1d6ffddce9ebc4f388bcc8460d616f17eb0
Successfully built numpy
Installing collected packages: numpy
Successfully installed numpy-1.21.0
WARNING: You are using pip version 20.2.4; however, version 21.1.3 is available.
You should consider upgrading via the '/home/homeassistant/.pyenv/versions/3.8.10/bin/python3 -m pip install --upgrade pip' command.
```

maury77, 2021-07-14:
I try it

homeassistant@sdomotica:/root$ mkdir -p /home/homeassistant/.pip
homeassistant@sdomotica:/root$  echo -e ' [global]\nextra-index-url=https://www.piwheels.org/simple/ '  > /home/homeassistant/.pip/pip.conf

omeassistant@sdomotica:/root$  . /home/homeassistant/pyenv-activate.sh
homeassistant@sdomotica:~$ python3 -m pip install numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting numpy
  Downloading numpy-1.21.0.zip (10.3 MB)
     |████████████████████████████████| 10.3 MB 8.4 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: numpy
  Building wheel for numpy (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/homeassistant/.pyenv/versions/3.8.10/bin/python3 /home/homeassistant/.pyenv/versions/3.8.10/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmp0k0oqgky
       cwd: /tmp/pip-install-yzvkley1/numpy_11507a2ab7f14c7db8a0b09be8716f4e
  Complete output (1403 lines):
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/mtrand.pyx
  Processing numpy/random/_pcg64.pyx
  Processing numpy/random/bit_generator.pyx
  Processing numpy/random/_sfc64.pyx
  Processing numpy/random/_common.pyx
  Processing numpy/random/_bounded_integers.pyx.in
  Processing numpy/random/_philox.pyx
  Processing numpy/random/_generator.pyx
  Processing numpy/random/_mt19937.pyx
  Cythonizing sources
  blas_opt_info:
  blas_mkl_info:
  customize UnixCCompiler
    libraries mkl_rt not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  blis_info:
    libraries blis not found in ['/home/homeassistant/.pyenv/versions/3.8.10/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
    NOT AVAILABLE
  
  openblas_info:
  C compiler: gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC
  
  creating /tmp/tmpxa9lc_ie/tmp
  creating /tmp/tmpxa9lc_ie/tmp/tmpxa9lc_ie
  compile options: '-c'
  gcc: /tmp/tmpxa9lc_ie/source.c
  gcc -pthread /tmp/tmpxa9lc_ie/tmp/tmpxa9lc_ie/source.o -L/usr/lib/arm-linux-gnueabihf -lopenblas -o /tmp/tmpxa9lc_ie/a.out
    FOUND:
      libraries = ['openblas', 'openblas']
      library_dirs = ['/usr/lib/arm-linux-gnueabihf']
      language = c

MichaIng, 2021-07-14:
No idea, I'm installing it on the exact same environment without this error 🤔. Is `g++` already installed on the system?
```sh
apt install g++
```
I recognised it is required now on RPi to build greenlet. And for Pillow additionally:
```sh
apt install libjpeg-dev
```
In my case, without those, the service failed to full start as those are required for the HTTP frontend. Please try to remove the installed dependencies as well and reinstall HA with the above two packages, ah and the Python patch version can be incremented as well:
```sh
mv /mnt/dietpi_userdata/homeassistant/deps{,_bak}
sed -i 's/3\.8\.10/3.8.11/' /boot/dietpi/dietpi-software
dietpi-software reinstall 157
```

maury77, 2021-07-15:
Unfortunately even the 3.8.11 and your procedure has not changed 

ug 15 20:10:46 sdomotica homeassistant[961]: 2021-07-15 20:10:46 ERROR (SyncWorker_6) [homeassistant.util.package] Unable to install package aiopylgtv==0.4.0: ERROR: Command errored out with exit status 1:
lug 15 20:10:46 sdomotica homeassistant[961]:    command: /home/homeassistant/.pyenv/versions/3.8.11/bin/python3.8 /home/homeassistant/.pyenv/versions/3.8.11/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpolbdmk9l
lug 15 20:10:46 sdomotica homeassistant[961]:        cwd: /tmp/pip-install-hakcj39_/numpy
lug 15 20:10:46 sdomotica homeassistant[961]:   Complete output (1520 lines):
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_bounded_integers.pxd.in
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/mtrand.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_pcg64.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/bit_generator.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_sfc64.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_common.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_bounded_integers.pyx.in
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_philox.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_generator.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Processing numpy/random/_mt19937.pyx
lug 15 20:10:46 sdomotica homeassistant[961]:   Cythonizing sources
lug 15 20:10:46 sdomotica homeassistant[961]:   blas_opt_info:
lug 15 20:10:46 sdomotica homeassistant[961]:   blas_mkl_info:
lug 15 20:10:46 sdomotica homeassistant[961]:   customize UnixCCompiler
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries mkl_rt not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   blis_info:
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries blis not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   openblas_info:
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries openblas not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   accelerate_info:
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   atlas_3_10_blas_threads_info:
lug 15 20:10:46 sdomotica homeassistant[961]:   Setting PTATLAS=ATLAS
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries tatlas not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   atlas_3_10_blas_info:
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries satlas not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomotica homeassistant[961]:   
lug 15 20:10:46 sdomotica homeassistant[961]:   atlas_blas_threads_info:
lug 15 20:10:46 sdomotica homeassistant[961]:   Setting PTATLAS=ATLAS
lug 15 20:10:46 sdomotica homeassistant[961]:     libraries ptf77blas,ptcblas,atlas not found in ['/home/homeassistant/.pyenv/versions/3.8.11/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
lug 15 20:10:46 sdomotica homeassistant[961]:     NOT AVAILABLE
lug 15 20:10:46 sdomot

maury77, 2021-07-16:
I Restore old backup with 3.8.5 and I resolve it

closed 2021-07-16 by maury77