MichaIng/DietPi #4552: DietPi-Software Update: Radarr

psi5asp, 2021-07-10:
### Creating a software request

#### Formal software information
- Software title | Radarr v3.2.1 - Master/Develop
- Short description | Automatically download your favorite movies, as they become available.
- Official URL (if available) |  https://radarr.video
- GitHub URL (if available) |  https://github.com/Radarr/Radarr
- (Official) install documentation | https://wiki.servarr.com/radarr/installation

#### Are there similar/alternative software titles available with DietPi-Software?
Yes - but older version using mono and more resources.
- ...

#### What makes your requested software better than the above solutions, if available?
Faster version using .net 5 instead of Mono. Less dependency install (no Mono install required, and .net is already installed as part of default image for other packages).
 ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Replace the existing install software with the new updated version of Radarr.
- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
Follow these simple steps: https://wiki.servarr.com/radarr/installation

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Not required
- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Not Required. This is a 'use the updated software version' request, not a new software request.

MichaIng, 2021-07-10:
Many thanks for your request.

In fact this migration has been done already and should have been applied with the update to DietPi v6.34 on all systems but ARMv6 RPi models (where .NET does not work). Are you running a newer DietPi version and Radarr is still running via Mono?

psi5asp, 2021-07-30:
Yea all ok. Don't know what happened, Radarr is .net version installed. Lidarr is not, so I'll raise a different request for Lidarr.

>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Checking for prerequisite software
> 
> **[ INFO ] DietPi-Software | SQLite will be reinstalled**
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Installing SQLite: Persistent single-file database system
> 
> [ INFO ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3, please wait...
> [  OK  ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Installing Radarr: automatically download movies
> 
> [  OK  ] DietPi-Software | Checking URL: https://github.com/Radarr/Radarr/releases/download/v3.2.2.5080/Radarr.master.3.2.2.5080.linux-core-arm64.tar.gz
> [  OK  ] DietPi-Software | cd /tmp/DietPi-Software
> [ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://github.com/Radarr/Radarr/releases/download/v3.2.2.5080/Radarr.master.3.2.2.5080.linux-core-arm64.tar.gz -o Radarr.master.3.2.2.5080.linux-core-arm64.tar.gz
> [ INFO ] DietPi-Software | APT install for: mediainfo libicu63, please wait...
> [  OK  ] DietPi-Software | APT install for: mediainfo libicu63
> [  OK  ] DietPi-Software | G_THREAD: All threads finished
> [  OK  ] DietPi-Software | tar xf Radarr.master.3.2.2.5080.linux-core-arm64.tar.gz
> [  OK  ] DietPi-Software | rm Radarr.master.3.2.2.5080.linux-core-arm64.tar.gz
> [  OK  ] DietPi-Software | mv Radarr /opt/radarr
>

closed 2021-07-30 by psi5asp