MichaIng/DietPi #4557: Installing G++ Multilib Problems

csharpee, 2021-07-14:
I am new to DietPi, but would I need to add to the sources list to install this package manually?

I am using the RaspPi 400 and v8arm. 
I am unsure why it is difficult to have this installed? 

I discovered related information on the web about it here: https://www.raspberrypi.org/forums/viewtopic.php?p=701320

MichaIng, 2021-07-14:
Many thanks for your question. I migrated the issue to the correct repository, as this is not related to our documentation site.

Does the following not work?
```sh
apt install g++-multilib
```

csharpee, 2021-08-09:
Yes, apt install g++-multilib doesn't work. 

I ended up cross-compiling a binary for Linux from amd64 to arm64 architecture through a virtual docker machine. 

Thanks for your reply.

MichaIng, 2021-08-09:
> Yes, apt install g++-multilib doesn't work.

What is the error message?

MichaIng, 2021-09-01:
I'll mark this as closed. Feel free to reopen if required. Would still be interesting what the error message of the package install was, as I cannot replicate any.

closed 2021-09-01 by MichaIng